mobizt/Firebase-ESP32 #161: Question - how fast can we set / update data to RTDB with this library?

derrickrc, 2021-04-30:
Hello,

I am excited to try this library. I am taking incoming data from a serial device and am effectively trying to stream it to RTDB (with timestamp). 

I have experience doing this with the Android Firebase RTDB SDK and when using `setValue()` it can write the data incredibly fast to the db without issue (near real-time).

For your library, how fast can we write values to a single node? I am hoping to repeatedly set data without having to wait for a `200 OK` or any other response from Firebase - thank you.

mobizt, 2021-05-01:
Please read [this](https://github.com/mobizt/Firebase-ESP32#authentication).

It's depends on the size of payload and auth token. Can't compare to tcp/udp socket used by Firebase SDK and Firebase Admin SDK which the header size can be very small and fastest.

The SSL handshake in first http request is nearly 1-2 seconds to wait, and after that which the session will be reused, the data transfer time will depending on the payload and auth token size as I said.

The id token when you sign in with Email/Password, size is almost 1k and used more time to transfer.

closed 2021-05-01 by mobizt

derrickrc, 2021-05-01:
Thank you for the quick response. To confirm then, will database secret
then be faster?

mobizt, 2021-05-01:
I should be like that unless insecure and obsoleted.

derrickrc, 2021-05-01:
Thanks I will give it a try.

derrickrc, 2021-05-02:
@mobizt hope you can help me, when I connect my serial device it sends the data quite fast and I can see it in the serial terminal at a fast rate using `Serial.println()`. 

However, when I add the line to set Firebase data, `loop()` slows down considerably and the serial messages seem to stack up in a queue - hopefully there is a way to write to Firebase at the rate at which the serial device sends the data. I have tried both `set` and `updateNodeSilent` methods. I am new to ESP32, any help would be much appreciated. 

My code:

```
#include "WiFi.h"
#include "FirebaseESP32.h"

const char* ssid = "SSID";
const char* password =  "pw";

#define FIREBASE_HOST "MYDB.firebaseio.com" 
#define FIREBASE_AUTH "MYSECRET" 

#define RXD2 41
#define TXD2 40

FirebaseData firebaseData;
FirebaseJson json;
String weight; 

void setup() {
   Serial.begin(115200);
   Serial2.begin(9600);
   WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
  delay(500);
  Serial.println("Connecting to WiFi..");
}
  Serial.println("Connected to the WiFi network");
  Firebase.begin(FIREBASE_HOST,FIREBASE_AUTH);
}
  
void loop() {
   weight = Serial2.readStringUntil('\x03');
   json.set("parent_001", weight);
   Serial.println("to send: " + weight); 
   Firebase.updateNodeSilent(firebaseData,"/test",json); // this seems to block and i don't know why
//      
//  if ((Firebase.setString(firebaseData,"/test",weight)) == true) {
//    Serial.println("successful write");
//  } else {
//    Serial.println("Error :(");
//  }
  Serial2.flush();
}
```

mobizt, 2021-05-02:
It's not possible because this is not the tcp/udp socket connection. 

The https connection has bigger header size and need to wait for server response for completion checking.  The async https request (multi-socket, non-block) is not available in this platform due to memory  reserved for SSL data is very huge, only async http is available in this platform with third party library but can't access all Google services as it required https.

The fastest speed for https request/response in this library is 300 ms for a request.

mobizt, 2021-05-02:
Think of you press F5 to refresh web page when the http/https request was sent.

derrickrc, 2021-05-02:
@mobizt thank you again for the quick response, that's too bad. The weird thing is though, if you look at this YT video, he is able to write random numbers to the node rapidly (I tried it as well): https://youtu.be/W1FbFqtu8J4?t=286

Is this because he is not also reading data from another serial port?

mobizt, 2021-05-02:
I don't know and it depends on your code, average 200 ms is from the iteration test to store the data in continuous loop.

derrickrc, 2021-05-02:
Thanks - I will look into async libraries. I appreciate it.

mobizt, 2021-05-02:
no such library for secure connection.

derrickrc, 2021-05-02:
Yeah, I am going to have to try and route http to my own server (e.g., via
Google Cloud Functions) and then write to RTDB from there. Thanks.