mobizt/Firebase-ESP32 #163: HELP firebase stop and restart

paxsipornax, 2021-05-17:
I use Esp32, I use bluetooth and wifi separately. That's why I stop wifi, connecting to bluetooth, I'm stopping bluetooth.
wifi connecting but firebase not working despite wifi connection

stop
   Firebase.getInt(firebaseData,"s1");
   if (firebaseData.intData() == 1) {
    Firebase.setInt(firebaseData,"s1",2);
    Firebase.end(firebaseData);
    Firebase.reconnectWiFi(false);
    WiFi.mode( WIFI_MODE_NULL );

restart

    void onDisconnect(BLEServer* pServer) {
     BLEAdvertising *pAdvertising = pServer->getAdvertising();
        pAdvertising->stop();
        BLEDevice::getAdvertising()->stop();
      WiFi.mode( WIFI_MODE_NULL );
      
      delay(500);
      WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.println("Bağlanır");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
   Serial.println(WIFI_SSID);

  Serial.println(WiFi.localIP());

  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);
  delay(500);
    }
};

mobizt, 2021-05-17:
Your problem is your WiFi configuration and not related to library.

You should set WiFi mode to WiFi_STA.

closed 2021-05-17 by mobizt

paxsipornax, 2021-05-18:
> Sizin sorununuz, WiFi yapılandırmanızdır ve kitaplıkla ilgili değildir.
> 
> WiFi modunu WiFi_STA olarak ayarlamalısınız.

WiFi.mode (WIFI_STA);
I tried but it didn't work I guess I can't stop firebase because wifi connecting is getting ip

mobizt, 2021-05-18:
Firebase is not a problem. 
It has the connection management inside which works normally.
Try Arduino examples about https request. If you can run example and sever connected, the library would work too.

mobizt, 2021-05-18:
At first, any third party integration conflict should be solved yourself.

If you cannot solve yourself, you should create the test example which minimum, clean (remove all third party libs) and clear to produce the error which I can test with no  further hardware requirement and post the error and the test code here.

If the problem is not related to the library after I've checked your example code, you should post your problem at the esp8266 forum.

mobizt, 2021-05-18:
All [examples](https://github.com/mobizt/Firebase-ESP32/blob/master/examples/Basic/Basic.ino) of library provide the code that always check the error in every function called which you never used it and never didn't provide any debug at any check point of your code.