mobizt/Firebase-ESP32 #164: Issue with token requesting.

msamoeed009, 2021-05-24:
Connecting to Wi-Fi.

19:15:51.086 -> Connected with IP: 192.168.10.19
19:15:51.086 ->
19:15:51.086 -> Firebase Client v2.2.4
19:15:51.086 ->
19:15:51.121 -> Token info: type = id token, status = on request
19:17:53.910 -> Token info: type = id token, status = error
19:17:53.910 -> Token error: code: -5, message: connection lost
19:17:53.910 -> Token info: type = id token, status = on request
19:19:53.980 -> Token info: type = id token, status = error
19:19:54.014 -> Token error: code: -5, message: connection lost
19:19:54.014 -> Token info: type = id token, status = on request
19:20:12.409 -> Token info: type = id token, status = error
19:20:12.409 -> Token error: code: -5, message: connection lost
19:20:12.409 -> Token info: type = id token, status = on request

Internet is stable. Still getting this issue?

mobizt, 2021-05-24:
The reasons can be your WiFi AP does not have the internet connectivity or the server is not valid.

First, try to reset router, change WiFi AP or connect via mobile WiFi hotspot.

Then, run this example to test your internet connection.

In Arduino IDE, select board, ESP32 Dev Module, select Files > Examples > HTTPClient  > BasicHttpsClient

If you can run above example, the internet is ok otherwise not ok.

If internet is ok, make sure the DATABASE_URL defined in the Firebase example sketch is valid.

This is where you get the DATABASE_URL or FIREBASE_HOST in old examples

[image]

closed 2021-05-24 by mobizt

msamoeed, 2021-05-24:
Thank you, it has started to work. Thank you for your help.

msamoeed009, 2021-05-26:
Hey there, I am having trouble trying to push an array of short into the
cloudfirestore, i guess i dont understand the syntax of it. Is there a
snippet that can help?

mobizt, 2021-05-26:
You need to use [Firebase ESP Client library](https://github.com/mobizt/Firebase-ESP-Client)
This is the [examples](https://github.com/mobizt/Firebase-ESP-Client/tree/main/examples/Firestore).

msamoeed009, 2021-05-26:
js.set("fields/watt/doubleValue/", measurements[0]);
js.set("fields/current/arrayValue/", data); //data is an array of type
short

js.toString(content);

Serial.print("Create a document... ");

(Firebase.Firestore.createDocument(&fbdo, FIREBASE_PROJECT_ID, "" /*
databaseId can be (default) or empty */, documentPath.c_str(), content.c_str()
));
Serial.printf("ok\n%s\n\n", fbdo.payload().c_str());

Using this example but getting errors

mobizt, 2021-05-26:
You should post the error,

mobizt, 2021-05-26:
You need to post at the Firebase ESP Client repos instead of here.

Use [Discussions](https://github.com/mobizt/Firebase-ESP-Client/discussions) to post the question as it is not the library issue.

mobizt, 2021-05-26:
The field value when you construct it with FirebaseJon should follow this API reference.

https://cloud.google.com/firestore/docs/reference/rest/v1/Value

mobizt, 2021-05-26:
I can't provide Firebase products getting started or how to.

msamoeed009, 2021-05-26:
Ok, I will check.

String documentPath = String((auth.token.uid).c_str());

for (int x = 0; x < LOCAL_MEASUREMENTS - 1; x++)
{
data[x] = measurements[x];
}

js.set("fields/watt/doubleValue/", measurements[0]);

js.set("fields/current/arrayValue/", data); //data is an array of String.
(GETTING ERROR HERE)

js.toString(content);

Serial.print("Create a document... ");

(Firebase.Firestore.createDocument(&fbdo, FIREBASE_PROJECT_ID, "" /*
databaseId can be (default) or empty */, documentPath.c_str(), content.c_str
()));
Serial.printf("ok\n%s\n\n", fbdo.payload().c_str());
// else
Serial.println(fbdo.errorReason());
}

//ERROR

Invalid value at 'document.fields[1].value.array_value' (
type.googleapis.com/google.firestore.v1.ArrayValue), \"\"

mobizt, 2021-05-26:
You need to learn Firestore before use.

mobizt, 2021-05-27:
If you understand the Firestore data structure well, you can create simple or complex data without the limit.

This is the example for create doc with field array

```cpp
String content;
FirebaseJson js;

String documentPath = "test_collection/test_document";

for (size_t i = 0; i < 10;i++)
{
    String filedPath = "fields/test_array/arrayValue/values/[" + String(i) + "]/stringValue";
    String arrValue = "Hello World! " + String(i);
    js.set(filedPath.c_str(), arrValue.c_str());
}

js.toString(content);

Serial.print("Create a document... ");

if (Firebase.Firestore.createDocument(&fbdo, FIREBASE_PROJECT_ID, "", documentPath.c_str(), content.c_str()))
    Serial.printf("ok\n%s\n\n", fbdo.payload().c_str());
else
    Serial.println(fbdo.errorReason());
```

msamoeed009, 2021-05-27:
Thanks a lot. I have not worked on C alot but have been using Firebase with
dart, js and C# and never had issues. Sorry for the trouble.

binayakbishnu, 2024-04-16:
> The reasons can be your WiFi AP does not have the internet connectivity or the server is not valid.
>
> First, try to reset router, change WiFi AP or connect via mobile WiFi hotspot.
>
> Then, run this [...]

Could you reshare the image? shows not found