mobizt/Firebase-ESP32 #165: Firebase pause and restart

am-canes, 2021-06-07:
Hello,

I am using two firebase objects, one for stream and other for update data. I am doing a Ping to detect if the internet connection is good. And when I detect a bad connection I use the function pauseFirebase to pause. And I restart just when connection is good again. 
But sometimes, after a while, firebase stops to receive the stream and to update. I think that maybe it is not restarting for some reason.
What is the return of pauseFirebase? Could I use the result to know if firebase restarted?
Is there any other way to know if it is connected? And if it is not connected, is there any way to restart?

I just try to update when wifi is connected and the connection is good.

mobizt, 2021-06-07:
No extra ping or stop/start is necessary, just call **[httpConnected()](https://github.com/mobizt/Firebase-ESP32/tree/master/src#determine-the-server-connection-status)** from Firebase Data object which gives the precise status of server connection.

The stream resuming is done appropriately when timed out (heart beat data or keep-alive event data from server was missed more than 40 seconds). 

Please don't use pauseFirebase for stream Firebase Data object and let the internal flow operate normally.

closed 2021-06-07 by mobizt

mobizt, 2021-06-07:
The internet connection will be managed (resume) internally as you call **Firebase.reconnectWiFi(true);**.

Do not use reconnectWiFi(true) with the WiFiManager or code that also handles the WiFi reconnection because it will interfere to each other.