mobizt/Firebase-ESP32 #166: BUG

kamalakarrao, 2021-06-12:
Hi ,
Firstly I am very happy in using this library as it really helped me a lot in building amazing things.
Very much grateful for that.

I have been struggling with an issue and hoping to get some help in fixing it.

**Describe the bug**
Program crashes and reboot after a Successful OTA update exactly at Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH).
But same code works when we upload it through Serial from laptop using USB (Traditional Code Upload).

-----ERROR----

CORRUPT HEAP: Bad tail at 0x3ffdc7f2. Expected 0xbaad5678 got 0xbaad5600
assertion "head != NULL" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c", line 214, function: multi_heap_free
abort() was called at PC 0x4013f187 on core 1

Backtrace: 0x4008c694:0x3ffb15e0 0x4008c8c5:0x3ffb1600 0x4013f187:0x3ffb1620 0x4008c309:0x3ffb1650 0x40084d72:0x3ffb1670 0x40085179:0x3ffb1690 0x4000bec7:0x3ffb16b0 0x40173f5d:0x3ffb16d0 0x4017432d:0x3ffb16f0 0x400d9e2e:0x3ffb1710 0x400e23db:0x3ffb1730 0x400e2f44:0x3ffb1990 0x400e3385:0x3ffb19b0 0x400e38e5:0x3ffb19f0 0x400d4c42:0x3ffb1a90 0x400ec7db:0x3ffb1fb0 0x40088ddd:0x3ffb1fd0

Rebooting...

-----ERROR----

**To Reproduce**

It works with typical firebase stream code but fails after successful OTA update

**IDE and its version:**
 - Arduino
 - Version [1.8.15]

I would be extremely grateful if i get any help in guiding me on fixing the error.

Thanks
Kamal

mobizt, 2021-06-12:
You should check your device free heap.

mobizt, 2021-06-12:
The device has not enough memory to run.

closed 2021-06-12 by mobizt

kamalakarrao, 2021-06-12:
I was generating bin file for OTA update from a windows pc which probably compiled with older libraries.
I had to do that because 'Export Compiled Binary' is not working on Mac Big Sur.
But I managed to find the Bin file under arduino internal folders [Path can be found in verbose logs while compiling]
After doing that, It's working perfectly fine after OTA update.

Your Library was perfect as usual and this issue was not related to it.

Thanks 😊