mobizt/Firebase-ESP32 #170: HELP

Karandalwani, 2021-06-22:
The speed to get data from firebase is too slow. How to increase it?

mobizt, 2021-06-22:
It's not too slow .

Post your clean, clear and minimized code here.

closed 2021-06-22 by mobizt

Karandalwani, 2021-06-22:
#include <FirebaseESP8266.h>
#include <FirebaseFS.h>
#include <ESP8266WiFi.h>
#include <ArduinoJson.h>
#include <ESP8266HTTPClient.h>
#define FIREBASE_HOST "dukh2-49941-default-rtdb.firebaseio.com"
#define FIREBASE_AUTH "jlgmWCQ0AYga3Dh3k9M0rVg5b1mPpNBzNigfJgoc"
String WIFI_SSID = "Note 5";
String WIFI_PASSWORD = "KARAN123";
#define Relay1 12
int val1;
#define Relay2 15
int val2;
#define Relay3 13
int val3;
#define Relay4 14
int val4;
FirebaseData fbdo;

void setup()
{
  Serial.begin(2000000);
  pinMode(Relay1, OUTPUT);
  pinMode(Relay2, OUTPUT);
  pinMode(Relay3, OUTPUT);
  pinMode(Relay4, OUTPUT);
  
  digitalWrite(Relay1, HIGH);
  digitalWrite(Relay2, HIGH);
  digitalWrite(Relay3, HIGH);
  digitalWrite(Relay4, HIGH);

    WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("connecting");
  delay(10000);
while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
  }
  Serial.println();
  Serial.print("connected:");
  Serial.println(WiFi.localIP());
   
  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.setString(fbdo,"S1", "0");                   
  Firebase.setString(fbdo,"S2", "0");
  Firebase.setString(fbdo,"S3", "0");                   
  Firebase.setString(fbdo,"S4", "0");
}
void firebasereconnect()
{
  Serial.println("Trying to reconnect");
  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
}

void loop()
{
          Firebase.getInt(fbdo,"S1");
  val1 =  fbdo.stringData().toInt();
   
  if (val1 == 1)
  {
    digitalWrite(Relay1, LOW);
    Serial.println("light ON");
  }
  else if (val1 == 0)
  {
    digitalWrite(Relay1, HIGH);
    Serial.println("light OFF");
  }

           Firebase.getInt(fbdo,"S2");
  val2 =  fbdo.stringData().toInt();

  if (val2 == 1)
  {
    digitalWrite(Relay2, LOW);
    Serial.println("Fan ON");
  }
  else if (val2 == 0)
  {
    digitalWrite(Relay2, HIGH);
    Serial.println("Fan OFF");
  }

 Firebase.getInt(fbdo,"S3");
  val3 =  fbdo.stringData().toInt();

  if (val3 == 1)
  {
    digitalWrite(Relay3, LOW);
    Serial.println("TV ON");
  }
  else if (val3 == 0)
  {
    digitalWrite(Relay3, HIGH);
    Serial.println("TV OFF");
  }

   Firebase.getInt(fbdo,"S4");
  val4 =  fbdo.stringData().toInt();

  if (val4 == 1)
  {
    digitalWrite(Relay4, LOW);
    Serial.println("AC ON");
  }
  else if (val4 == 0)
  {
    digitalWrite(Relay4, HIGH);
    Serial.println("AC OFF");
  }

}

mobizt, 2021-06-22:
Please use github markdown and indent your code.

Karandalwani, 2021-06-22:
`#include <FirebaseESP8266.h>
#include <FirebaseFS.h>
#include <ESP8266WiFi.h>
#include <ArduinoJson.h>
#include <ESP8266HTTPClient.h>
#define FIREBASE_HOST "dukh2-49941-default-rtdb.firebaseio.com"
#define FIREBASE_AUTH "jlgmWCQ0AYga3Dh3k9M0rVg5b1mPpNBzNigfJgoc"
String WIFI_SSID = "Note 5";
String WIFI_PASSWORD = "KARAN123";
#define Relay1 12
int val1;
#define Relay2 15
int val2;
#define Relay3 13
int val3;
#define Relay4 14
int val4;
FirebaseData fbdo;

void setup()
{
  Serial.begin(2000000);
  pinMode(Relay1, OUTPUT);
  pinMode(Relay2, OUTPUT);
  pinMode(Relay3, OUTPUT);
  pinMode(Relay4, OUTPUT);
  
  digitalWrite(Relay1, HIGH);
  digitalWrite(Relay2, HIGH);
  digitalWrite(Relay3, HIGH);
  digitalWrite(Relay4, HIGH);

    WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("connecting");
  delay(10000);
while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
  }
  Serial.println();
  Serial.print("connected:");
  Serial.println(WiFi.localIP());
   
  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.setString(fbdo,"S1", "0");                   
  Firebase.setString(fbdo,"S2", "0");
  Firebase.setString(fbdo,"S3", "0");                   
  Firebase.setString(fbdo,"S4", "0");
}
void firebasereconnect()
{
  Serial.println("Trying to reconnect");
  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
}

void loop()
{
          Firebase.getInt(fbdo,"S1");
  val1 =  fbdo.stringData().toInt();
   
  if (val1 == 1)
  {
    digitalWrite(Relay1, LOW);
    Serial.println("light ON");
  }
  else if (val1 == 0)
  {
    digitalWrite(Relay1, HIGH);
    Serial.println("light OFF");
  }

           Firebase.getInt(fbdo,"S2");
  val2 =  fbdo.stringData().toInt();

  if (val2 == 1)
  {
    digitalWrite(Relay2, LOW);
    Serial.println("Fan ON");
  }
  else if (val2 == 0)
  {
    digitalWrite(Relay2, HIGH);
    Serial.println("Fan OFF");
  }

 Firebase.getInt(fbdo,"S3");
  val3 =  fbdo.stringData().toInt();

  if (val3 == 1)
  {
    digitalWrite(Relay3, LOW);
    Serial.println("TV ON");
  }
  else if (val3 == 0)
  {
    digitalWrite(Relay3, HIGH);
    Serial.println("TV OFF");
  }

   Firebase.getInt(fbdo,"S4");
  val4 =  fbdo.stringData().toInt();

  if (val4 == 1)
  {
    digitalWrite(Relay4, LOW);
    Serial.println("AC ON");
  }
  else if (val4 == 0)
  {
    digitalWrite(Relay4, HIGH);
    Serial.println("AC OFF");
  }

}`

mobizt, 2021-06-22:
Look like you don't know how to use the library properly and you are new to Firebase.

Calling get continuously in this way in the loop is not efficient and cost your data usage too much and nobody do this in practical usage.

Do thing like this is similar to you are refreshing the webpage continuously and it's nonsense in making HTTP request all the time.

If you want to listen to the value changed in the database, use stream instead.
https://github.com/mobizt/Firebase-ESP32/tree/master/examples/DataChangesListener

If you want to send data faster, use setAsync function instead.

The event data stream in REST APIs is as fast or slower a bit as MQTT or web socket technology.

Spend the time to learn the concept and how to use Firebase which available anywhere on internet.

Try another Firebase client e.g, web using Javascrip or android to learn the basic of server event listening concept.

Try [all examples](https://github.com/mobizt/Firebase-ESP32/tree/master/examples) provided in this library.

Read the document about REST API implemented in this library.
https://firebase.google.com/docs/reference/rest/database

Karandalwani, 2021-06-22:
```
#include <FirebaseESP8266.h>
#include <FirebaseFS.h>
#include <ESP8266WiFi.h>
#include <ArduinoJson.h>
#include <ESP8266HTTPClient.h>
#define FIREBASE_HOST "dukh2-49941-default-rtdb.firebaseio.com"
#define FIREBASE_AUTH "jlgmWCQ0AYga3Dh3k9M0rVg5b1mPpNBzNigfJgoc"

 
  #define Relay1 12
  int val1;
  #define Relay2 15
  int val2;
  #define Relay3 13
  int val3;
  #define Relay4 14
  int val4;
  FirebaseData fbdo;

  String WIFI_SSID = "Note 5";
  String WIFI_PASSWORD = "KARAN123";

    void setup()
      {
          Serial.begin(2000000);
          pinMode(Relay1, OUTPUT);
          pinMode(Relay2, OUTPUT);
          pinMode(Relay3, OUTPUT);
          pinMode(Relay4, OUTPUT);
  
          digitalWrite(Relay1, HIGH);
          digitalWrite(Relay2, HIGH);
          digitalWrite(Relay3, HIGH);
          digitalWrite(Relay4, HIGH);

          WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
          Serial.print("connecting");
          delay(10000);
          while (WiFi.status() != WL_CONNECTED) 
            {
                Serial.print(".");
                delay(500);
             }
           Serial.println();
           Serial.print("connected:");
           Serial.println(WiFi.localIP());
   
               Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
               Firebase.setString(fbdo,"S1", "0");                   
               Firebase.setString(fbdo,"S2", "0");
               Firebase.setString(fbdo,"S3", "0");                   
               Firebase.setString(fbdo,"S4", "0");
        }

        
    void firebasereconnect()
      {
         Serial.println("Trying to reconnect");
         Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
      }

    void loop()
      {
          Firebase.getInt(fbdo,"S1");
          val1 =  fbdo.stringData().toInt();
          
          if (val1 == 1)
           {
               digitalWrite(Relay1, LOW);
               Serial.println("light ON");
           }
          else if (val1 == 0)
           {
              digitalWrite(Relay1, HIGH);
              Serial.println("light OFF");
           }

          Firebase.getInt(fbdo,"S2");
          val2 =  fbdo.stringData().toInt();

          if (val2 == 1)
            {
              digitalWrite(Relay2, LOW);
              Serial.println("Fan ON");
            }
          else if (val2 == 0)
            {
              digitalWrite(Relay2, HIGH);
              Serial.println("Fan OFF");
            }

          Firebase.getInt(fbdo,"S3");
          val3 =  fbdo.stringData().toInt();

          if (val3 == 1)
            {
              digitalWrite(Relay3, LOW);
              Serial.println("TV ON");
            }
          else if (val3 == 0)
            {
              digitalWrite(Relay3, HIGH);
              Serial.println("TV OFF");
            }

          Firebase.getInt(fbdo,"S4");
          val4 =  fbdo.stringData().toInt();

          if (val4 == 1)
           {
              digitalWrite(Relay4, LOW);
              Serial.println("AC ON");
           }
          else if (val4 == 0)
           {
              digitalWrite(Relay4, HIGH);
              Serial.println("AC OFF");
           }

      }
```

mobizt, 2021-06-22:
It's not necessary to post your code again please check my comment above.

Karandalwani, 2021-06-22:
> Look like you don't know how to use the library properly and you are new to Firebase.
> 
> Calling get continuously in this way in the loop is not efficient and cost your data usage too much and nobody do this in practical usage.
> 
> Do thing like this is similar to you are refreshing the webpage continuously and it's nonsense in making HTTP request all the time.
> 
> If you want to listen to the value changed in the database, use stream instead.
> https://github.com/mobizt/Firebase-ESP32/tree/master/examples/DataChangesListener
> 
> If you want to send data faster, use setAsync function instead.
> 
> The event data stream in REST APIs is as fast or slower a bit as MQTT or web socket technology.
> 
> Spend the time to learn the concept and how to use Firebase which are available anywhere on internet.
> 
> Try other Firebase clients e.g, web using Javascript or Android to learn the basic of server event listening concept.
> 
> Try [all examples](https://github.com/mobizt/Firebase-ESP32/tree/master/examples) provided in this library.
> 
> Read the document about REST API implemented in this library.
> https://firebase.google.com/docs/reference/rest/database

Thank You I will definitely work on my concepts. Thanks for your help 😊