mobizt/Firebase-ESP32 #173: BUG

floatAsNeeded, 2021-06-30:
Hi! 
**Describe the bug**
I just found myself in this problem recently when updating to a versione of the library from 3.10.0 or newer. It basically reset the board with the following error:

```
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
19:46:57.839 -> Core 1 register dump:
19:46:57.839 -> PC      : 0x4000c2e4  PS      : 0x00060e30  A0      : 0x8018f5dc  A1      : 0x3ffb1e30  
19:46:57.839 -> A2      : 0x00000000  A3      : 0x3ffccc08  A4      : 0x00000024  A5      : 0x00000000  
19:46:57.839 -> A6      : 0x74616577  A7      : 0x2d726568  A8      : 0x00000000  A9      : 0x3ffb1e10  
19:46:57.839 -> A10     : 0x3ffccb1c  A11     : 0x400874d4  A12     : 0x00000000  A13     : 0x3f423a6f  
19:46:57.839 -> A14     : 0x00000024  A15     : 0x3ffccc08  SAR     : 0x0000000a  EXCCAUSE: 0x0000001d  
19:46:57.839 -> EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000001  
19:46:57.839 -> 
19:46:57.839 -> ELF file SHA256: 0000000000000000
19:46:57.839 -> 
19:46:57.839 -> Backtrace: 0x4000c2e4:0x3ffb1e30 0x4018f5d9:0x3ffb1e40 0x4018f668:0x3ffb1e60 0x4018f707:0x3ffb1e80 0x400d5602:0x3ffb1eb0 0x400d576d:0x3ffb1f50 0x400d2381:0x3ffb1f70 0x400f767a:0x3ffb1fb0 0x4008a652:0x3ffb1fd0
19:46:57.839 -> 
19:46:57.839 -> Rebooting...

```
**To Reproduce**
This is my code I'm using

`///////////////////////////////LIBRARY DECLARATION////////////////////////////////////////
#include <WiFi.h>
#include <FirebaseESP32.h>
#include "ClosedCube_SHT31D.h"
#include "Adafruit_VEML6075.h"
#include "DFRobot_SHT20.h"
#include "LC709203F.h"
#include "Max44009.h"
#include <WiFiManager.h>         // https://github.com/tzapu/WiFiManager
#include <Wire.h>
WiFiManager wifiManager;

//--------------------------------------------------------------------------------------//

///BATTERY OBJECT//
LC709203F gg;  // create a gas gauge object.

//DECLARATION SENSORS//
ClosedCube_SHT31D sht3xd;
DFRobot_SHT20    sht20;

//LIGHT SENSOR DEFINE//
Adafruit_VEML6075 uv = Adafruit_VEML6075();
Max44009 myLux(0x4A);

//I2C COMUNICATION ADDRESS//
const int I2CSlaveAddress = 8;      // I2C Address.

////////////////////*********FIREBASE DETAILS************///////////////////////////////////
#define FIREBASE_HOST ""                 // the project name address from firebase id
#define FIREBASE_AUTH ""            // the secret key generated from firebase
FirebaseData Weather;

//WIND DIRECTION//
#define WindDirectionDavis 32 // anemometer
#define WindDirectionAnemometer 33 // anemometer

////////////RAIN////////////////
float mmGoccia; // tipping bucket count in mm
float mmPioggia = 0.0; // daily rain
RTC_DATA_ATTR float mmPioggia24H = 0;
RTC_DATA_ATTR float rainrate = 0; // real-time rainrate
RTC_DATA_ATTR float rainrateMax = 0; // daily rainrateMax
byte rainratedetect = 0;
int rainrateMaxIntensity = 0;
byte PluvioFlag = 0; // detect interrupt of rain
[... 341 lines cut ...]
{
  if (Firebase.getInt(Weather, "Time/CurrentDay"))
  {
    CurrentDay = Weather.intData();
  }

  if (CurrentDay != PreviousDay)
  {
    PreviousDay = CurrentDay;
    mmPioggia24H = 0;
    rainrateMax = 0;
    Firebase.setIntAsync(Weather, "Time/PreviousDay",  PreviousDay);
    Firebase.setFloatAsync(Weather, "Rain/Rain24H", mmPioggia24H);
    Firebase.setFloatAsync(Weather, "Rain/RainRateMax", rainrateMax);
  }
}`

Deleted firebase secret for privacy.

**IDE and its version:**
 - Arduino, PlatformIO
 - Version 1.8.13

**ESP32 Arduino Core SDK version**
 - Version 1.0.6

mobizt, 2021-06-30:
Your code can't reproduce.

You should remove all third party libraries from the code.

Share your clean, clear and minimal code instead of integration code.

Otherwise you should debug your code first and address where in the code that causes the error and what's your expected result and what's the error you get instead of crash that possibilities of crash are unclear and not related directly to the library.

mobizt, 2021-06-30:
You need to update the library to the latest version, post your clean code without third party libs which everyone can test and reproduce the error.

floatAsNeeded, 2021-06-30:
Hi

Thanks for the respond! I tried the latest version already, the issue came after the 3.10.0 as stated, I didn't try to debug because everything was working always, but just after updating the issue came up, that's why I thought it was related to the library, but it could be a mix of things. I will do the test without any library and let you know!

mobizt, 2021-06-30:
Now I'm testing something that possible causes the crash with legacy initialize function (begin using Firebase host and database secret).

I will  inform you soon.

mobizt, 2021-06-30:
Now the update, [v3.10.4](https://github.com/mobizt/Firebase-ESP32/releases/tag/v3.10.4) is available with this bug fixed.

Please update the library.

closed 2021-06-30 by mobizt

floatAsNeeded, 2021-07-01:
Thank you very much! It is now working well! The only thing I noticed is that sometimes if I use the setAsync it doens't actually update the data. Why you think that is happening? Is it okay to use on float values?

mobizt, 2021-07-01:
The async in this library used is for store only and is not actually async request unless the response was ignored (no wait) and continue to send other requests which Firebase supports concurrent request like this.

The session reused and closed by server or by library (every 3 minutes).  Sometimes server closed the connection by itself, and due to no response was check then your database was not changed.

mobizt, 2021-07-01:
You can check the connection by calling httpConnected() from Firebase Data object to determine the request of async function was sent or not.

floatAsNeeded, 2021-07-06:
Sorry for late reply! Thank you very much for the info