mobizt/Firebase-ESP32 #174: Connection Refused!

Amey112, 2021-07-04:
I am simultaneously (streaming using setMultiPathStreamCallback()) and checking for WiFi connectivity using keepWiFiAlive(); 
When I try to use Firebase.setInt() function, the firebase Data Object returns error "Connection Refused".

if I remove the keepWiFiAlive() function; Firebase.setInt() function works alright. 
Please can You help me find the error?

The keepWiFiAlive function is :
`void keepWiFiAlive(void *parameter) {
  for (;;) {
    //    if (WiFi.status() == WL_CONNECTED && Ping.ping("www.google.com", 1)) {
    if (WiFi.status() == WL_CONNECTED) {
      Serial.println("Internet is Active!");
      configTime(0, 0, ntpServer);
      Serial.println("Syncing Time...");
      //      prevSyncMillis = millis();
      if (modeVal == false) {
        //        fbdo1.pauseFirebase(false);
        delay(200);
        Firebase.readStream(fbdo1);
        portENTER_CRITICAL_ISR(&modeMux);
        modeVal = true;
        portEXIT_CRITICAL_ISR(&modeMux);
      }
      vTaskDelay(10000 / portTICK_PERIOD_MS);
      continue;
    }
    //    fbdo1.pauseFirebase(true);
    portENTER_CRITICAL_ISR(&modeMux);
    modeVal = false;
    portEXIT_CRITICAL_ISR(&modeMux);
    portENTER_CRITICAL_ISR(&initMux);
    shouldInit = true;
    portEXIT_CRITICAL_ISR(&initMux);
    Serial.println("Trying to connect to WiFi..");
    WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
    unsigned long startAttemptTime = millis();
    while (WiFi.status() != WL_CONNECTED &&
           millis() - startAttemptTime < WIFI_TIMEOUT_MS) {
      Serial.print(".");
      delay(1000);
    }
    vTaskDelay(WIFI_RECOVER_TIME_MS);
  }
}
`
And 
`  xTaskCreatePinnedToCore(
    keepWiFiAlive,
    "keepWiFiAlive", // Task name
    50000,            // Stack size (bytes)
    NULL,            // Parameter
    1,               // Task priority
    NULL,            // Task handle
    tskNO_AFFINITY);

}`
xTaskCreatePinnedToCore() function is used in the setup() for it.

using **Firebase.setInt(fbdo2, parentPath + switch2.FirebaseId, (int)switch2.currState)** is hit or miss;
sometimes it works and sometimes it doesn't.

Thank You for your time.

Amey112, 2021-07-04:
Hey, I found a workAround!
I always delete the keepWiFiAlive() task and then create it again whenever I use setInt function; 
Though I would like to know what causes this problem, if You know please tell me.

mobizt, 2021-07-04:
This library has its own WiFi reconnection routine. This can be enabled via Firebase.reconnectWiFi(true/* or false "/);

As in all examples of library, this option is enable and WiFi will be resumed automatically.

Every http request, the WiFi status will check and  WiFi.reconnect() will be called internally. The operations behind this function are disconnect WiFi and reconnect again  with the settings that saved in flash memory.

The WiFi reconnection will be done once for each http request in all CRUD and periodically every 10 seconds in stream connection if WiFi status was not connected.

This algorithm used in this library works perfectly and effectively for ESP8266 and ESP32 without user code or other WiFi reconnection library requirement.

In case where WiFi reconnection process was taken control by user code or other WiFi management library e.g. all WiFiManager libraries, the option Firebase.reconnectWiFi should set false to prevent the interferences between these.

closed 2021-07-04 by mobizt

mobizt, 2021-07-05:
> PS.: Using a getJSON() or any other get function works but using set functions is not working

There is no different between calling these functions, as the reason I explained above.

yukimach, 2021-07-17:
Code was working perfectly until 6th of july. It was running for around a year and no issues and then suddenly getting error of connection refused.

Checked the rules: true , to test. 
I'm still using database secrets fo FIREBASE_AUTH . Could that be the issue that its too old a way to connect?