mobizt/Firebase-ESP32 #175: How much heap is required to push to Firebase?

miag676, 2021-07-09:
Hello,

first of all thank you for making and maintaining this library. I am using it on my ESP32 but am having some issues when using BLE to scan for other devices. First I connect to the internet and scan using BLE, then I disable BLE using the following command: esp_bt_controller_disable() (as I have seen WiFi and BLE can't work together). However,  when trying to push an integer to Firebase I get the following error: "connection refused". Before pushing to Firebase my free heap is around 30 000. Is that not enough to establish a connection? How much free heap is needed to push to Firebase?

Here is also my code:
`void setup()
{ 
    Serial.begin(115200);
    WiFiManager wm;
wm.setTimeout(60);
 if (!wm.autoConnect("AutoConnectAP", "password")) {
 Serial.println("failed to connect and hit timeout");
  }
Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
if (!Firebase.beginStream(firebaseData1, path))
 {
   Serial.print("------------------------------------");
   Serial.print("Can't begin stream connection...");
   Serial.print("REASON: " + firebaseData1.errorReason());
   Serial.print("------------------------------------");
  }
    Serial.println("Starting Arduino BLE Client application...");
  BLEDevice::init("ESP32-BLE-Client");
  

  /* Retrieve a Scanner and set the callback we want to use to be informed when we
     have detected a new device.  Specify that we want active scanning and start the
     scan to run for 5 seconds. */
  BLEScan* pBLEScan = BLEDevice::getScan();
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setInterval(1349);
  pBLEScan->setWindow(449);
  pBLEScan->setActiveScan(true);
  pBLEScan->start(5, false);
     esp_bt_controller_disable();

}

void loop()
{
  if (millis() - prevMillis > 10000){
    prevMillis = millis();
     delay(5);
int stev = 3;
firebaseData2.pauseFirebase(false);
  Serial.println(ESP.getFreeHeap());
   if (Firebase.push(firebaseData2, "Devices/BLE", stev))
    {
                  Serial.println("PASSED");
            Serial.println("PATH: " + firebaseData2.dataPath());
            Serial.println("TYPE: " + firebaseData2.dataType());
            Serial.print("VALUE: ");
            printResult(firebaseData2);
            Serial.println("------------------------------------");
            Serial.println();
    }
    else
    {
                  Serial.println("FAILED");
            Serial.println("REASON: " + firebaseData2.errorReason());
            Serial.println("------------------------------------");
            Serial.println();
    }
    firebaseData2.clear();
  }
  }`

Thanks in advance!

Kind regrads, 
Mia

mobizt, 2021-07-09:
The connection refuse means your device can't access the internet (can't get ip from host).

Make sure your database url or host is valid.

You need to test by ping or create https request to some server for internet connection checking. The examples for test this are available in Arduino IDE examples.

This usage problem should ask in [discussions](https://github.com/mobizt/Firebase-ESP32/discussions) instead of issue.

mobizt, 2021-07-09:
The memory may be a problem and ESP32 required > 80k for making https request.

mobizt, 2021-07-09:
Did you try [NimBLE](https://github.com/h2zero/NimBLE-Arduino) fork for ESP32 which use less memory.

miag676, 2021-07-09:
Thank you for your help! I will remeber to ask in the discussions next time. Curretnly I am using the BLEdevice library, but will look at NimBLE. I think my database url and host are valid, since it successfuly pushes to Firebase when I'm not using BLE.

closed 2021-07-09 by mobizt