mobizt/Firebase-ESP32 #176: HELP - streamCallback

vanderareosa, 2021-07-09:
When I update some value directly in the firebase console, the streamCallBack only shows the path and the value of the field that has changed. But when I make the change through my app the stream shows all the fields of my object, even the fields that haven't changed. How can I solve this? I only need to know the field that has been changed to take some action.

mobizt, 2021-07-09:
Fix your app on how it makes change to  database. 

From your explanation, your app changes whole data at the stream path instead of change to some child node.

The stream event data is the changes under your stream path on the database.

closed 2021-07-09 by mobizt

vanderareosa, 2021-07-09:
correct! Thanks so much. I did the changes and its all works fine.