mobizt/Firebase-ESP32 #177: BUG

pei-dev, 2021-07-14:
I have a nodemcu ESP32-S, which I connect to RealTime Database to send and read data. Everything works correctly, but it always 'breaks' when it reaches 100 read or write requests. Each one follows its own independent counter, and of course, when it restarts, it takes about 10s to re-establish the connection and send data again. I used the example of the library and it does exactly the same, so it must be a bug in the library that I hope I can fix as soon as possible, because it is urgent.

**The simplest code to reproduce my error**

`#include <FirebaseESP32.h>
#include <FirebaseFS.h>
#include <WiFi.h>

#define WIFI_SSID ""
#define WIFI_PASSWORD ""

#define FIREBASE_HOST ""
#define FIREBASE_AUTH ""

FirebaseData fbdo;

String path = "/test";

int tiempoConsumidoSesion = 0;
int timeouts = 0; // Error counter

void setup()
{

  Serial.begin(115200);

  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);

  tiempoConsumidoSesion = 0;

  //Size and its write timeout e.g. tiny (1s), small (10s), medium (30s) and large (60s).
  Firebase.setReadTimeout(fbdo, 1000 * 60);
  Firebase.setwriteSizeLimit(fbdo, "tiny");
}

void loop()
{
  Serial.println(timeouts);
  tiempoConsumidoSesion = tiempoConsumidoSesion + 1;
  if (Firebase.setString(fbdo, path, String(tiempoConsumidoSesion))) 
  {
    Serial.println("PASSED");
    Serial.println("PATH: " + fbdo.dataPath());
    Serial.println(tiempoConsumidoSesion);
    Serial.println("------------------------------------");
    Serial.println();

  } else {

    Serial.println("FAILED");
    Serial.println("REASON: " + fbdo.errorReason());
    timeouts = timeouts + 1;
    Serial.println("------------------------------------");
    Serial.println();
  }
  delay(1000);
}`

**Expected behavior**
When it adds up to 100 iterations, the sending stops for a maximum of 10s and gives the following error cause:
FAILED
REASON: read timed out

(in case it is useful, I have seen that it corresponds to the following line of code of the lib: `static const char fb_esp_pgm_str_69[] PROGMEM = "read timed out`")

**IDE and its version:**
 - Arduino
 - Version 1.8.11

**ESP32 Arduino Core SDK version**
 - Version 1.0.4
 
**Additional context**
I have tried with another board, with another cable, with more or less complex codes, with the library for the ESP8266 and ESP32, and I get the same bug in all of them.

mobizt, 2021-07-14:
You should update the library and ESP32 SDK first.

closed 2021-07-14 by mobizt

mobizt, 2021-07-14:
As I mention you from last post, you should debug it first and don't conclude anything as I test all intensively 24/7 to identify all possible issue.

mobizt, 2021-07-14:
If you test with ESP32 and ESP8266 and get the same issue, your network is most the possible issue.

pei-dev, 2021-07-14:
I had ESP32 SDK version 1.0.4 because it was the right one to be able to seamlessly integrate Wifi and BLE on the same board. I have updated it to the latest version and the bug has disappeared. I'll keep trying with more complex codes, but so far it seems to be solved. Thank you very much for your help.

mobizt, 2021-07-14:
ESP32 Core v1.0.6 WiFIClientSecure is more reliable and stable than 1.0.4 with better lwip handle as I ever found the issue as it hangs in long time open socket connection for file download and upload issue.

pei-dev, 2021-07-14:
Yes, I have noticed. Well, after trying with more complex codes everything works correctly, so problem solved. Again, thanks for the quick response and efficient help.

parthesh-hub, 2022-02-05:
> You should update the library and ESP32 SDK first.

I am also facing the same issue.. I tried updating the libraries and sdk.
I am using esp32s with firebase realtime database.
Versions are as follows:
1) Arduino version : 1.8.19
2) ESP32 board manager : 1.0.6 (latest available) 
3) SDK  version : sdk esp32 : v3.3.5-1-g85c43024c  (Also tried with 4.4, installed from github)

Can you please help me with this :)

mobizt, 2022-02-05:
@[parthesh-hub](https://github.com/parthesh-hub) 

You can post in [Discussions](https://github.com/mobizt/Firebase-ESP32/discussions) with the code and debug message and describe your problem.