mobizt/Firebase-ESP32 #178: multipath stream + set data

ayyysh04, 2021-07-14:
I am doing a project of a switch and led where i have to update the on /off data sent from arduino to firebase and firebase to arduino.
I am using multipath stream and setString to receive and update data.
Receiving is fine but when i update data using setstring the data gets updated in firebase but multipath stream detects it again and update the data again .This make the on and off commands slow.

Is there any function which allow us to set/update data without getting deteced by multipath stream

mobizt, 2021-07-14:
You should avoid the recursive like operation.

It's not library issue unless your data flow problem.

Don't use any stream update data to send something  to update the node under the parent path.

closed 2021-07-14 by mobizt

ayyysh04, 2021-07-14:
#include <ESP8266WiFi.h>
#include <FirebaseESP8266.h>

/*Define the WiFi credentials */
#define WIFI_SSID "WIFI_AP"
#define WIFI_PASSWORD "WIFI_PASSWORD"
#define FIREBASE_HOST "prototypexreal-default-rtdb.firebaseio.com/"
#define FIREBASE_AUTH "FoUFEsd9oeh1iQfxswlxJwkCP5ybHEWKTp3i3SyP"

/*Pins
#define LED4 1
#define LED2 2
#define LED3 4
*/
#define ButtonPin 5
#define LED1 16
int bstate=0;
FirebaseData firebaseData1;

unsigned long sendDataPrevMillis = 0;

String parentPath = "/Devices/Room1";
String childPath[1] = {"/Board1"};

size_t childPathSize = 1;
void handlePin(String path,String value)
{
 if(path == "/Board1/Switch1")
  {
    Serial.println("conditon mathced");
  
    if(value=="1")
     {digitalWrite(LED1,HIGH);
     bstate=1;
     }
    else if(value=="0")
    {
     digitalWrite(LED1,LOW);
     bstate=0;
    }
    else
     Serial.println("Wrong input\n");
     
  }
 else
  Serial.println("No condition matched");
  }

void streamCallback(MultiPathStreamData stream)
{
 Serial.println();
 Serial.println("Stream Data1 available");
 
 size_t numChild = sizeof(childPath) / sizeof(childPath[0]);

 for (size_t i = 0; i < numChild; i++)
  {
    if (stream.get(childPath[i]))
    {
      Serial.println("stream ");
      Serial.println(stream.get(childPath[i]));
      
      Serial.println("path: "+ stream.dataPath + ", type: " + stream.type + ",value: " + stream.value);

      //Custom function to make pin high and low
      handlePin(stream.dataPath,stream.value);
    }
  }

  Serial.println();
}

void streamTimeoutCallback(bool timeout)
{
  if (timeout)
    Serial.println("stream timeout, resuming...\n");   
}
void setup()
{
 pinMode(LED1,OUTPUT);
  pinMode(ButtonPin,OUTPUT);
  Serial.begin(115200);

  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  Firebase.begin(FIREBASE_HOST,FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);

  if (!Firebase.beginMultiPathStream(firebaseData1, parentPath,childPath,childPathSize))
   {
    Serial.println("cant begin stream connection\n");
    Serial.println("Reason: "+firebaseData1.errorReason());
    }
  Firebase.setMultiPathStreamCallback(firebaseData1,streamCallback,streamTimeoutCallback);

}
void loop()
{
 if(digitalRead(ButtonPin)==1)
 {
  bstate+=1;
   if(bstate%2 == 0)
     {
      Serial.println("Led is switched Off");
      digitalWrite(LED1,LOW);
      Firebase.setStringAsync(firebaseData1, "firebaseData1","0");  
      bstate=0;
     }
   else
    {
     Serial.println("Led is switched On");
     Firebase.setStringAsync(firebaseData1,"/Devices/Room1/Board1/Switch1","1"); 
     digitalWrite(LED1,HIGH); 
    }
 }  
}

library is excellent love ur work ,i just need some help is my code
the problem is when i update the value from arduino the pathstream also detects it and find which path is detected which delays the off operation after an on operation

mobizt, 2021-07-14:
```cpp
Firebase.setStringAsync(firebaseData1, "firebaseData1","0"); 
```

You use the same Firebase Data object, **firebaseData1** for stream with callback and other CRUD (create/read/update/delete) operations which is not recommended and you will not see usage like this in the examples.

You can use only one Firebase Data object for stream read operation and CRUD, not stream callback like [this](https://github.com/mobizt/Firebase-ESP32/blob/master/examples/DataChangesListener/NoCallback/NoCallback.ino).

Stream connection session need to be keep open after request sent, if you interrupt it by using the same Firebase Data object that holds the TCP client to connect to other request in non-stream mode, the current stream connection will be stop to open new connection of non-stream request.

ayyysh04, 2021-07-14:
thanks it worked