mobizt/Firebase-ESP32 #179: HELP Multipath stream+ settimg data

ayyysh04, 2021-07-14:
#include <ESP8266WiFi.h>
#include <FirebaseESP8266.h>

/*Define the WiFi credentials */
#define WIFI_SSID "WIFI_AP"
#define WIFI_PASSWORD "WIFI_PASSWORD"
#define FIREBASE_HOST "prototypexreal-default-rtdb.firebaseio.com/"
#define FIREBASE_AUTH "FoUFEsd9oeh1iQfxswlxJwkCP5ybHEWKTp3i3SyP"

#define ButtonPin 5
#define LED1 16
int bstate=0;
FirebaseData firebaseData1;

unsigned long sendDataPrevMillis = 0;

String parentPath = "/Devices/Room1";
String childPath[1] = {"/Board1"};

size_t childPathSize = 1;
void handlePin(String path,String value)
{
 if(path == "/Board1/Switch1")
  {
    Serial.println("conditon mathced");
  
    if(value=="1")
     {digitalWrite(LED1,HIGH);
     bstate=1;
     }
    else if(value=="0")
    {
     digitalWrite(LED1,LOW);
     bstate=0;
    }
    else
     Serial.println("Wrong input\n");
     
  }
 else
  Serial.println("No condition matched");
  }

void streamCallback(MultiPathStreamData stream)
{
 Serial.println();
 Serial.println("Stream Data1 available");
 
 size_t numChild = sizeof(childPath) / sizeof(childPath[0]);

 for (size_t i = 0; i < numChild; i++)
  {
    if (stream.get(childPath[i]))
    {
      Serial.println("stream ");
      Serial.println(stream.get(childPath[i]));
      
      Serial.println("path: "+ stream.dataPath + ", type: " + stream.type + ",value: " + stream.value);

      //Custom function to make pin high and low
      handlePin(stream.dataPath,stream.value);
    }
  }

  Serial.println();
}

void streamTimeoutCallback(bool timeout)
{
  if (timeout)
    Serial.println("stream timeout, resuming...\n");   
}
void setup()
{
 pinMode(LED1,OUTPUT);
  pinMode(ButtonPin,OUTPUT);
  Serial.begin(115200);

  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  Firebase.begin(FIREBASE_HOST,FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);

  if (!Firebase.beginMultiPathStream(firebaseData1, parentPath,childPath,childPathSize))
   {
    Serial.println("cant begin stream connection\n");
    Serial.println("Reason: "+firebaseData1.errorReason());
    }
  Firebase.setMultiPathStreamCallback(firebaseData1,streamCallback,streamTimeoutCallback);

}
void loop()
{
 if(digitalRead(ButtonPin)==1)
 {
  bstate+=1;
   if(bstate%2 == 0)
     {
      Serial.println("Led is switched Off");
      digitalWrite(LED1,LOW);
      Firebase.setStringAsync(firebaseData1, "firebaseData1","0");  
      bstate=0;
     }
   else
    {
     Serial.println("Led is switched On");
     Firebase.setStringAsync(firebaseData1,"/Devices/Room1/Board1/Switch1","1"); 
     digitalWrite(LED1,HIGH); 
    }
 }  
}

library is excellent love ur work ,i just need some help is my code
the problem is when i update the value from arduino the pathstream also detects it and find which path is detected which delays the off operation after an ON operation

ayyysh04, 2021-07-14:
Also other path stream gets called when i update one path 
plz help me by correcting above code
[image]

mobizt, 2021-07-14:
Not need to open new same issue, I reply your previous issue already.

mobizt, 2021-07-14:
https://github.com/mobizt/Firebase-ESP32/issues/178#issuecomment-879938771

closed 2021-07-14 by mobizt