mobizt/Firebase-ESP32 #183: delay issue in esp32 with firebase

pranavjasani, 2021-08-22:
hello @mobizt  I reffered your code but it seems a bit confusing...

I am using 8 gpio pins to control leds with firebase. I am facing the delay of nearly @16 seconds. can you help me with this???

thank you

Here's my code:

#include <WiFi.h>                                                // esp32 library
#include <IOXhop_FirebaseESP32.h>                                             // firebase library
#define FIREBASE_HOST "xxxxxxxxxxxxxxxxxxxxxxxxxx"                         // the project name address from firebase id
#define FIREBASE_AUTH "xxxxxxxxxxxxxxxxxxxxxxxxxx"                    // the secret key generated from firebase

#define WIFI_SSID "xxxxxxx"                                         //input your home or public wifi name
#define WIFI_PASSWORD "xxxxxxxxx"                                    //password of wifi ssid
String fireStatus = "";                                              //led status received from firebase
String bedroom_path_switch1="vishal/bedroom/switch1";
String bedroom_path_switch2="vishal/bedroom/switch2";
String bedroom_path_switch3="vishal/bedroom/switch3";
String bedroom_path_switch4="vishal/bedroom/switch4";
String bedroom_path_switch5="vishal/bedroom/switch5";
String bedroom_path_switch6="vishal/bedroom/switch6";
String bedroom_path_switch7="vishal/bedroom/switch7";
String bedroom_path_switch8="vishal/bedroom/switch8";

int switch1=32;
int switch2=33;
int switch3=27;
int switch4=14;
int switch5=12;
int switch6=13;
int switch7=4;
int switch8=0;
String number1;
String number2;
String number3;
String number4;
String number5;
String number6;
String number7;
String number8;

void setup() {
  Serial.begin(115200);
  pinMode(switch1, OUTPUT);
  pinMode(switch2, OUTPUT);
  pinMode(switch3, OUTPUT);
  pinMode(switch4, OUTPUT);
  pinMode(switch5, OUTPUT);
  pinMode(switch6, OUTPUT);
  pinMode(switch7, OUTPUT);
 pinMode(switch8, OUTPUT);

  //connect to wifi
   WiFi.begin(WIFI_SSID, WIFI_PASSWORD);                                      //try to connect with wifi
  Serial.print("Connecting to ");
  Serial.print(WIFI_SSID);
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
  }
 // Serial.println();
  Serial.print("Connected to ");
  Serial.println(WIFI_SSID);
  Serial.println(WiFi.localIP());                                                      //print local IP address

  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);                                       // connect to firebase

}

void loop() {

   // get value  
 
number1=Firebase.getInt(bedroom_path_switch1);
number2=Firebase.getInt(bedroom_path_switch2);
number3=Firebase.getInt(bedroom_path_switch3);
number4=Firebase.getInt(bedroom_path_switch4);
number5=Firebase.getInt(bedroom_path_switch5);
number6=Firebase.getInt(bedroom_path_switch6);
number7=Firebase.getInt(bedroom_path_switch7);
number8=Firebase.getInt(bedroom_path_switch8);

//switch1
if (number1 == "1"){
  digitalWrite(switch1, HIGH);
}
 else{
  digitalWrite(switch1, LOW);
  }

//switch2
if (number2 == "1"){
  digitalWrite(switch2, HIGH);
}
 else{
  digitalWrite(switch2, LOW);
  }

  //switch3
if (number3 == "1"){
  digitalWrite(switch3, HIGH);
}
 else{
  digitalWrite(switch3, LOW);
  }

  //switch4
if (number4 == "1"){
  digitalWrite(switch4, HIGH);
  Serial.println("switch4 on");
}
 else{
  digitalWrite(switch4, LOW);
  Serial.println("offffff");
  }

  //switch5
if (number5 == "1"){
  digitalWrite(switch5, HIGH);
}
 else{
  digitalWrite(switch5, LOW);
  }

//switch6
if (number6 == "1"){
  digitalWrite(switch6, HIGH);
}
 else{
  digitalWrite(switch6, LOW);
  }

//switch7
if (number7 == "1"){
  digitalWrite(switch7, HIGH);
}
 else{
  digitalWrite(switch7, LOW);
  }

  //switch8
if (number8 == "1"){
  digitalWrite(switch8, HIGH);
}
 else{
  digitalWrite(switch8, LOW);
  }

}

mobizt, 2021-08-22:
You're posting in the wrong repository. This is not my library.

mobizt, 2021-08-22:
This issue will be deleted.

closed 2021-08-22 by mobizt