mobizt/Firebase-ESP32 #184: Strange behavior from library to a specific data after new update

floatAsNeeded, 2021-08-24:
Hi!

I just found a weird issue that I was thinking is related to float numbers but it also seems not be related at the same time... 
I try to explain better: 

Basically, I was using the version of the library 3.11.2 and I encountered a problem when doing the get function 

  if (Firebase.getFloat(Weather, "SHT3x/Temperature/Temperature"))
  {
    temp = Weather.floatData() + OffsetTemp;
  }
  else
  {
    Serial.println(Weather.errorReason());
  }

temp is float data and also OffsetTemp. Basically it will get a data completely random from database and not the actual data, and this happens JUST for this raw. I've other raw with others float data, the same like that and this happens just to the first data. It is very strange. Temperature max and min will get without no problems, but not temperature, that will return a random number without error from database (I checked if there are any errors but everything works fine)

Temperature: 
11.15702
TemperatureMax: 
12.31861
TemperatureMin: 
8.64157

Temperature max and min and also other data float will get with no issues! Just one in particular won't get that is the one I posted.
After downgrading to the version 3.11.0 of the library it works again as always. I was wondering if is something related to some changes on how to get float data.

It will be good to follow up on this.

My version of Arduino is 1.8.13
Esp32 1.0.6

mobizt, 2021-08-24:
Can you test with download the zip file of latest version and replace the version that installed on the Arduino Library Manager?

There are some changes in mainstream but not the released version.

mobizt, 2021-08-24:
I did not found any issue related to your post as I test from the example.

You should create the simple example code which clean and complete that can reproduce the issue and I will check from that.

floatAsNeeded, 2021-08-24:
I will check now downloading the zip version. I let you know! Thank you

mobizt, 2021-08-24:
Since v3.11.0, there is some bugs in data type checking.

In v3.11.1 and latest, this bugs were fixed.

With the latest version, data type mismatch checking is disabled by default, if you intend to get some data with **getBool**, **getInt**, **getFloat**, **getDouble** functions which the response payload data type does not match to the type you request,, and you call **intData()**, **floatData()** or **doubleData** from FirebaseData object, the value that returns will be the original value of previous (old) payload data prior to the current request.

This is because the internal payload was unchanged as you try to get the wrong type from data.

To enable data type checking, you need to set this config.

`config.rtdb.data_type_stricted = true;`

floatAsNeeded, 2021-08-24:
It could be there the problem maybe, but that's strange happens just with that specific raw, the other raw is basically the same, and it doesn't have the error. Maybe because as you said is maybe some old data. Where I can set that line, in setup? Is that line also make slower the receiving of data?

mobizt, 2021-08-24:
You can set/unset any place in your code, may be before you call getXXX and can be disable after that. Nothing is different in speed  due to small peace of type checking code from the currently downloaded response payload.

I recommend to use generic get function and check the data type before use. The latest version, I introduce **to<type>()** which you can cast to any value type instead of fix type e.g. intData or floatData.

You can check the new examples.

Data type mismatch checking is applied to all my Firebase libraries.

floatAsNeeded, 2021-08-24:
Thanks for the help! I just tried with the new temp = Weather.to<float>() + OffsetTemp; but when I added the line it says that "config was not declared in this scope"

config.rtdb.data_type_stricted = true;

floatAsNeeded, 2021-08-24:
I'm using like this, but still same issue. mmm 
Do I wrong something?
```
  config.rtdb.data_type_stricted = true;
  if (Firebase.getFloat(Weather, "SHT3x/Temperature/Temperature"))
  {
    temp = Weather.to<float>() + OffsetTemp;
  }
  else
  {
    Serial.println(Weather.errorReason());
  }
```

floatAsNeeded, 2021-08-24:
If I use this simple code works. That's odd

```
#include <FirebaseESP32.h>
#include <WiFi.h>
////////////////////*********FIREBASE DETAILS************///////////////////////////////////
#define FIREBASE_HOST ""                 // the project name address from firebase id
#define FIREBASE_AUTH ""            // the secret key generated from firebase
FirebaseData Weather;
FirebaseConfig config;
float temp;
float OffsetTemp;

void setup() {
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.begin();
  while (WiFi.status() != WL_CONNECTED) {
    delay(250);
    //Serial.print(".");
  }
  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);   // connect to firebase
  //Serial.println(Weather.errorReason());
  Firebase.reconnectWiFi(true);
  Firebase.setMaxRetry(Weather, 2);
  config.rtdb.data_type_stricted = true;
  if (Firebase.getFloat(Weather, "SHT3x/Temperature/Temperature"))
  {
    temp = Weather.to<float>() + OffsetTemp;
  }
  else
  {
    Serial.println(Weather.errorReason());
  }
}

void loop() {
  Serial.println(temp);
  delay(2000);
}
```

floatAsNeeded, 2021-08-24:
I found the issue. Eventually was my fault. Seems the issue is related not on that line of the Firebase, but instead on this line 

```
  if (Firebase.getFloat(Weather, "SHT3x/Offset"))
  {
    OffsetTemp = Weather.to<float>();
  }
```

Basically, I was getting that data in wrong spot of the database and it returned me 62 that then plus to the temperature it gave a weird reading. That raw doesn't exist anymore.. I don't know how in the previous version it could detect, or maybe before it just was ignoring that. 

Anyway now I fixed that raw with the correct get from the database and now is all working good

closed 2021-08-24 by floatAsNeeded