mobizt/Firebase-ESP32 #187: Can't send json to Firebase (please help)

juliusfelix12, 2021-08-30:
Hi sir, why i can't send JSON string to firebase, which part i do wrong?? I used to read the documentation and still not find the solution, please help thankyou.
[image]
[image]

mobizt, 2021-08-30:
You should read the document.
https://github.com/mobizt/Firebase-ESP32/tree/master/src#set-child-nodes-key-and-value-using-firebasejson-object-to-the-defined-database-path

You can't set the serialized JSON string to that function which accept only FirebaseJson object.

The library has built-in JSON builder and parser, [FirebaseJson](https://github.com/mobizt/Firebase-ESP32/tree/master/src#firebasejson-object-functions) to serialize and deserialize JSON which is easier and efficient.

mobizt, 2021-08-30:
The examples related to FirebaseJson are [here](https://github.com/mobizt/Firebase-ESP32/tree/master/examples/FirebaseJson/examples).

mobizt, 2021-08-30:
You should update the library to latest version to use the new core of FirebaseJson which is powered by the best JSON parser, [cJSON](https://github.com/DaveGamble/cJSON).

closed 2021-08-30 by mobizt

mobizt, 2021-08-30:
This is the [example](https://github.com/mobizt/Firebase-ESP32/blob/master/examples/FirebaseJson/examples/Client/Firebase/Firebase.ino) when using FirebaseJson and Firebase RTDB.

juliusfelix12, 2021-08-30:
thankyou sir for helping!

juliusfelix12, 2021-08-30:
sir @mobizt , i already read and apply code from the documentation like this and it works!
[image]

but why it doesn't generate random key like this as a child for every read data??
[image]

pls help thanks

mobizt, 2021-08-30:
You can use pushJSON to push the new random key name to that node.

juliusfelix12, 2021-08-30:
thanks a lot sir!!!