mobizt/Firebase-ESP32 #188: How to push a value with a timestamp in the same key?

0x776172, 2021-08-31:
Hi sir, i want to add timestamp in my value. look like this:
[image]

json1.add("Value", valLight);
json1.add("TImestamp", Firebase.pushTimestamp(fbdo, ""));
Firebase.pushJSON(fbdo, pathJSON, json1); 

i have tried this code but it doesnt work properly, and the return of the Firebase.pushTimestamp is false with the error of bad request.. please help me with the issues.

mobizt, 2021-08-31:
You should read the document for function descriptions.

All functions returned boolean for the operrating status instead of value.

For the pushTimestamp function, there is a bug that returns bad request which was already fixed, please update the library to latest version, [v3.11.3](https://github.com/mobizt/Firebase-ESP32/releases/tag/v3.11.3).

With the latest version library, the timestamp example was updated which you can follow how to set and get the timestamp correctly.
https://github.com/mobizt/Firebase-ESP32/blob/master/examples/Timestamp/Timestamp.ino

closed 2021-08-31 by mobizt

0x776172, 2021-08-31:
Thank you very much, i will try the code soon!

0x776172, 2021-08-31:
> You should read the document for function descriptions.
> 
> All functions returned boolean for the operating status instead of value.
> 
> For the pushTimestamp function, there is a bug that returns bad request which was already fixed, please update the library to latest version, [v3.11.3](https://github.com/mobizt/Firebase-ESP32/releases/tag/v3.11.3).
> 
> With the latest version library, the timestamp example was updated which you can follow how to set and get the timestamp correctly.
> https://github.com/mobizt/Firebase-ESP32/blob/master/examples/Timestamp/Timestamp.ino

okay i will try the code, Thank you very much!!