mobizt/Firebase-ESP32 #191: BeginMultiPathStream

Amey112, 2021-09-26:
**'mBeginMultiPathStream' was not declared in this scope**
{
	"resource": "/platformIO/SmartBoard-Arduino/.pio/libdeps/esp32dev/Firebase ESP32 Client/src/Firebase.h",
	"owner": "cpp",
	"severity": 8,
	"message": "'mBeginMultiPathStream' was not declared in this scope",
	"startLineNumber": 1868,
	"startColumn": 131,
	"endLineNumber": 1868,
	"endColumn": 131
}

**Expected behavior**
The stream is supposed to be started, rather a compilation message occurs. 

**IDE and its version:**
 - PlatformIO
 - Version Core 5.2.0

**Firebase ESP32 Client**
 - Version 3.11.3

**Additional context**
There is also a bug in FirebaseConfig config, which prevents me from using cofig.database_url and config.signer.tokens.legacy_token

Amey112, 2021-09-26:
**The line of code which produces the error**

  if (!Firebase.beginMultiPathStream(fbdo1, parentPath, childPath, childPathSize))
  {
    Serial.println("------------------------------------");
    Serial.println("Can't begin stream connection...");
    Serial.println("REASON: " + fbdo1.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }

mobizt, 2021-09-26:
I don't know how you install the library but it's not correct.

You should use Library Manager in PIO Home -> Libraries and search for the library then click the blue button "Add to Project" and select the project to add.

You can also install via Library Manager CLI.
https://docs.platformio.org/en/latest/core/userguide/lib/cmd_install.html
https://docs.platformio.org/en/latest/core/userguide/lib/cmd_uninstall.html

closed 2021-09-26 by mobizt

Amey112, 2021-09-26:
I did the exact same thing.

mobizt, 2021-09-26:
Uninstall first.

mobizt, 2021-09-26:
This is the Github action that used for compile, build and run the examples using PlatformIO on Linux.
https://github.com/mobizt/Firebase-ESP32/runs/3650252027?check_suite_focus=true

Amey112, 2021-09-26:
Ok, I don't know if I being a real idiot here, but the build link is for a version released 6 days ago, and the latest version on platformio is 3.11.3 i.e. one which was released 26 days ago. 
Am I right here?

mobizt, 2021-09-26:
I have found the possible error that when you use the old API of older version library which accepts 4 arguments.

The version you're trying to install has the breaking changes in APIs since version 3.11.0.

Please refer to the new function usage which accepts 2 aruments.
https://github.com/mobizt/Firebase-ESP32/blob/58410a61afab2dfd89aea45cad6b963cb5988c62/examples/DataChangesListener/MultiPath/MultiPath.ino#L134

mobizt, 2021-09-26:
> Ok, I don't know if I being a real idiot here, but the build link is for a version released 6 days ago, and the latest version on platformio is 3.11.3 i.e. one which was released 26 days ago.
> Am I right here?

No matter the version is, the breaking changes since 3.11.0.

Amey112, 2021-09-26:
Thanks, will do it, also can you check if one can download version 3.11.6 from platformio since I am only able to download version 3.11.3 and below, even though the release notes mention 3.11.6

mobizt, 2021-09-26:
There's an issue on PlatformIO Library Manager system since this library was added to PIO registry by the PlatformIO bot.

When I add library.json file to the repos since v3.11.4, that causes no updates in Library Manager and stuck at v3.11.3.

I will remove this library.json file from repos to fix the PlatformIO update issue.

mobizt, 2021-09-26:
I will update the library soon (in an hour) to fix registry issue.
The PIO Library Manager may update in 24 hours later.

mobizt, 2021-09-26:
The library updated to v3.11.7.

To install it now, use

```
pio lib -g install  https://github.com/mobizt/Firebase-ESP32.git
```