mobizt/Firebase-ESP32 #192: MultiPathStream and DHT22

fubl84, 2021-09-26:
Thank you for this whole library! It makes communication with the firebase really easy. 
I'm using your library to sync data between a firebase realtime database and an esp32. 
And for some reason the .setMutliPathStreamCallback function conflicts with the DHT22 (DHT sensor library@^1.4.2 & Adafruit Unified Sensor@^1.1.4).
With the StreamCallback in setup I'm getting nan-values from the DHT and if I comment the function out the readings are just fine. 
To further test this, I set the first dht reading just before the .setMultiPathStreamCallback and it put out normal values - and after the callback-function I get the nan-messages from the dht.

Do you have any idea, why that might be the case? Is there something conflicting with the libraries or is it just stupidity on my part?

Here's the code:

`
/* ****************************************************************************/
/* ******************************** INCLUDES **********************************/
/* ****************************************************************************/

#include <Arduino.h>
#include <time.h>
#include "time.h"
#include <DHT.h>             // Sensor DHT22
#include <WiFi.h>            // WiFi
#include <Firebase.h> // Google Firebase
#include "addons/RTDBHelper.h"
#include "addons/TokenHelper.h"

/* ****************************************************************************/
/* *************************** DEFINES AND MACROS *****************************/
/* ****************************************************************************/

/* 1. Define the WiFi credentials */
#define WIFI_SSID ""
#define WIFI_PASSWORD ""

/* 2. Define the API Key */
#define API_KEY ""

/* 3. Define the RTDB URL */
#define DATABASE_URL "" 

/* 4. Define the user Email and password that alreadey registerd or added in your project */
#define USER_EMAIL ""
#define USER_PASSWORD ""

//Define I/O pins
#define PIN_DHT1    25
#define PIN_DHT2    26
#define RELAY_HUMIDIFIER    18
#define RELAY_VENTILATOR    5
#define RELAY_HEATMAT       17
#define RELAY_REMOTE        4
#define RELAY_LIGHTS        19

//Initialize DHT
#define DHTTYPE DHT22
DHT dht1(PIN_DHT1, DHTTYPE);
DHT dht2(PIN_DHT2, DHTTYPE);

//define states
bool statusStandby = false;

//define limits for humditiy check -> can be changed in app or via touch
int hLowLimit = 65;
int offsetHumidity = 8;
int h = 0;
int t = 0;
int remoteStartH;
int remoteStartM;
int remoteStopH;
int remoteStopM;
int lightsStartH;
int lightsStartM;
[... 348 lines cut ...]
                    remoteMan = true;
                } else {
                    remoteMan = false;
                }
                break;
                case 18: //manual Lights mode
                if (isOn == "true") {
                    lightsMan = true;
                } else {
                    lightsMan = false;
                }
                break;
                default:
                break;
            }
        } else{}
    }
}

void streamTimeoutCallback(bool timeout)
{
  if (timeout)
    Serial.println("stream timeout, resuming...\n");
}
`

mobizt, 2021-09-26:
It's not a library issue.  

There's nothing related between the libraries, no shared memory, static function and variable, everything is defined inside class.

This may be memory issue instead.

When you assign the callback function, the RTOS task for stream will be created with the stack memory reserved for specific amount.

If you call F**irebase.setMultiPathStreamCallback**, without providing the forth argument (**streamTaskStackSize**), the default stack size for stream task will set to 8192 k.

Your device may running low heap memory and some of your code included third party libs or core libs may not work properly or fail to allocate memory. 

You should debug and checking free memory.

mobizt, 2021-09-26:
You should test with clean and simple code between Firebase and sensor libraries, without your class or object that consumed extra memory.

I recommend to use this [stream example](https://github.com/mobizt/Firebase-ESP32/blob/master/examples/DataChangesListener/Callback/Callback.ino) to test with sensor.

The multipath stream and stream operations are the same unless the multipath stream has the function to parse the JSON payload as stated in the example.
https://github.com/mobizt/Firebase-ESP32/blob/e73423a2de65d16bc09f012908e25a1a87d0b804/examples/DataChangesListener/MultiPath/MultiPath.ino#L132

closed 2021-09-26 by mobizt

popeyaho001, 2022-12-18:
@fubl84 have you fix this, i had the same error. can not read from dht22 while using callback

mobizt, 2022-12-18:
> @fubl84 have you fix this, i had the same error. can not read from dht22 while using callback

 If you ask my reply, there is nothing related.

Users should debug this themselves.

Stream function with callback creates the RTOS task on CPU core 1 with 8192 bytes reserved stack memory. Inside the task is infinite loop calling readStream with 3 ms delay, that's it.

You should debug your code for memory and timing and should use the sensor library properly. Or change the sensor library that dosn't use delay inside or something that waits for onewire, i2c interfaces.

The good sensor library must not use delay or blocking code to wait onewire or i2c bus to be ready which is the blocking process that affects other code to run properly.