mobizt/Firebase-ESP32 #199: Firebase connection timeout

manoj-kumar-MS, 2021-12-06:
Hi,
I am trying to read data from firebase and i want to set a read time out of 1 minute.
I used the command `Firebase.setReadTimeout(fbdo, 60000);` and the loop function 
```
void loop() {
  String path = "/References/connAlive";
  if (Firebase.getString(fbdo, path)) {
    String temp = fbdo.to<String>();
    Serial.println(temp);
  }
  else{
    Serial.println("Conn Failed");
  }
}
```

But I am getting the following output

````
17:00:31.429 -> true
17:00:31.829 -> true
17:00:32.229 -> true
17:00:32.629 -> true
17:00:33.069 -> true
17:00:33.469 -> true
17:00:43.496 -> Conn Failed
17:01:20.098 -> Conn Failed
17:01:57.128 -> Conn Failed
17:02:34.132 -> Conn Failed
17:03:11.135 -> Conn Failed
17:03:48.149 -> Conn Failed
17:04:25.157 -> Conn Failed
17:04:43.688 -> Conn Failed
17:04:43.688 -> Conn Failed
````

The Time out is 37 seconds for 8 times and then the `Serial.println("Conn Failed");` runs continuously
Thanks in Advance

mobizt, 2021-12-06:
This is the server side option to reject the long request instead of server connection which you should read the APIs references [here](https://firebase.google.com/docs/reference/rest/database#section-param-timeout).

Don't use it if you are not sure what it is.

If you want to know the reason of operation error, use **fbdo.errorReason()** instead.

The maximum timeout is 900,000 ms (15 min).

mobizt, 2021-12-06:
You are using the library in the wrong way which you should avoid.

https://github.com/mobizt/Firebase-ESP32/blob/b6c96948a3561d70c9857a014447315d3b0315d7/examples/Basic/Basic.ino#L204-L245

closed 2021-12-06 by mobizt

manoj-kumar-MS, 2021-12-06:
Sorry for any miscommunication. 
I changed a the code as below
````
String path = "/References/connAlive";
  Serial.println("Starting Now");
  if (Firebase.getString(fbdo, path)) {
    String temp = fbdo.to<String>();
    Serial.println(temp);
  }
  else {
    Serial.println( fbdo.errorReason());
  }

  delay(5000);
````
I want to retrive Data from realtime database. The condition I was trying to check was 
When ESP32 is connected to Wifi and if the Wifi has no internet access, I want to run the `else` condition.

I want this to run within 1 minute.

I have also checked by turning of the Wifi, If the Wifi is disconnected, the response is immediate and else condtion runs immediately but if ESP32 is connected to Wifi and that Wifi network is down it takes 10 secs to print

`response payload read timed out due to network issue or too large data size`

 The next time the loop runs the reponse is after 37 seconds

` connection lost`

OUTPUT

`18:21:34.031 -> Starting Now
18:21:34.438 -> true
18:21:39.435 -> Starting Now
18:21:49.441 -> response payload read timed out due to network issue or too large data size
18:21:54.436 -> Starting Now
18:22:31.266 -> connection lost
18:22:36.247 -> Starting Now
18:23:13.252 -> connection lost
18:23:18.295 -> Starting Now
18:23:55.295 -> connection lost
18:24:00.284 -> Starting Now
18:24:00.284 -> connection lost
`

mobizt, 2021-12-06:
Did you check the example or just code with your understanding?
https://github.com/mobizt/Firebase-ESP32/blob/b6c96948a3561d70c9857a014447315d3b0315d7/examples/Basic/Basic.ino#L101-L129

Don't expect the library to resolve your internet or WiFi connection.

It's impossible for device to know what happen to your network status, it is just sending the packet and wait the reply response and if the waiting is timed out, before exit the waiting (block) loop with error.

YOU MUST HANDLE AND CHECK YOUR INTERNET CONNECTION IN YOUR CODE USING PING OR TEST THE SERVER CONNECTION INSTEAD OF ALLOW LIBRARY TO PROVE THE CONNECTION AND IT WILL DECIDE THE STATUS BASED ON TIMEOUT OF THE SSL CLIENT SETTINGS