mobizt/Firebase-ESP32 #202: Can't complie code in ESP32 Node-MCU

PaoMian0806, 2021-12-19:
Hello, I am a student from Taiwan. I am using this library to get the Json file on the network and transfer it to the RTDB on Firebase. But I encountered an error when I compiled the following code:

```

#if defined(ESP32)
#include <WiFi.h>
#include <FirebaseESP32.h>
#elif defined(ESP8266)
#include <ESP8266WiFi.h>
#include <FirebaseESP8266.h>
#endif

#include <WiFiClientSecure.h>

#define FBJS_ENABLE_WIFI_CLIENT_SECURE
#include <FirebaseJson.h>

//Provide the token generation process info.
#include <addons/TokenHelper.h>

//Provide the RTDB payload printing info and other helper functions.
#include <addons/RTDBHelper.h>

/* 1. Define the WiFi credentials */
#define WIFI_SSID "INSERTMYWIFI"
#define WIFI_PASSWORD "INSERTMYPASSWORD"

/* 2. Define the API Key */
#define API_KEY "INSERTMYKEY"

/* 3. Define the RTDB URL */
#define DATABASE_URL "MYDBURL"

/* 4. Define the user Email and password that alreadey registerd or added in your project */
#define USER_EMAIL "MYGMAIL"
#define USER_PASSWORD "MYGPASSWORD"

FirebaseData fbdo;

FirebaseAuth auth;
FirebaseConfig config;

unsigned long sendDataPrevMillis = 0;
unsigned long ms = 0;

unsigned long count = 0;

void setup()
{

    Serial.begin(115200);

    WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
    Serial.print("Connecting to Wi-Fi");
    while (WiFi.status() != WL_CONNECTED)
    {
        Serial.print(".");
        delay(300);
    }
    Serial.println();
    Serial.print("Connected with IP: ");
    Serial.println(WiFi.localIP());
    Serial.println();

    Serial.printf("Firebase Client v%s\n\n", FIREBASE_CLIENT_VERSION);

    config.api_key = API_KEY;

    auth.user.email = USER_EMAIL;
    auth.user.password = USER_PASSWORD;
[... 143 lines cut ...]
libraries\FirebaseJson\extras\print\printf.c.o: In function `vprintf_':
printf.c:(.text.vprintf_+0x0): multiple definition of `vprintf_'
libraries\Firebase_ESP32_Client\json\extras\print\printf.c.o:printf.c:(.text.vprintf_+0x0): first defined here
libraries\FirebaseJson\extras\print\printf.c.o: In function `vsnprintf_':
printf.c:(.text.vsnprintf_+0x0): multiple definition of `vsnprintf_'
libraries\Firebase_ESP32_Client\json\extras\print\printf.c.o:printf.c:(.text.vsnprintf_+0x0): first defined here
libraries\FirebaseJson\extras\print\printf.c.o: In function `fctprintf':
printf.c:(.text.fctprintf+0x0): multiple definition of `fctprintf'
libraries\Firebase_ESP32_Client\json\extras\print\printf.c.o:printf.c:(.text.fctprintf+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
exit status 1

```

However, I was able to test the following examples(Firebase_ESP32_Client) separately and they worked fine:

- FirebaseJson >> Client >> WiFiClientSecure

- FirebaseJson >> Client >> Firebase

- Basic

I apologize if this seems like a stupid question, maybe there is a better way to solve it, but my English reading skills are not very good. I'd appreciate any pointers, and I'm sorry to have wasted your time.

If the grammar seems strange, I used a translator to turn the Chinese into English.

mobizt, 2021-12-19:
Update Firebase library to latest version.

Move this to the top of your sketch
**#define FBJS_ENABLE_WIFI_CLIENT_SECURE**

Don't include the FirebaseJson as it is already built in the Firebase library.

closed 2021-12-19 by mobizt

PaoMian0806, 2021-12-19:
Thank you very much! now it can be compiled. But after connecting to HTTPS and sending a GET request there is another problem:

```
21:54:42.958 -> Connecting to Wi-Fi.........
21:54:45.678 -> Connected with IP: 192.168.0.100
21:54:45.678 -> 
21:54:45.678 -> Firebase Client v3.12.7
21:54:45.678 -> 
21:54:45.678 -> Token info: type = id token, status = on request
21:54:47.916 -> Token info: type = id token, status = ready
21:54:47.916 -> Connecting to server... ok
21:54:50.937 -> Send GET request...
21:54:50.937 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
21:54:50.971 -> Core 1 register dump:
21:54:50.971 -> PC      : 0x400d5dd3  PS      : 0x00060a30  A0      : 0x800d6a36  A1      : 0x3ffb1d40  
21:54:50.971 -> A2      : 0x00000000  A3      : 0x3ffb1d64  A4      : 0x00000010  A5      : 0x00000010  
21:54:50.971 -> A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x800d19d0  A9      : 0x3ffb1d30  
21:54:50.971 -> A10     : 0x00000010  A11     : 0x3f40053b  A12     : 0x00000010  A13     : 0x0000ff00  
21:54:51.005 -> A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
21:54:51.005 -> EXCVADDR: 0x0000000c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
21:54:51.005 -> 
21:54:51.005 -> ELF file SHA256: 0000000000000000
21:54:51.005 -> 
21:54:51.005 -> Backtrace: 0x400d5dd3:0x3ffb1d40 0x400d6a33:0x3ffb1d60 0x400d4db5:0x3ffb1d90 0x400d16f9:0x3ffb1db0 0x400dec04:0x3ffb1fb0 0x40089a7a:0x3ffb1fd0
21:54:51.005 -> 
21:54:51.005 -> Rebooting...
21:54:51.038 -> ets Jun  8 2016 00:22:57
21:54:51.038 -> 
21:54:51.038 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
21:54:51.038 -> configsip: 0, SPIWP:0xee
21:54:51.038 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
21:54:51.038 -> mode:DIO, clock div:1
21:54:51.038 -> load:0x3fff0018,len:4
21:54:51.038 -> load:0x3fff001c,len:1044
21:54:51.038 -> load:0x40078000,len:10124
21:54:51.038 -> load:0x40080400,len:5856
21:54:51.038 -> entry 0x400806a8
```

The rest of the code is the same as the previous one, I used [EspExceptionDecoder](https://github.com/me-no-dev/EspExceptionDecoder) to parse the error and then these messages appeared:

```
0x400d5d7f: get_value_buffer_length at C:\Users\PaoMian\Documents\Arduino\libraries\Firebase_ESP32_Client\src\json\MB_JSON/MB_JSON.c line 120
0x400d69df: MB_JSON_SerializedBufferLength at C:\Users\PaoMian\Documents\Arduino\libraries\Firebase_ESP32_Client\src\json\MB_JSON/MB_JSON.c line 120
0x400d4d61: FirebaseJsonBase::mGetSerializedBufferLength(bool) at C:\Users\PaoMian\Documents\Arduino\libraries\Firebase_ESP32_Client\src\json/FirebaseJson.cpp line 1229
0x400d16c5: FirebaseJson::serializedBufferLength(bool) at c:\users\paomian\documents\arduino\libraries\firebase_esp32_client\src\json/firebasejson.h line 236
:  (inlined by) loop() at C:\Users\PaoMian\Documents\Arduino\Test3/Test3.ino line 125
0x400debb0: loopTask(void*) at C:\Users\PaoMian\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/main.cpp line 23
0x40089a7a: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 355 (discriminator 1)
```

I looked up some information on the web and reread the tutorials for this library, but I may have missed something and couldn't get the GET result to execute successfully.

mobizt, 2021-12-19:
I'm not sure you understand the http GET method request headers usage.

With GET request, the content-length header is not needed and invalid header.

And calling serializedBufferLength from NULL (constructed object but not yet initialized) FirebaseJson is useless.

The FirebaseJson constructor is only placeholder and should be initialized with some value before calling that function.