mobizt/Firebase-ESP32 #203: Invalid token in FCM firebase

razisek, 2021-12-20:
I want to use Firebase Cloud Messaging (FCM). but when entering Server key from cloud messaging in firebase I get response :
```
The ID token or registration token was not not found at index
The ID token or registration token was not not found at index
The ID token or registration token was not not found at index
```

the server key I use is the same as the one in the firebase console
[image]
this my method :
```c++
void sendNotification()
    {
        fbdo.fcm.begin("AAAAJn7rcU0:APA91bECX7dFufaRYc9LzLbSXmIQCFpqmHiJnhg9_h-Mk2W03gaSWPeUmZTxw5235Caqcchm36tyeblsDSFLHyZEfmrA0w9gjTrWtLG6mdprG7tUw8tOUn64ak4W2Emk_Zxxxxxxxxx");
        fbdo.fcm.addDeviceToken("fh3SkcqSTFK7zhmjalPS7f:APA91bGeLJG5gtsNIFU4qsf7uiZchQi9-ZkvQlvJZr_cj6fKr1LGYY8rnoqH7iCRLgg7ZQmKFgCYKM5EmGsiK8c5ovXcLNvCrI9kp3ygoTiYEeOX0TudBkZPw-i83-xxxxxxxxxxxx");
        fbdo.fcm.setPriority("normal");
        fbdo.fcm.setTimeToLive(5000);
        fbdo.fcm.setNotifyMessage("Notification", "Hello World!", "firebase-logo.png", "http://www.google.com");
        fbdo.fcm.setDataMessage("{\"myData\":\"myValue\"}");
        if (Firebase.sendMessage(fbdo, 1))
        {
            //Success, print the result returned from server
            Serial.println(fbdo.fcm.getSendResult());
        }
        else
        {
            //Failed, print the error reason
            Serial.println(fbdo.errorReason());
        }
    }
```

can anyone help? thanks

mobizt, 2021-12-20:
The index start at 0. Do you new to program ming?.

closed 2021-12-20 by mobizt

razisek, 2021-12-21:
how to send to multiple device?

mobizt, 2021-12-21:
Add more tokens and use broadcast.

https://github.com/mobizt/Firebase-ESP32/blob/150be42799b5602a12103db8985b0cdd620b4c69/examples/Messaging/Messaging.ino#L100

razisek, 2021-12-21:
thankss sir