mobizt/Firebase-ESP32 #208: Simultaneous data sent from streamCallback getting ignored

Ritwik1798, 2022-01-17:
I have been working on a IoT device and everything works great. 
But when I update multiple variables simultaneously, the firebase values are updates but nothing reflects on the device.

Please let me know if I'm missing out on something!

The code is below:
(Tried updating STATUS for SWITCH21,22,23,24 from android app simultaneously, doesnt reflect anything on device)
`#include <FirebaseESP32.h>
#include <EEPROM.h>
#define FIREBASE_HOST "xxxx"
#define FIREBASE_AUTH "xxxx"
String path = "/SHS002";
FirebaseData firebaseData;
FirebaseJson json;
FirebaseData fbdo1;
FirebaseData fbdo2;

//void allowMultipleRequests(bool enable);
//#define FIREBASE_LOCAL_IP "/NODE_IP/IP_M1_C1"
int state1, state2, state3, state4;

String device1, device2, device3, device4;

int sch1, sch2, sch3, sch4, on_prev = 3;

char dev1[20], dev2[20], dev3[20], dev4[20];

void printResult(StreamData &data);

void streamCallback(StreamData data) {
  yield();
  if (data.dataPath() == "/SWITCH21/STATUS") {
    state1 = data.intData();
    EEPROM.write(240, state1);
    EEPROM.commit();
    Serial.println(state1);
  }
  else if (data.dataPath() == "/SWITCH22/STATUS") {
    state2 = data.intData();
    EEPROM.write(250, state2);
    EEPROM.commit();
    Serial.println(state2);
  }
  else if (data.dataPath() == "/SWITCH23/STATUS") {
    state3 = data.intData();
    EEPROM.write(260, state3);
    EEPROM.commit();
    Serial.println(state3);
  }
  else if (data.dataPath() == "/SWITCH24/STATUS") {
    state4 = data.intData();
    EEPROM.write(270, state4);
    EEPROM.commit();
    Serial.println(state4);
    if(state4!=0) on_prev = state4;
    EEPROM.write(280, on_prev);
    EEPROM.commit();
  }
   if (data.dataPath() == "/SWITCH21/SCHEDULE") {
      sch1 = data.intData();
      Serial.println(sch1);
    }
   if (data.dataPath() == "/SWITCH22/SCHEDULE") {
      sch2 = data.intData();
      Serial.println(sch2);
    }
    if (data.dataPath() == "/SWITCH23/SCHEDULE") {
      sch3 = data.intData();
      Serial.println(sch3);
    }
    if (data.dataPath() == "/SWITCH24/SCHEDULE") {
      sch4 = data.intData();
      Serial.println(sch4);
    }

 
   
}
void printResult(StreamData &data)
{

  if (data.dataType() == "int")
    Serial.println(data.intData());
  else if (data.dataType() == "float")
    Serial.println(data.floatData(), 5);
  else if (data.dataType() == "double")
    printf("%.9lf\n", data.doubleData());
  else if (data.dataType() == "boolean")
    Serial.println(data.boolData() == 1 ? "true" : "false");
  else if (data.dataType() == "string" || data.dataType() == "null")
    Serial.println(data.stringData());
}

void streamTimeoutCallback(bool timeout)
{
  if (timeout)
  {
    Serial.println();
    Serial.println("Stream timeout, resume streaming...");
    Serial.println();
  }
}

void firebase_initialization() {

  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);

  if (!Firebase.beginStream(fbdo1, path))
  {
    Serial.println("------------------------------------");
    Serial.println("Can't begin stream connection...");
    Serial.println("REASON: " + fbdo1.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }

  Firebase.setStreamCallback(fbdo1, streamCallback, streamTimeoutCallback);
}`

mobizt, 2022-01-17:
Nothing is the issue unless the path you checked from data.dataPath() is not what you expected .

You should debug what's the path returned from data.dataPath().

closed 2022-01-17 by mobizt

Ritwik1798, 2022-01-17:
The paths are updated on server and device when triggered one at a time but when more than one of them are triggered simultaneously, the changes are not reflected.

To give you more clarity, /SWITCH21/STATUS when triggered reflects changes properly but when /SWITCH21/STATUS, /SWITCH22/STATUS are triggered simultaneously, only one of them is updated.

mobizt, 2022-01-17:
You may use the outdated version which the later versions come with the [javascript test](https://github.com/mobizt/Firebase-ESP32/blob/master/examples/Browser/index.html) that test the stream performance and data will not missed.

You should update the library.

mobizt, 2022-01-17:
When you updated the library, please follow the stream usage example [here](https://github.com/mobizt/Firebase-ESP32/blob/master/examples/DataChangesListener/Callback/Callback.ino).

There are many changes since the old version you used.

Ritwik1798, 2022-01-17:
Thank you so much! The latest version resolved all the issues.