mobizt/Firebase-ESP32 #213: Reboots using Callback example

simondddd, 2022-02-18:
**Describe the bug**
Reboots (when connection fails?) using Callback example.
Guru Meditation Error: (LoadProhibited) and (InstrFetchProhibited)

**To Reproduce**
Steps to reproduce the behavior:
1. Use Callback.ino example during some hours
2. Reboots happen

**Expected behavior**
Stable operation

**Versions:**
 - Arduino IDE 1.8.19
 - arduino-esp32 2.0.2
 - Firebase-ESP32 3.14.5

**Additional context**
Trying to solve stability issues in the project I am working on I have been removing code, and as it fails with the more simple version, I have tried the example, and it also fails.
They are not the same errors, but as this is the use case example I expected it to be stable.
I have tried different ESP32 and different wifi connections (work and home). Maybe the connection fails because of my internet, but I do not experience any glitch in other uses.
The code is the example with changed wifi and firebase data and the loop from 15 to 60 seconds as our projects upload data every 60 seconds.
I have tried removing "#define FIREBASE_USE_PSRAM" from FirebaseFS.h as the ESP32 I am using does not have PSRAM.
ESP Exception Decoder says "Decode Failed"

[CallbackSimon.txt](https://github.com/mobizt/Firebase-ESP32/files/8095170/CallbackSimon.txt)
[ERROR-FIREBASE_USE_PSRAM.txt](https://github.com/mobizt/Firebase-ESP32/files/8095186/ERROR-FIREBASE_USE_PSRAM.txt)
[ERROR-NO_FIREBASE_USE_PSRAM-1.txt](https://github.com/mobizt/Firebase-ESP32/files/8095187/ERROR-NO_FIREBASE_USE_PSRAM-1.txt)
[ERROR-NO_FIREBASE_USE_PSRAM-2.txt](https://github.com/mobizt/Firebase-ESP32/files/8095189/ERROR-NO_FIREBASE_USE_PSRAM-2.txt)
[ERROR-NO_FIREBASE_USE_PSRAM-3.txt](https://github.com/mobizt/Firebase-ESP32/files/8095191/ERROR-NO_FIREBASE_USE_PSRAM-3.txt)

simondddd, 2022-02-18:
Maybe I am doing something wrong or reboot is expected behaviour when loosing connection and both my internet connections are mediocre, but I am not and expert in this.
What more can I test or do?

Also I think FIREBASE_CLIENT_VERSION from FirebaseESP32.h is not updated from 3.14.4 to 3.14.5 number (It says 3.14.4 version in serial output)

mobizt, 2022-02-18:
The PSRAM is not the issue.

The define FIREBASE_USE_PSRAM is just to allow library to use PSRAM.

If PSRAM is never enable in IDE during build the PSRAM will never use as we check the existence of PSRAM using ESP32 core sdk even this FIREBASE_USE_PSRAM was defined.

You should enable WiFI and SSL debug in Arduino IDE.

You should debug the stack trace and post the result here.

Please don't attach as file or image to the post because of I'm not convenient to download and open it on my PC.

With the well manage wrapper TCP client as you can use other Arduino Client with this library too, with my intensive test, this issue will not happen in all WIFI, ethernet and internet situations.

The most possible causes are in somewhere in the core libraries which fail to read/write memory due to unstable, noisy or weak supply voltage.

mobizt, 2022-02-18:
I test with latest ESP32 core sdk with the older Arduino IDE v1.18.16.

I will test with latest Arduino IDE and reply here.

At this moment, I guess that may be a compiler related issue about add-ons header files

mobizt, 2022-02-18:
P
P

simondddd, 2022-02-18:
Thank you for the PSRAM clarification.
I have enabled: Tools -> Core Debug Level -> Verbose
What more do you suggest me to do for debug? I have just bought an esp-prog jtag device, but I have still to learn how to use.
Are you suggesting to use a specific TCP library, or that the included is very good? Sorry for having to ask. 
I have reproduced the issue and this is what I see:

[V][ssl_client.cpp:61] start_ssl_client(): Free internal heap before TLS
This reduces from ~250000 at start, to ~200000 after the first loop, and then slowly and gradually to ~128000 where it fails with: [E][ssl_client.cpp:36] _handle_error(): [start_ssl_client():254]: (-32512) SSL - Memory allocation failed.

This is the serial output near the reboot.

`Received stream payload size: 69 (Max. 74)

[21255935][V][ssl_client.cpp:311] stop_ssl_socket(): Cleaning SSL connection.
[21255937][V][ssl_client.cpp:311] stop_ssl_socket(): Cleaning SSL connection.
[21255938][V][ssl_client.cpp:61] start_ssl_client(): Free internal heap before TLS 131112
[21255946][V][ssl_client.cpp:67] start_ssl_client(): Starting socket
[21256005][V][ssl_client.cpp:143] start_ssl_client(): Seeding the random number generator
[21256006][V][ssl_client.cpp:152] start_ssl_client(): Setting up the SSL/TLS structure...
[21256010][I][ssl_client.cpp:173] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
[21256018][V][ssl_client.cpp:244] start_ssl_client(): Setting hostname for TLS session...
[21256027][V][ssl_client.cpp:259] start_ssl_client(): Performing the SSL/TLS handshake...
[21257333][V][ssl_client.cpp:280] start_ssl_client(): Verifying peer X.509 certificate...
[21257333][V][ssl_client.cpp:288] start_ssl_client(): Certificate verified.
[21257336][V][ssl_client.cpp:303] start_ssl_client(): Free internal heap after TLS 88040
[21257344][V][ssl_client.cpp:351] send_ssl_data(): Writing HTTP request with 37 bytes...
[21257354][V][ssl_client.cpp:351] send_ssl_data(): Writing HTTP request with 921 bytes...
[21257363][V][ssl_client.cpp:351] send_ssl_data(): Writing HTTP request with 250 bytes...
[21257370][V][ssl_client.cpp:351] send_ssl_data(): Writing HTTP request with 26 bytes...
Set json... ok

sream path, /test/stream/data
event path, /json
data type, json
event type, put

Pretty printed JSON data:
{
	"data": "hello",
	"num": 355
}
Iterate JSON data:

0, Type: object, Name: data, Value: "hello"
1, Type: object, Name: num, Value: 355

Received stream payload size: 69 (Max. 74)

Token info: type = id token, status = on request
[21280999][V][ssl_client.cpp:311] stop_ssl_socket(): Cleaning SSL connection.
[21281000][V][ssl_client.cpp:311] stop_ssl_socket(): Cleaning SSL connection.
[21281001][V][ssl_client.cpp:311] stop_ssl_socket(): Cleaning SSL connection.
[21281013][V][ssl_client.cpp:311] stop_ssl_socket(): Cleaning SSL connection.
[21281015][V][ssl_client.cpp:311] stop_ssl_socket(): Cleaning SSL connection.
[21281028][V][ssl_client.cpp:311] stop_ssl_socket(): Cleaning SSL connection.
[21281029][V][ssl_client.cpp:311] stop_ssl_socket(): Cleaning SSL connection.
[21281040][V][ssl_client.cpp:61] start_ssl_client(): Free internal heap before TLS 127720
[21281042][V][ssl_client.cpp:311] stop_ssl_socket(): Cleaning SSL connection.
[21281055][V][ssl_client.cpp:67] start_ssl_client(): Starting socket
[21281057][V][ssl_client.cpp:311] stop_ssl_socket(): Cleaning SSL connection.
[21281070][V][ssl_client.cpp:311] stop_ssl_socket(): Cleaning SSL connection.
[21281077][V][ssl_client.cpp:311] stop_ssl_socket(): Cleaning SSL connection.
[21281084][V][ssl_client.cpp:311] stop_ssl_socket(): Cleaning SSL connection.
[21281091][V][ssl_client.cpp:311] stop_ssl_socket(): Cleaning SSL connection.
[21281098][V][ssl_client.cpp:311] stop_ssl_socket(): Cleaning SSL connection.
[21281104][V][ssl_client.cpp:311] stop_ssl_socket(): Cleaning SSL connection.
[21281111][V][ssl_client.cpp:311] stop_ssl_socket(): Cleaning SSL connection.
[21281116][V][ssl_client.cpp:143] start_ssl_client(): Seeding the random number generator
[21281118][V][ssl_client.cpp:311] stop_ssl_socket(): Cleaning SSL connection.
[21281131][V][ssl_client.cpp:152] start_ssl_client(): Setting up the SSL/TLS structure...
[... 143 lines cut ...]
[1631560][V][ssl_client.cpp:259] start_ssl_client(): Performing the SSL/TLS handshake...
[1632499][V][ssl_client.cpp:280] start_ssl_client(): Verifying peer X.509 certificate...
[1632499][V][ssl_client.cpp:288] start_ssl_client(): Certificate verified.
[1632502][V][ssl_client.cpp:303] start_ssl_client(): Free internal heap after TLS 152400
[1632510][V][ssl_client.cpp:351] send_ssl_data(): Writing HTTP request with 37 bytes...
[1632519][V][ssl_client.cpp:351] send_ssl_data(): Writing HTTP request with 921 bytes...
[1632528][V][ssl_client.cpp:351] send_ssl_data(): Writing HTTP request with 250 bytes...
[1632535][V][ssl_client.cpp:351] send_ssl_data(): Writing HTTP request with 25 bytes...
`

This is the memory in some middle loops:

`[11895782][V][ssl_client.cpp:61] start_ssl_client(): Free internal heap before TLS 160744
[11896834][V][ssl_client.cpp:303] start_ssl_client(): Free internal heap after TLS 117712
[12075785][V][ssl_client.cpp:61] start_ssl_client(): Free internal heap before TLS 160276
[12076863][V][ssl_client.cpp:303] start_ssl_client(): Free internal heap after TLS 117220
[12255787][V][ssl_client.cpp:61] start_ssl_client(): Free internal heap before TLS 159816
[12256898][V][ssl_client.cpp:303] start_ssl_client(): Free internal heap after TLS 116688
[... cut ...] l_client(): Free internal heap before TLS 159300
[12436965][V][ssl_client.cpp:303] start_ssl_client(): Free internal heap after TLS 116184
[12615793][V][ssl_client.cpp:61] start_ssl_client(): Free internal heap before TLS 158864
[12616960][V][ssl_client.cpp:303] start_ssl_client(): Free internal heap after TLS 115776
[12795796][V][ssl_client.cpp:61] start_ssl_client(): Free internal heap before TLS 158404
[12797068][V][ssl_client.cpp:303] start_ssl_client(): Free internal heap after TLS 115348
`

mobizt, 2022-02-20:
The library update [v3.15.0](https://github.com/mobizt/Firebase-ESP32/releases/tag/v3.15.0) is available with the bugs fixed.
Please try.

closed 2022-02-20 by mobizt

simondddd, 2022-02-21:
Yes, example works stable now.
Thank you.