mobizt/Firebase-ESP32 #219: 'SearchCriteria' is not a member of 'FirebaseJson'

halilozdgn, 2022-04-06:
Hey there, i've just updated esp core version from 1.0.6 to 2.0.2. After that i got some errors and upgraded Firebase-ESP32 library to 3.15.5. When i try compile after update i got following erorrs:
`
'SearchCriteria' is not a member of 'FirebaseJson'
error: 'SearchCriteria' is not a member of 'FirebaseJson'
error: 'criteria' was not declared in this scope
error: 'class FirebaseJson' has no member named 'search'
error: 'class FirebaseJson' has no member named 'search'
error: 'class FirebaseJsonData' has no member named 'searchPath'
In function 'void getDownloadURL(String)':
sketch_apr05a:635:17: error: 'SearchCriteria' is not a member of 'FirebaseJson'
   FirebaseJson::SearchCriteria criteria;
error: 'criteria' was not declared in this scope
   criteria.path = "/" + uid + "/url";

`

And this is the function i used:

`void getDownloadURL(String uid)
{
  // json elimizde hala tutulu tekrar istek göndermeye gerek yok
  FirebaseJson::SearchCriteria criteria;
  FirebaseJsonData result;
  String downloadURL;
  Serial.printf("U id : ");
  Serial.println(uid);
  criteria.path = "/" + uid + "/url";
  criteria.searchAll = true;
  int count = json.search(result, criteria, true);
  json.toString(Serial, true);
  if (count > 0)
  {
    if (result.success)
    {
      Serial.println("Result Success...");
      downloadURL = result.to<String>();
    }
  }
  int deger = downloadURL.indexOf('"');
  downloadURL = downloadURL.substring(deger + 1, downloadURL.length());
  deger = downloadURL.indexOf('"');
  downloadURL = downloadURL.substring(0, deger);
  Serial.println(downloadURL);
  Serial.println("updateFirmware çalışacak");
  updateFirmware(downloadURL);
}`

After i got these errors i tried new esp32/esp8266 client library and and firebasejson library but nothing changed.  With new update is there any change of using FirabeseJson::SearchCriteria?

Arduino Version 1.8.15
ESP32 Core Version 2.0.2

mobizt, 2022-04-06:
The search features removed from FirebaseJson included in this Firebase library due to bugs then I decide to remove it.

halilozdgn, 2022-04-06:
Ok then which version you've exactly removed seach feature?

mobizt, 2022-04-06:
This is the [last version](https://github.com/mobizt/Firebase-ESP32/releases/tag/v3.12.10) that FirebaseJson supports search.

halilozdgn, 2022-04-06:
With 3.12.10 and 3.12.9 still same issue
`'SearchCriteria' is not a member of 'FirebaseJson'`

mobizt, 2022-04-06:
If you install FirebaseJson, remove it, use the built-in version.

halilozdgn, 2022-04-06:
After removing FirebaseJson it works, Thanks

closed 2022-04-06 by halilozdgn