mobizt/Firebase-ESP32 #227: BUG sometimes it get stuck at "on initializing"

cfficaurzua, 2022-04-21:
Sometimes it works just fine 
```
Firebase Client v3.15.5

Token info: type = OAuth2.0 access token, status = on initializing
Token info: type = OAuth2.0 access token, status = on signing
Token info: type = OAuth2.0 access token, status = on request
Token info: type = OAuth2.0 access token, status = ready
```

but sometimes it get stuck at 

```
Firebase Client v3.15.5

Token info: type = OAuth2.0 access token, status = on initializing
```
Sometimes it takes 5 min, but sometimes I wait more than an hour with no response.
Is there anyway I can put a timeout?

Thanks

mobizt, 2022-04-21:
It is the NTP time requesting process and no timeout because it's not blocking process. 

It will stay at that process when the device time was not set yet.

You should check your internet connection (not WiFi).

closed 2022-04-21 by mobizt

mobizt, 2022-04-21:
Or you can set the device time manually before calling `Firebase.begin` and make sure you get the valid timestamp from calling `time(nullptr)`.

There are many examples on internet for device time setting from NTP server and RTC.

cfficaurzua, 2022-04-22:
Much appreciated

> I recommended you to update the library to [v3.15.8](https://github.com/mobizt/Firebase-ESP32/releases/tag/v3.15.8).
> 
> This is the important update to fix stream and token initialization issues.
> 
> The token initialization process is now can be resumed when the internet connection lost and the timed out error during NTP server time synching will be reported.