mobizt/Firebase-ESP32 #228: ็HELP Firebase.pushJSON

keecook, 2022-04-28:
I'm using pushJSON with library version 3.16.1 but it doesn't work with error like this:

exit status 1
no matching function for call to 'FirebaseESP32::pushJSON(FirebaseData&, String&, String&)'

This is my Code:
  String jsonData = "{\"photo\":\"" + Photo2Base64() + "\"}";
  String photoPath = "/photo";
  Firebase.pushJSON(firebaseData, photoPath, jsonData);

I'm using with ESP32-CAM (AI Thinker ESP32-CAM version 1.0.5)

mobizt, 2022-04-28:
You should check the documentation for the function description first.
https://github.com/mobizt/Firebase-ESP32/tree/master/src#append-new-child-node-key-and-value-using-firebasejson-object-to-the-defined-database-path

closed 2022-04-28 by mobizt

keecook, 2022-04-28:
Sorry, I don't know how to fix this problem. can you help me

keecook, 2022-04-28:
this is my code`/*
ESP32-CAM Load images from SD card to enroll faces and recognize faces automatically.
Author : ChungYi Fu (Kaohsiung, Taiwan)  2021-6-29 21:30
https://www.facebook.com/francefu
*/
#include "esp_camera.h"
#include <WiFi.h>
#include <FirebaseESP32.h>
#include <time.h>
#define SDA 15
#define SCL 14
#include <VL53L0X.h>
#include <Wire.h>

VL53L0X sensor;
FirebaseData firebaseData;
FirebaseData fbdo;

#define FIREBASE_HOST "project-5713534941910068953-default-rtdb.firebaseio.com"
#define FIREBASE_AUTH "HsTpNMk6siFDvQvAdzgB6h7JTPpA4UK82zQVOpAF"

#define Relay 2
#define Red 12
#define Green 13
#include "Base64.h"

const char* ssid = "2.4G_55"; //Wifi Name SSID
const char* password = "00000000"; //WIFI Password

//人臉辨識同一人人臉註冊影像數
#define ENROLL_CONFIRM_TIMES 5

//可由網頁get-still按鈕取得解析度CIF影像另存於SD卡 http://192.168.xxx.xxx/capture  (FRAMESIZE_CIF)
String filepath[5] = {"/1.jpg", "/2.jpg", "/3.jpg", "/4.jpg", "/5.jpg"};  //1.jpg, 2.jpg, ...., 35.jpg
int image_width = 400;  
int image_height = 296;

//人臉辨識同一人人臉註冊影像數
#define ENROLL_CONFIRM_TIMES 5
//人臉辨識註冊人數
#define FACE_ID_SAVE_NUMBER 7

//設定人臉辨識顯示的人名
String recognize_face_matched_name[7] = {"Name0","Name1","Name2","Name3","Name4","Name5","Name6"};

#include "soc/soc.h"             //用於電源不穩不重開機 
#include "soc/rtc_cntl_reg.h"    //用於電源不穩不重開機 
#include "esp_camera.h"          //視訊函式
#include "img_converters.h"      //影像格式轉換函式
#include "fb_gfx.h"              //影像繪圖函式
#include "fd_forward.h"          //人臉偵測函式
#include "fr_forward.h"          //人臉辨識函式
#include "FS.h"                  //檔案系統函式
#include "SD_MMC.h"              //SD卡存取函式

//安信可ESP32-CAM模組腳位設定
#define PWDN_GPIO_NUM     32
#define RESET_GPIO_NUM    -1
#define XCLK_GPIO_NUM      0
#define SIOD_GPIO_NUM     26
#define SIOC_GPIO_NUM     27

#define Y9_GPIO_NUM       35
#define Y8_GPIO_NUM       34
#define Y7_GPIO_NUM       39
#define Y6_GPIO_NUM       36
#define Y5_GPIO_NUM       21
#define Y4_GPIO_NUM       19
#define Y3_GPIO_NUM       18
#define Y2_GPIO_NUM        5
[... 398 lines cut ...]
      if(door_app == "true"){
        Serial.println("Door open");
        digitalWrite(Relay,HIGH);
        digitalWrite(Green,HIGH);
        digitalWrite(Red,LOW);
        Firebase.setString(firebaseData,"STATUS/Realtime","ประตูเปิด");
      }
      if(door_app == "false"){
        Serial.print("Doooooor"); 
        digitalWrite(Relay,LOW);
        digitalWrite(Green,LOW);
        digitalWrite(Red,HIGH);
        Firebase.setString(firebaseData,"STATUS/Realtime", "ประตูปิด");
      }
    }
  }
  Serial.println(sensor.readRangeContinuousMillimeters());
  if(sensor.readRangeContinuousMillimeters()<500){
  String jsonData = "{\"photo\":\"" + Photo2Base64() + "\"}";
  String photoPath = "/photo";
  Firebase.pushJSON(firebaseData, photoPath, jsonData);
  faceRecognition();
  delay(100);
}
}`

mobizt, 2022-04-28:
The type of third parameter of `Firebase.pushJson` functon is `FirebaseJson` instead of `String`.

mobizt, 2022-04-28:
For `FirebaseJson`, see [this example](https://github.com/mobizt/Firebase-ESP32/blob/master/examples/FirebaseJson/BasicUsage/Create_Edit_Parse/Create_Edit_Parse.ino) for the usage.

Instead of convert the image data as base64 encoded string and store it to database, I recommend to store the data to file and use [these functions](https://github.com/mobizt/Firebase-ESP32/tree/master/src#set-binary-data-from-the-file-store-on-sd-cardflash-memory-to-the-defined-database-path) to read data from file and store to database as base64 string.

The file functions above prevent large memory allocation for base64 encoded string which caused the low memory situation which may not enough for SSL client to work.

See [these examples](https://github.com/mobizt/Firebase-ESP32/tree/master/examples/File) for how `File` functions work.

keecook, 2022-04-28:
thank you very much I can fix this problem now. thank you for helping me