mobizt/Firebase-ESP32 #237: BUG: after 1 hour aprox, ESP32 doesn´t read changes in Real Tieme Database (RTDB)

fjmduran, 2022-08-08:
Hello, 
**Describe the bug**
Until this weekend everything work OK, but yesterday I realized that ESP32 did not update when data change in real time database (RTDB).
Today I have checked and the behaviour is the same, but I have been able to see ESP32 leave update change in RTDB after second token refresh.

**To Reproduce**
Steps to reproduce the behavior:
1. Load sketch
2. Wait until second token refresh time
3. To check ESP32 does not read updating RTDB

**Expected behavior**
I hope ESP32 always can read changes in RTDB

**Screenshots**
[I have attached screenshot of Arduino serial monitor](https://github.com/fjmduran/images/blob/master/githubBug/firebaseESP32/cannt-read-after-second-token-request.png?raw=true)

**IDE and its version:**
 - Arduino v1.8.15
 - Library Firebase-ESP32 v4.0.3

**Code**

```
#include <WiFi.h>
#include <FirebaseESP32.h>
#include <HTTPClient.h>
#include <TimeLib.h>
#include <esp_int_wdt.h>
#include <esp_task_wdt.h>

// Provide the token generation process info.
#include <addons/TokenHelper.h>

// Provide the RTDB payload printing info and other helper functions.
#include <addons/RTDBHelper.h> //TODO

#define FIREBASE_HOST "https://*****.europe-west1.firebasedatabase.app"
#define API_KEY "*****"
#define USER_EMAIL "*****"
#define USER_PASSWORD "*****"

#define WIFI_SSID "*****"
#define WIFI_PASSWORD "*****"

// Your Domain name with URL path or IP address with path
String serverName = "https://*****.cloudfunctions.net/"; // function for send ESP32 info
String ubicationId = "/test";
String microId = "/porton";

#define NTP_SERVER "pool.ntp.org"

#define BlinkLED 12
#define rele01 15
#define strokeEnd 14
#define strokeEndLed 13

int period = 1000;
unsigned long time_now = 0;

const int millisecondsDoorOpened = 1000;
int intSendESPInfo = 3600 * 12;
int contSendESPInfo = intSendESPInfo; // para que nada más iniciar la placa envíe su info
int timeOutCounter = 0;
bool openStrokeEnd = false;

// Define the Firebase Data object
FirebaseData fbdo;
FirebaseData stream;

[... 257 lines cut ...]
    String _minute = static_cast<String>(minute(t));
    String _second = static_cast<String>(second(t));
    String date = _day + "-" + _hour + ":" + _minute + ":" + _second;
    return date;
}

void hard_restart()
{
    Serial.println("REINICIO FORZADO");
    esp_task_wdt_init(1, true);
    esp_task_wdt_add(NULL);
    while (true)
        ;
}

void upResetCounter()
{
    Serial.println(timeOutCounter++);
    if (timeOutCounter >= 1000)
    {
        timeOutCounter = 0;
        hard_restart();
    }
}
```

mobizt, 2022-08-08:
You never provide the actual debug print from the library and library itself does not have this kind of issue as it is intensively test 24/7 for years and updated frequently unless user code bugs.

You should debug your code and enable debug options in Arduino IDE.

The function Firebase.ready should be called repeatedly to handle authentication tasks and provide the state that you can access the Firebase services.

The token refreshing failed when your device free Heap is low below 100k.

closed 2022-08-08 by mobizt

mobizt, 2022-08-08:
You must keep the database under the streaming path minimal.

The stream payload to store in the `FirebaseData` was limited to 4096 bytes and can change via [this function](https://github.com/mobizt/Firebase-ESP32/tree/master/src#set-the-http-response-size-limit). 

When new stream connection established, all data at the streaming path can not store in `FirebaseData` if it is larger than the limit, the stored data will be truncated and cannot be parsed correctly in case of JSON payload.

Instead of increase the payload limit, you should avoid to add or append data to the nodes under your streaming path that causes your data to grow up.

mobizt, 2022-08-08:
Ok, I just seen your debug image.

Your code is lack of debug information that you can follow what is the line that is currently executing.

mobizt, 2022-08-08:
I can run your code and the stream works normally but your should avoid using String object and string literal joining with library string argument because this programming syntax is for Arduino String only.

`if (Firebase.updateNode(fbdo, ubicationId + "/" + deviceToModify, json))` 

should be

` if (Firebase.updateNode(fbdo, String(ubicationId + "/" + deviceToModify), json))`

mobizt, 2022-08-08:
The String and string literal is not support in all my libraries that use MB_String class.

This will give unexpected result in your code above and is what you called bug.

fjmduran, 2022-08-08:
Thanks for your time.
I have updated my code with your suggestions:
- I have deleted String object and string literal joining
- I check free memory and neves is low by 100k
- I have added line number to debug message
But it has same behaviour. 

[You can see serial monitor output here](https://github.com/fjmduran/images/blob/master/githubBug/firebaseESP32/cannt-read-after-second-token-request-02.png?raw=true)

Mi code now is:
```#include <WiFi.h>
#include <FirebaseESP32.h>
#include <HTTPClient.h>
#include <TimeLib.h>
#include <esp_int_wdt.h>
#include <esp_task_wdt.h>

// Provide the token generation process info.
#include <addons/TokenHelper.h>

// Provide the RTDB payload printing info and other helper functions.
#include <addons/RTDBHelper.h>

#define FIREBASE_HOST "https://****rtdb.europe-west1.firebasedatabase.app"
#define API_KEY "****"
#define USER_EMAIL "****"
#define USER_PASSWORD "****"

#define WIFI_SSID "****"
#define WIFI_PASSWORD "****"

// Your Domain name with URL path or IP address with path
String serverName = "https://europe-west2****.cloudfunctions.net/"; 
String ubicationId = "/test";
String microId = "/porton";

#define NTP_SERVER "pool.ntp.org"

#define BlinkLED 12
#define rele01 15
#define strokeEnd 14
#define strokeEndLed 13

int period = 1000;
unsigned long time_now = 0;

const int millisecondsDoorOpened = 1000;
int intSendESPInfo = 3600 * 12;
int contSendESPInfo = intSendESPInfo; // para que nada más iniciar la placa envíe su info
int timeOutCounter = 0;
bool openStrokeEnd = false;

// Define the Firebase Data object
FirebaseData fbdo;
FirebaseData stream;

// Define the FirebaseAuth data for authentication data
FirebaseAuth auth;

// Define the FirebaseConfig data for config data
FirebaseConfig config;

FirebaseJson updateData;
FirebaseJson json;

void setup()
{
    pinMode(BlinkLED, OUTPUT);
    pinMode(rele01, OUTPUT);
    pinMode(strokeEndLed, OUTPUT);
[... 237 lines cut ...]
    String date = _day + "-" + _hour + ":" + _minute + ":" + _second;
    return date;
}

void hard_restart()
{
    printInConsole("REINICIO FORZADO", __LINE__);
    esp_task_wdt_init(1, true);
    esp_task_wdt_add(NULL);
    while (true)
        ;
}

void upResetCounter()
{
    if (timeOutCounter >= 1000)
    {
        timeOutCounter = 0;
        hard_restart();
    }
}```

Please, you see anything else?

Regards,

mobizt, 2022-08-09:
Are you new to this Firebase library. 

You may don't understand the stream operation of library correctly.

You must play with example until you understand the stream behavior in all stream event cases before adapt it.

At first stream connection or reconnection after timed out, library sends the http request to server with additional headers to tell server that this connection is in stream mode and server will send the response payload back which is all data at the streaming path.

Library will fire the `streamAvialible` with whole data as it get from sever which is not data from the changes  in database.

Then if the changes occurred in your database under the streaming path, server will push that changed data to devices.

You must use this functions to determine the stream event, its path and data.

https://github.com/mobizt/Firebase-ESP32/blob/6722bef924d06477043006dcc71132111019e88f/examples/DataChangesListener/NoCallback/NoCallback.ino#L146-L162

mobizt, 2022-08-09:
As long as this function `stream.streamTimeout()` returns false, your device was connected to server normally and stream won't be missed.

This [web client example](https://github.com/mobizt/Firebase-ESP32/blob/master/examples/Browser/index.html) shows the performance of library and Firebase that guarantee no data missed during stream. 

You can force token to expire in 20 seconds with this config

`config.signer.preRefreshSeconds = 5600 - 20;`

mobizt, 2022-08-09:
**THIS IS FOR YOUR INFORMATION**

> The library was intensively tested 24/7 for years and I strongly confirm that it works normally in all properly use cases.
> 
> As library developer and maintainer, I'm so busy and can't help someone to code or find the bug in the program but provide the full documentation and examples that user can learn and test.
> 
> Then before you open the issue, try and follow the examples and make sure you understand the concepts of Firebase and the usage of the library.
> 
> Don't ignore the debug printing during test as most functions return boolean status of operation. The `fbdo.errorReason()` will give you the error detail.
> 
> If you are sure that it is library issue, you should provide the minimum, clean (no 3rd library), clear but complete code that I can test to reproduce the issue.
> 
> If you are not sure, start with the [Discussions](https://github.com/mobizt/Firebase-ESP32/discussions) instead.

fjmduran, 2022-08-09:
Thanks for your time again.

I am running basic example and I think I have been able to find my problem. After second token refresh, the event type is auth_revoked.

I have projects with more one years using your library and everthing was OK until this weekend.

[image]

Do you know what I must do for solve it?
My RTDB rules are:

[image]

I have search about auth_revoked but it is unsuccess.

Regards,

mobizt, 2022-08-09:
You may need to upgrade to blaze plan when quota limits reached.

fjmduran, 2022-08-09:
I have already blaze plan

[image]

mobizt, 2022-08-09:
Firebase may limit your account due to some security reasons.

You can contact the Firebase support team or create the new project.

Don't ask me about Firebase policy about security, privacy and limitations.

fjmduran, 2022-08-09:
OK, thanks