mobizt/Firebase-ESP32 #240: ESP is not able to read using getString function when using v3.11 onwards

Hassaan18, 2022-08-25:
Firebase read function works fine until v3.10.5 but 3.11 takes the database path as a constant template. Even doing so, it can write to the database using setString function but on reading, it's returning empty string.

mobizt, 2022-08-26:
The library was updated. You should also update your code to pass String pointer to readString function.

closed 2022-08-26 by mobizt

Hassaan18, 2022-08-26:
Yes, I'm passing that to the function. Please refer to the code below

```
  String tmp;
  String path = "/" + String(i);
  Firebase.setString(firebasedata, path, "XYZ");
  Firebase.getString(firebasedata, path, tmp);
```

mobizt, 2022-08-26:
No, you can't pass as object like that in the new version because the library APIs changed for almost year. You should  pass the pointer of String e.g. &tmp.

mobizt, 2022-08-26:
For example.

`Firebase.getString(firebasedata, path, &tmp);`

mobizt, 2022-08-26:
You should update the library to latest version first.