mobizt/Firebase-ESP32 #241: BUG

ludvigaldrin, 2022-09-19:
**Describe the bug**
Im trying to get the example from this page to work:
https://medium.com/@vibrologic/serverless-iots-with-firebase-realtime-database-and-esp32-def049181b57

Have just copied Node1 code, setup Realtime database according to instructions, have added HTTPClientESP32Ex from url. Have installed version 4.1.0 Firebase ESP32 Client by Mobizt. Then added my details to FIREBASE_HOST, FIREBASE_AUTH, WIFI_SSID, WIFI_PASSWORD.

I have a Wemos D1 Mini
During boot it connectes to wifi correct but then I get the error bellow.

Doing some commenting I found that its this line that produces the error:
`Firebase.beginStream(firebaseData1, path + "/" + nodeID)`

```
14:30:04.748 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
14:30:04.748 -> configsip: 0, SPIWP:0xee
14:30:04.786 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
14:30:04.786 -> mode:DIO, clock div:1
14:30:04.786 -> load:0x3fff0030,len:1184
14:30:04.786 -> load:0x40078000,len:13160
14:30:04.786 -> load:0x40080400,len:3036
14:30:04.786 -> entry 0x400805e4
14:30:05.147 -> 
14:30:05.147 -> 
14:30:05.260 -> Connecting to Wi-Fi.
14:30:05.542 -> Connected with IP: xxxx
14:30:05.575 -> 
14:30:05.575 -> Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
14:30:05.575 -> 
14:30:05.575 -> Core  1 register dump:
14:30:05.575 -> PC      : 0x400d2a65  PS      : 0x00060f30  A0      : 0x800d2a91  A1      : 0x3ffb2640  
14:30:05.575 -> A2      : 0x00000000  A3      : 0x646f4e2f  A4      : 0x00000000  A5      : 0x00000000  
14:30:05.575 -> A6      : 0x00000000  A7      : 0xff000000  A8      : 0x800d2a5f  A9      : 0x3ffb2620  
14:30:05.609 -> A10     : 0x00000000  A11     : 0x646f4e2f  A12     : 0x00000000  A13     : 0x3ffbb048  
14:30:05.609 -> A14     : 0x00060d20  A15     : 0x00000089  SAR     : 0x00000008  EXCCAUSE: 0x0000001d  
14:30:05.609 -> EXCVADDR: 0x00000000  LBEG    : 0x400898b4  LEND    : 0x400898bf  LCOUNT  : 0x00000000  
14:30:05.609 -> 
14:30:05.609 -> 
14:30:05.609 -> Backtrace:0x400d2a62:0x3ffb26400x400d2a8e:0x3ffb2660 0x400d2b2c:0x3ffb2680 0x400d4ea1:0x3ffb26c0 0x400df049:0x3ffb26f0 0x400d2ff7:0x3ffb27a0 0x400e99d2:0x3ffb2820 
14:30:05.644 ->
```

**To Reproduce**
Steps to reproduce the behavior:
Running example

**Expected behavior**
Working :)

**Screenshots**
NA

**IDE and its version:**
 - Arduino
 - Version 1.8.19

**ESP32 Arduino Core SDK version**
 - Version [e.g. 2.0.2]

**Additional context**
Add any other context about the problem here.

mobizt, 2022-09-19:
This library is for ESP32.

You are using 4 years obsoleted example with the last library. Everything was changed so far.

You should use [these examples](https://github.com/mobizt/Firebase-ESP32/tree/master/examples) for latest version of library.

closed 2022-09-19 by mobizt

mobizt, 2022-09-19:
With that obsolete example, you should use 

`Firebase.beginStream(firebaseData1, String(path + "/" + nodeID))`

You can't assign concatenate string i.e `path + "/" + nodeID` to all functions directly.

ludvigaldrin, 2022-09-19:
Thanks for your help. Maybe changing the Readme then since I just read that and got the links from it :)
Thanks any how

ludvigaldrin, 2022-09-19:
A found better samples thanks. Great work