mobizt/Firebase-ESP32 #246: BUG : Debug exception reason: Stack canary watchpoint triggered (task_stream1073)

mrezzaaa, 2022-10-27:
**Describe the bug**

Hi, i just creating a program using WiFi and BLE for esp32 to listen rtdb and sending command via BLE(listener) to nearby devices(node/command executor) when value on rtdb changed. 

First handling stream callback run smoothly.
After handling some stream callback function the devices (listener) rebooted.

**To Reproduce**
Steps to reproduce the behavior:
I had decode stack trace exception via EspExceptionDecoder tools and here what i got 
```
0x400d7065: NimBLEClient::getService(NimBLEUUID const&) at .pio/libdeps/esp32dev/NimBLE-Arduino/src/NimBLEClient.cpp line 605
0x400d7065: NimBLEClient::getService(NimBLEUUID const&) at .pio/libdeps/esp32dev/NimBLE-Arduino/src/NimBLEClient.cpp line 605
0x400d7065: NimBLEClient::getService(NimBLEUUID const&) at .pio/libdeps/esp32dev/NimBLE-Arduino/src/NimBLEClient.cpp line 605
0x400d7065: NimBLEClient::getService(NimBLEUUID const&) at .pio/libdeps/esp32dev/NimBLE-Arduino/src/NimBLEClient.cpp line 605
0x400d7065: NimBLEClient::getService(NimBLEUUID const&) at .pio/libdeps/esp32dev/NimBLE-Arduino/src/NimBLEClient.cpp line 605
0x400d7065: NimBLEClient::getService(NimBLEUUID const&) at .pio/libdeps/esp32dev/NimBLE-Arduino/src/NimBLEClient.cpp line 605
0x400d7065: NimBLEClient::getService(NimBLEUUID const&) at .pio/libdeps/esp32dev/NimBLE-Arduino/src/NimBLEClient.cpp line 605
0x400d7065: NimBLEClient::getService(NimBLEUUID const&) at .pio/libdeps/esp32dev/NimBLE-Arduino/src/NimBLEClient.cpp line 605
0x400d7065: NimBLEClient::getService(NimBLEUUID const&) at .pio/libdeps/esp32dev/NimBLE-Arduino/src/NimBLEClient.cpp line 605
0x400d7065: NimBLEClient::getService(NimBLEUUID const&) at .pio/libdeps/esp32dev/NimBLE-Arduino/src/NimBLEClient.cpp line 605
0x400d7065: NimBLEClient::getService(NimBLEUUID const&) at .pio/libdeps/esp32dev/NimBLE-Arduino/src/NimBLEClient.cpp line 605
0x400d7065: NimBLEClient::getService(NimBLEUUID const&) at .pio/libdeps/esp32dev/NimBLE-Arduino/src/NimBLEClient.cpp line 605
0x400d7065: NimBLEClient::getService(NimBLEUUID const&) at .pio/libdeps/esp32dev/NimBLE-Arduino/src/NimBLEClient.cpp line 605
0x400d7065: NimBLEClient::getService(NimBLEUUID const&) at .pio/libdeps/esp32dev/NimBLE-Arduino/src/NimBLEClient.cpp line 605
0x400d7065: NimBLEClient::getService(NimBLEUUID const&) at .pio/libdeps/esp32dev/NimBLE-Arduino/src/NimBLEClient.cpp line 605
0x400d7065: NimBLEClient::getService(NimBLEUUID const&) at .pio/libdeps/esp32dev/NimBLE-Arduino/src/NimBLEClient.cpp line 605
0x400d7065: NimBLEClient::getService(NimBLEUUID const&) at .pio/libdeps/esp32dev/NimBLE-Arduino/src/NimBLEClient.cpp line 605
0x400d7065: NimBLEClient::getService(NimBLEUUID const&) at .pio/libdeps/esp32dev/NimBLE-Arduino/src/NimBLEClient.cpp line 605
0x400d7065: NimBLEClient::getService(NimBLEUUID const&) at .pio/libdeps/esp32dev/NimBLE-Arduino/src/NimBLEClient.cpp line 605
0x400d7065: NimBLEClient::getService(NimBLEUUID const&) at .pio/libdeps/esp32dev/NimBLE-Arduino/src/NimBLEClient.cpp line 605
0x400d75c6: NimBLEClient::getService(char const*) at /Users/icewalker/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch3/xtensa-esp32-elf/include/c++/8.4.0/ext/new_allocator.h line 79
0x400d3e4d: sendingCommand(String) at src/main.cpp line 236
0x400d44d6: streamCallback(FirebaseStream) at src/main.cpp line 319
0x400f1e32: FB_RTDB::sendCB(FirebaseData*) at .pio/libdeps/esp32dev/Firebase Arduino Client Library for ESP8266 and ESP32/src/rtdb/FB_RTDB.cpp line 3406
0x400f4502: FB_RTDB::handleResponse(FirebaseData*, fb_esp_rtdb_request_info_t*) at .pio/libdeps/esp32dev/Firebase Arduino Client Library for ESP8266 and ESP32/src/rtdb/FB_RTDB.cpp line 3162
0x400f4b3a: FB_RTDB::waitResponse(FirebaseData*, fb_esp_rtdb_request_info_t*) at .pio/libdeps/esp32dev/Firebase Arduino Client Library for ESP8266 and ESP32/src/rtdb/FB_RTDB.cpp line 2409
0x400f4ee5: FB_RTDB::handleStreamRead(FirebaseData*) at .pio/libdeps/esp32dev/Firebase Arduino Client Library for ESP8266 and ESP32/src/rtdb/FB_RTDB.cpp line 741
0x400f4f29: FB_RTDB::readStream(FirebaseData*) at .pio/libdeps/esp32dev/Firebase Arduino Client Library for ESP8266 and ESP32/src/rtdb/FB_RTDB.cpp line 636
0x400f4f95: FB_RTDB:: ::_FUN(void *) at .pio/libdeps/esp32dev/Firebase Arduino Client Library for ESP8266 and ESP32/src/rtdb/FB_RTDB.cpp line 889
```

- Using Partition for Huge app ( 3MB , 1MB spiffs)
- Checking my free heap before starting firebase connection about `140KB from 254KB`
- After firebase connection established free heap decreased to `93KB from 254KB`
- Heap after doing BLE function also stable on `93KB from 254KB`

**IDE and its version:**
 - PlatformIO `core:6.1.4` and `home:3.4.3`

**ESP32 Arduino Core SDK version**
 - Version 4.4.0
 - 
**Another libraries**
- NimBLE-Arduino version 1.3.6

**Additional context** 
Question : 
is streamcallback cannot be interrupted with some `delay()` function ? 
because when my listener handling stream data from rtdb, it will scan to nearby device about 5 seconds (time blocking) and sending command to nearby node (about 30-100 devices nearby) and trying to filtering each devices.
Why i asking this because NimBLE connect and retrieve some BLE attributes, it take some seconds.

mobizt, 2022-10-27:
That is your application that based on your code which you should manage the memory usage from available memory yourself.
Most of memory used by the library is dynamic allocation which taken by mbedTLS library approx 70k per server openning session which reducing much free heap. 

The stack memory in ESP32 core is limited to 4 kB.

closed 2022-10-27 by mobizt

mrezzaaa, 2022-10-27:
So, you suggest to increasing ESP32 core stack memory ?

mobizt, 2022-10-27:
I did not see the original error message that show the error detail that related to stack or heap.

The minimum free heap for mbedTLS to work is 100k.

The FirebaseData object create the instance of TCP client that utilized the mbedTLS for SSL/TLS data encryption/decryption.

Then when you use it for stream, which the connection must be kept open, the heap memory used by this all the time is approx. 70k

And if this object used for other tasks to update and read the database which the connection will keep open by default to reduce the time for SSL/TLS handshake every time, the memory will use all the time approx. 70k.

For the stack limit by hard code changing the file in core SDK which you can search google for how to do it.

mrezzaaa, 2022-10-27:
How do you think what if i connect firebase with `xTaskCreatePinnedToCore` to maintain connection still open ?

mobizt, 2022-10-27:
It's not related to the session (http connection).

The http connection is normally closed by server after returning the response and can be kept open by client by sending the "Connection: keep-alive" header with the request.