mobizt/Firebase-ESP32 #247: [Question] How can I solve 'token is not ready' problem?

jatu-studiobox, 2022-11-03:
Hi,

I have some problem about 'token'. I use 'email & password' authentication with Firebase. If ESP32 don't working for a while, and then came back to work with RTDB. There is an error 'token is not ready'. 

Please advise me, how can I solve this problem?

mobizt, 2022-11-03:
The library uses token for authentication except for legacy token (database secret) which will be expired in 1 hour after created and need to be refresh. 

You should run `Firebase.ready()` in loop to handle the handle the authentication process or before when ever you want to access the Firebase services, like this.

https://github.com/mobizt/Firebase-ESP32/blob/19d4e3efd83c3fb61da0da8bb434174ceae202c4/examples/Basic/Basic.ino#L134-L139

The return value from `Firebase.ready()` is the authentication status.

Please follow the examples first and read the comments in the examples for the details and important information.

closed 2022-11-03 by mobizt

jatu-studiobox, 2022-11-03:
Thank you very much for your help. :)