mobizt/Firebase-ESP32 #248: [BUG] After update Firebase-ESP32 library from version 4.1.0 to version 4.2.0, the error 'method not allow' is occur.

jatu-studiobox, 2022-11-03:
**Describe the bug**
After update Firebase-ESP32 library from version 4.1.0 to version 4.2.0, the error 'method not allow' is occur. (version4.1.0 work fine.)
Some update methods (setTimestamp, setInt) have error 'method not allow'. Sometime, data in RTDB lost when it is updated.

**To Reproduce**
call method 'setTimestame' -> result error
call method 'setInt' -> result error

**Expected behavior**
Data is correctly updated. 

**Screenshots**
[image]

[image]

[image]

**IDE and its version:**
 - Arduino IDE
 - Version 1.8.19

**ESP32 Arduino Core SDK version**
 - Version 2.0.5

mobizt, 2022-11-04:
The update is not affected to all Firebase operations included request and response management i.e. no changes in the RTDB operations.

The update is related to the TCP client which now shared usage from user assigned FirebaseData object and used in authentication process as normally when authentication process was performing, the user FirebaseData object will not be able to connect to the server as usual.

The library v4.2.0 works normally with intensively test with all examples.

Please note that using Strings and string literal joining or concatenation are not allowed in all library functions that accept string as parameter.

I recommend removing all installed Firebase libraries in Arduino libraries folder and install the latest version from library manager to prevent the compiler wrong file links from several files with the same name issue.

If you feel that there is bug, please verify your code changes then create clean and simple example code that replicate the bug post it and describe the bug here.

closed 2022-11-04 by mobizt

mobizt, 2022-11-04:
Running this example with latest library [version 4.2.1](https://github.com/mobizt/Firebase-ESP32/blob/master/examples/Timestamp/Timestamp.ino) is fine.

jatu-studiobox, 2022-11-04:
Ok. After I follow your advise and update to version 4.2.1, it back to work usual.

Thank you very much :)