mobizt/Firebase-ESP32 #262: ESP32 Multipath Streaming with large Json tree

Liew279, 2023-01-27:
Hi, I'm new to GitHub, and also firebase. I was wondering if the multipath.ino example can be extended to larger Json trees.

For example:
{
 "System":{
     "FoodDispenser": {
            "DateTime":{
                    "DateTime0": "XXXXX",
                    "DateTime1": "XXXXX"
            },
           "Dispense": 1
     },
     "WaterDispenser": {
           "Level": "YYYY",
           "Dispense":1
     }
  }
}
If I used /System as the parent node and /FoodDispenser and /WaterDispenser as child nodes, would the multipath feature still give the specific nodes that have changed?

Will appreciate the help. Thanks!

Liew279, 2023-01-27:
Apologize for the formatting. Basically I'm asking if my chosen child node has many more subchilds, would the callback still give the specific node down to the bottom?

mobizt, 2023-01-28:
The `multiPathStream` is the normal Stream with JSON pre-parsing for the specific child nodes via `stream.get` function.

Normally you should get the values from parsing as manually JSON parsing.

Instead of using `multiPathStream`, I recommend using Sream instead and parse the stream data yourself.

For Realtime stream event data reading, you should keep the data under stream path small as possible.

The data under stream path should not grow or append the new data to it which you should think of time for large data transferring as well as the device memory limit in mind.

closed 2023-01-28 by mobizt

Liew279, 2023-01-28:
Thanks a lot man! I know how to learn the things I need now 😁