mobizt/Firebase-ESP32 #263: ESP32 unable to connect to multipath stream

Liew279, 2023-01-28:
Hi, I am receiving the following error for multipath stream:
Can't begin stream connection
REASON: unknown error

The following is my code:
`#include <WiFiManager.h>
#include <WiFi.h>
#include <FirebaseESP32.h>
#include "addons/TokenHelper.h"
#include "addons/RTDBHelper.h"
#include <ESP32Servo.h>
#include <HX711.h>

WiFiManager wm;
#define TRIGGER_PIN 0
int timeout = 120;  //time period the portal stays open

//#define API_KEY "AIzaSyCCGwHWrq6hfrVKE71MWXOHS-6vBxSFbtg"
#define DATABASE_URL "https://capstone-773c6-default-rtdb.firebaseio.com/"
// Define Firebase Data object
FirebaseData stream, fbdo;
FirebaseAuth auth;
FirebaseConfig config;
bool signupOK = false;

bool streamTO = true; //used for storing stream timeout status
unsigned long sendDataPrevMillis = 0;
String parentPath = "/FeederScheduling";
String childPath[2] = {"/Date", "/dispense"};
size_t childPathSize = 2;

const int buttonwifi = 0; //internal pullup resistor
const int servopin = 36;
const int echof = 35;
const int trigf = 32;
const int Data = 33;
const int clk = 25;
const int Buzzer = 27;

const int buttonw = 13;
const int relaypump = 23;
const int relaysuck = 22;
const int sig = 21;
const int power = 19;

Servo myservo;
int servoPosition = 0;
int servoDegrees = 2;
const int servoMinDegrees = 0; // the limits to servo movement
const int servoMaxDegrees = 90;
unsigned long currentMillis = 0;
unsigned long previousServoMillis = 0; // the time when the servo was last moved
unsigned long intervalservo = 10; //20 - very slow interval
bool startFlag = true;
HX711 LoadCell; //load sensor
#define calfactor 101.73
int weightperserving = 57;   //in grams (cat needs 1/3 - 1/2 cup, which is around 37-57g)
int weightinput = 0;
bool IsDispense = false;

String dateTime[2];  //store the input date and time
uint16_t count = 0;

void handleStreamValue(String Path, String value);
void streamCallback(MultiPathStreamData stream) {
  //if stream timeout status is true, set it to false and return (ignore only the first stream since the stream is first established or reconnected)
  if (streamTO) {
    streamTO = false;
    return;
  }
  size_t numChild = sizeof(childPath) / sizeof(childPath[0]);
  for (size_t i = 0; i < numChild; i++) {
    if (stream.get(childPath[i])) {
      Serial.println("path: " + stream.dataPath + ", type: " + stream.type + ", value: " + stream.value);
    }
  }
  handleStreamValue(stream.dataPath, stream.value);   //parse the stream
  Serial.println();
}

void streamTimeoutCallback(bool timeout) {
  //store stream timeout status
  streamTO = timeout;
  if (timeout) {
    Serial.println();
    Serial.println("Stream timeout, resume streaming...");
    Serial.println();
  }
}

void setup() {
  Serial.begin(115200);
  
  if(!wm.autoConnect("ESP32TEST_AP")){
    Serial.println("failed to connect and hit timeout");
    ESP.restart();
    delay(1000);
  }
  Serial.println("WiFi connected");
  pinMode(TRIGGER_PIN, INPUT_PULLUP);
  //config.api_key = API_KEY;
  config.database_url = DATABASE_URL;
  config.signer.test_mode = true;  //optional, delete for non-test mode
  /*if (Firebase.signUp(&config, &auth, "", "")) {
    Serial.println("signUp OK");
    signupOK = true;
  } else {
    Serial.printf("%s\n", config.signer.signupError.message.c_str());
  }*/

  config.token_status_callback = tokenStatusCallback;
  Firebase.begin(&config, &auth);
  Firebase.reconnectWiFi(true);
  /*if (!Firebase.RTDB.beginStream(&fbdo_s1, "/Food/Schedule/DateTime"))
    Serial.printf("stream 1 begin error, %s\n\n", fbdo_s1.errorReason().c_str());
    if (!Firebase.RTDB.beginStream(&fbdo_s2, "/Food/amount"))
    Serial.printf("stream 2 begin error, %s\n\n", fbdo_s2.errorReason().c_str());*/
  if (!Firebase.beginMultiPathStream(stream, parentPath, childPath, childPathSize))
  {
    Serial.println("------------------------------------");
    Serial.println("Can't begin stream connection...");
    Serial.println("REASON: " + stream.errorReason());
    Serial.println("------------------------------------");
    Serial.println();
  }
  //Set the reserved size of stack memory in bytes for internal stream callback processing RTOS task.
  //8192 is the minimum size.
  Firebase.setMultiPathStreamCallback(stream, streamCallback, streamTimeoutCallback);

  myservo.attach(36);
  myservo.write(servoMinDegrees);  //set servo to initial position
  pinMode(Buzzer, OUTPUT);   //Buzzer for dispenser
  LoadCell.begin(Data, clk);  //Load cell

  Serial.begin(115200);
}

void loop() {
  /*if (digitalRead(TRIGGER_PIN) == LOW) {
    wm.resetSettings();   //for testing
    wm.setConfigPortalTimeout(timeout);
    if (!wm.startConfigPortal("OnDemandAP")) { //can be password protected
      Serial.println("failed to connect and hit timeout");
      delay(3000);
      ESP.restart();
      delay(5000);
    }
    Serial.println("connected...yeey :)");
  }*/

  float templevel = ultrasonic(trigf, echof);
  if (Firebase.ready() && ((millis() - sendDataPrevMillis > 2000) || sendDataPrevMillis == 0)) {
    sendDataPrevMillis = millis();
    Serial.printf("Set Foodlevel... %s\n", Firebase.setFloat(fbdo, F("/Foodlevelread"), templevel) ? "ok" : fbdo.errorReason().c_str());
    /*if(IsWaterLow()){
      Serial.printf("Set Low Water Level Alert... %s\n", Firebase.setString(fbdo, F("/Waterlevelread"), "WATER LEVEL LOW!") ? "ok" : fbdo.errorReason().c_str());}
      else{
      Serial.printf("Set Low Water Level Alert... %s\n", Firebase.setString(fbdo, F("/Waterlevelread"), "Water Level Sufficient") ? "ok" : fbdo.errorReason().c_str());
      }*/
  }

  struct tm *CT = getTime();
  //store date and time in format YYYY:MM:DDTHH:MM
  String currdateTime = String(CT->tm_year) + "-" +
                        (CT->tm_mon < 10 ? "0" : "") + String(CT->tm_mon) + "-" +
                        (CT->tm_mday < 10 ? "0" : "") + String(CT->tm_mday) + "T" +
                        (CT->tm_hour < 10 ? "0" : "") + String(CT->tm_hour) + ":" +
                        (CT->tm_min < 10 ? "0" : "") + String(CT->tm_min);

  if (currdateTime == dateTime[0] || currdateTime == dateTime[1]){
    startFlag = true;   //start the servo
    buzzer();
  }
  if (startFlag)
    servosweep();
}

void servosweep () {
  currentMillis = millis();
  if (currentMillis - previousServoMillis >= intervalservo) {
    previousServoMillis = currentMillis;
    if (getWeight() < weightperserving) {
      if (servoPosition <= servoMaxDegrees) {
        servoPosition += servoDegrees;
      }
    }
    else {
      servoPosition -= servoDegrees;
    }
    myservo.write(servoPosition);
  }
  if (servoPosition == 0) {
    startFlag = false;
  }
}

float getWeight() { //Callibrate..............................
  float weight;
  if (LoadCell.is_ready()) {
    float reading = LoadCell.read();
    weight = (reading + 90200) / calfactor;
    if (weight < 0) { //if Weight fluctuates to a negative value
      weight = 0.0;
    }
    Serial.print("HX711 reading: ");
    Serial.print(weight);
    Serial.println(" g");
  } else {
    Serial.println("HX711 not found.");
  }
  return weight;
}

void buzzer() {
  tone(Buzzer, 1000); // Send 1KHz sound signal...
  delay(1000);        // ...for 1 sec
  noTone(Buzzer);     // Stop sound...
  delay(1000);        // ...for 1sec
}

float ultrasonic (int TRIGPIN, int ECHOPIN) {
  digitalWrite(trigf, LOW);
  delayMicroseconds(2);
  // Sets the trigPin HIGH (ACTIVE) for 10 microseconds
  digitalWrite(trigf, HIGH);
  delayMicroseconds(10);
  digitalWrite(trigf, LOW);
  // Reads the echoPin, returns the sound wave travel time in microseconds
  int duration = pulseIn(echof, HIGH);
  // Calculating the distance
  float distance = duration * 0.034 / 2; // Speed of sound wave divided by 2 (go and back)

  return distance;
}

int getWaterLevel() {    //calibrate water level.............................
  digitalWrite(power, HIGH);  // Turn the sensor ON
  delay(10);              // wait 10 milliseconds
  int val = analogRead(sig);    // Read the analog value form sensor
  digitalWrite(power, LOW);   // Turn the sensor OFF
  return val;             // send current reading
}

struct tm * getTime() {
  time_t now = time(NULL);
  struct tm *local_time = localtime(&now);
  return local_time;
}

void handleStreamValue(String Path, String value) {
  if (Path == "/dispense") {
    IsDispense = value.toInt();
  }
  else if (Path == "/Date/DateTime0") {
    dateTime[0] = value;
  }
  else if (Path == "/Date/DateTime1") {
    dateTime[1] = value;
  }
}`

Liew279, 2023-01-28:
My wifi is connected, but somehow the stream doesn't connect. It's just at the beginning, so I have no clue at all. Will appreciate the help!

mobizt, 2023-01-29:
You should update the library which the condition of "Unknown error" should not exist.

Stream is resumable. Don't be worried about the first Stream connection attempt. 

The database URL should be `capstone-773c6-default-rtdb.firebaseio.com` without `https://` at the beginning and `/` at the end.

You should read the examples comments for the guidelines.

Liew279, 2023-01-29:
Hi, thanks for the reply. I ran the Multipath.ino example and whenever a json is set, I get a "stream timed out, resuming" output on the serial monitor, for all the json.set. Does this mean my stream is connected?

Liew279, 2023-01-29:
I tested this code, a little edit to Multipath.ino. I am not getting any callbacks :((

`#include <WiFiManager.h>
#include <WiFi.h>
#include <FirebaseESP32.h>
// Provide the token generation process info.
#include <addons/TokenHelper.h>

// Provide the RTDB payload printing info and other helper functions.
#include <addons/RTDBHelper.h>

#include <ESP32Servo.h>
#include <HX711.h>

WiFiManager wm;
#define TRIGGER_PIN 0
int timeout = 120;  //time period the portal stays open

/* 1. Define the WiFi credentials */
#define WIFI_SSID "WIFI_AP"
#define WIFI_PASSWORD "abcd1234"

// For the following credentials, see examples/Authentications/SignInAsUser/EmailPassword/EmailPassword.ino

/* 2. Define the API Key */
#define API_KEY "AIzaSyCCGwHWrq6hfrVKE71MWXOHS-6vBxSFbtg"

/* 3. Define the RTDB URL */
#define DATABASE_URL "esp32-firebase-demo-46c3d-default-rtdb.asia-southeast1.firebasedatabase.app" //<databaseName>.firebaseio.com or <databaseName>.<region>.firebasedatabase.app

/* 4. Define the user Email and password that alreadey registerd or added in your project */
#define USER_EMAIL "USER_EMAIL"
#define USER_PASSWORD "USER_PASSWORD"

// Define Firebase Data object
FirebaseData fbdo;
FirebaseData stream;

FirebaseAuth auth;
FirebaseConfig config;

unsigned long sendDataPrevMillis = 0;

String parentPath = "/FeederScheduling";
String childPath[2] = {"/Date", "/dispense"};

int count = 0;
bool signupOK = false;
bool IsDispense = false;

void streamCallback(MultiPathStreamData stream)
{
  size_t numChild = sizeof(childPath) / sizeof(childPath[0]);

  for (size_t i = 0; i < numChild; i++)
  {
    if (stream.get(childPath[i]))
    {
      Serial.printf("path: %s, event: %s, type: %s, value: %s%s", stream.dataPath.c_str(), stream.eventType.c_str(), stream.type.c_str(), stream.value.c_str(), i < numChild - 1 ? "\n" : "");
      handleStreamValue(stream.dataPath, stream.value);   //parse the stream
    }
  }
  
  Serial.println();

  // This is the size of stream payload received (current and max value)
  // Max payload size is the payload size under the stream path since the stream connected
  // and read once and will not update until stream reconnection takes place.
  // This max value will be zero as no payload received in case of ESP8266 which
  // BearSSL reserved Rx buffer size is less than the actual stream payload.
  Serial.printf("Received stream payload size: %d (Max. %d)\n\n", stream.payloadLength(), stream.maxPayloadLength());
}

void streamTimeoutCallback(bool timeout)
{
  if (timeout)
    Serial.println("stream timed out, resuming...\n");

  if (!stream.httpConnected())
    Serial.printf("error code: %d, reason: %s\n\n", stream.httpCode(), stream.errorReason().c_str());
}

String dateTime[2];  //store the input date and time

void setup()
{

  Serial.begin(115200);

  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  Serial.printf("Firebase Client v%s\n\n", FIREBASE_CLIENT_VERSION);

  /* Assign the api key (required) */
  config.api_key = API_KEY;

  /* Assign the RTDB URL (required) */
  config.database_url = DATABASE_URL;

  if (Firebase.signUp(&config, &auth, "", "")) {  //last two empty - Anonymous user (See Auth library example), dont forget to enable auth method in firebase
    Serial.println("ok");
    signupOK = true;
  }
  else {
    Serial.printf("%s\n", config.signer.signupError.message.c_str());
  }

  /* Assign the callback function for the long running token generation task */
  config.token_status_callback = tokenStatusCallback; // see addons/TokenHelper.h

  Firebase.begin(&config, &auth);

  Firebase.reconnectWiFi(true);

  if (!Firebase.beginMultiPathStream(stream, parentPath))
    Serial.printf("sream begin error, %s\n\n", stream.errorReason().c_str());

  Firebase.setMultiPathStreamCallback(stream, streamCallback, streamTimeoutCallback);

}

void loop()
{

  // Firebase.ready() should be called repeatedly to handle authentication tasks.
}

void handleStreamValue(String Path, String value) {
  if (Path == "/dispense") {
    IsDispense = value.toInt();
  }
  else if (Path == "/Date/DateTime0") {
    dateTime[0] = value;
  }
  else if (Path == "/Date/DateTime1") {
    dateTime[1] = value;
  }
}`

mobizt, 2023-01-29:
As I replied above, you should read all comments in the examples which you should follow.

`Firebase.ready` function plays the important roles for this library for authentication handling and should be executed repeatedly because we use short-live token for Google Firebase API authentication which will be expired in 59 minutes and need to re-authenticate. 

I already recommend to use [Stream](https://github.com/mobizt/Firebase-ESP32/blob/master/examples/DataChangesListener/Callback/Callback.ino) instead which you can manage your data yourself.

With Stream you will know what is actually stream response received and why the JSON parsing does not return what you expected.

You should print the stream response for debugging before process that data.

mobizt, 2023-01-29:
When you want to post the c++ code in Github, please insert your code inside the markup like this.

\`\`\`cpp

// your code here

\`\`\`

avinashboy, 2024-02-03:
Hi, it's me again. I have an issue while trying to access the multi-stream, which has been declared by `stream` and `firebaseData`, one for changes happening in RTDB and to update the RTDB respectively. So, after looking into your documentation, I understand a little bit. So here is my code. 

```cpp

#define TINY_GSM_MODEM_SIM7600
#define TINY_GSM_USE_GPRS true
#define TINY_GSM_USE_WIFI false
#define TINY_GSM_MUX_COUNT 10
#define SerialAT Serial1
#include <TinyGsmClient.h>
#include <Arduino.h>
#include <Firebase_ESP_Client.h>

#include "addons/TokenHelper.h"
#include "addons/RTDBHelper.h"

// Firebase project API Key
#define API_KEY "xxxxx"

// Authorized Email and Password
#define USER_EMAIL "xxxxxxx"
#define USER_PASSWORD "xxxxxxxxx"

// GSM Module Pin Definitions
const int MODEM_RX = 27;
const int MODEM_TX = 26;
const int POWER_PIN = 4; // Assuming this is used to power cycle the modem

// Network Credentials
const char APN[] = "";
const char GPRS_USER[] = "";
const char GPRS_PASS[] = "";
#define GSM_PIN ""

const int LEDSS = 2;

String rxString;

const char* myConstant = "";
String listenerPath = "l/" + String(myConstant) + "/a";

// Firebase RTDB URL
#define DATABASE_URL "test-xxxxxxxx-xxxxx-firebasedatabase.app"

// Define Firebase objects
FirebaseData firebaseData;
FirebaseData stream;
FirebaseAuth auth;
FirebaseConfig config;

// Configure TinyGSM
TinyGsm modem(SerialAT);
TinyGsmClient gsmClient(modem, 0);
TinyGsmClient gsmClient2(modem, 1);

void streamCallback(FirebaseStream data) {
  Serial.printf("stream path, %s\nevent path, %s\ndata type, %s\nevent type, %s\n\n",
                data.streamPath().c_str(),
                data.dataPath().c_str(),
                data.dataType().c_str(),
                data.eventType().c_str());
  //printResult(data);  //see addons/RTDBHelper.h
  Serial.println();

  // Get the path that triggered the function
  String streamPath = String(data.dataPath());

  // if the data returned is an integer, there was a change on the GPIO state on the following path /{gpio_number}
  if (data.dataTypeEnum() == fb_esp_rtdb_data_type_integer) {
    String gpio = streamPath.substring(1);
[... 121 lines cut ...]

        // Construct the data to post
        FirebaseJson json;
        json.set("RSSI", rssi);
        json.set("BER", ber);
        json.set("timestamp", networkTime);

        // Post the data to Firebase
        if (Firebase.RTDB.setJSON(&firebaseData, path.c_str(), &json)) {
        Serial.println("Data posted to Firebase successfully.");
        } else {
        Serial.print("Failed to post data to Firebase: ");
        Serial.println(firebaseData.errorReason().c_str());
        }

       }
      
  }

  delay(15000); // Delay to slow down the loop
}

```

Please look into it and give feedback, which would be great for me.

mobizt, 2024-02-03:
@avinashboy 

You should not use delay as it blocks your device and library from operation too.

avinashboy, 2024-02-03:
Not Working 

This is Serial Monitor Output

```
21:29:39.441 -> Waiting for network...
21:29:42.108 -> Network connected
21:29:42.108 -> Connecting to APN: airtelgprs.com
21:29:42.404 -> GPRS connected
21:29:48.160 -> stream path, /l/XX-XX-XX-XXX-XXX-XXXX/a
21:29:48.160 -> event path, /
21:29:48.160 -> data type, json
21:29:48.160 -> event type, put
21:29:48.160 -> 
21:29:48.160 -> 
21:29:48.160 -> Count: 1
21:29:48.160 -> 
21:29:48.160 -> ---------
21:29:48.160 -> STATE-2: 2
21:29:48.160 -> GPIO-2:0
21:29:48.160 -> 
21:29:48.160 -> Received stream payload size: 46 (Max. 46)
21:29:48.204 -> 
21:29:48.500 -> GPRS connection lost. Reconnecting...
21:29:49.543 -> Waiting for network... success
21:29:49.543 -> Connecting to airtelgprs.com success
```

mobizt, 2024-02-03:
@avinashboy 

This is most important. Did you done this.
https://github.com/mobizt/Firebase-ESP32/blob/9c63f30b1a72093f16395a6b83ad13ecde0b5ecb/examples/ExternalClient/GSM/StreamCallback/StreamCallback.ino#L14-L16

avinashboy, 2024-02-03:
Yes, I have done this.

mobizt, 2024-02-03:
@avinashboy 

What is your issue as this is quite off topic.

avinashboy, 2024-02-03:
I don't know what to do. 🤷😢

mobizt, 2024-02-03:
@avinashboy

I don't understand what 's your problem and I can't help.

You should post only library related question.

Asking for programming help will not be accepted.

avinashboy, 2024-02-03:
My problem is that I am unable to post data to RTDB and also listen to the stream data. I am unable to handle multiple streams; I can only listen to one stream, which changes happening in the RTDB. So, that is my problem, and I am using the A7672 SimCom module for LTE. 

Anyway, I'm so sorry for disturbing you, and thank you for your time 🙏.

mobizt, 2024-02-03:
@avinashboy

You can use another FirebaseData object with another gsmclient on another socket (slot) for another stream path.

`TinyGsmClient gsmClient3(modem, 2);`

`FirebaseData stream2;`

`stream2.setGSMClient(&gsmClient3,....`

closed 2024-02-03 by mobizt