mobizt/Firebase-ESP32 #265: stream begin error, unknown error

6N5tr, 2023-02-11:
Im using this one... 

https://github.com/mobizt/Firebase-ESP32/blob/master/examples/ExternalClient/WiFi/StreamCallback/StreamCallback.ino 

but it doesnt work... two weeks ago with the same code it works perfectly but now it doesnt... Im using a stream callback with tesmode authentication.

For reproduce it... just try to use the code of the example. Need help. Thanks. 

Im working with:
esp32 
visual studio code 
PlatformIO IDE.
Firebase Client v4.3.5

mobizt, 2023-02-11:
Nothing I can do because of server issue.

You should use authentication and follow Google recommendation. Google may stop the services for this kind of usage as policy violation.

closed 2023-02-11 by mobizt

6N5tr, 2023-02-11:
> 

Well I just want to test this example withouth auth but im gonna try with auth and give feedback about this new try.

mobizt, 2023-02-11:
Please check the library installation. You should uninstall all Firebase libraries that found in libraries folder.

The Arduino IDE library installation from zip download and Library Manager updates causes the libraries in different version install in different folder that leads to error during compile and run.

6N5tr, 2023-02-11:
> Please check the library installation. You should uninstall all Firebase libraries that found in libraries folder.
> 
> The Arduino IDE library installation from zip download and Library Manager updates causes the libraries in different version install in different folders that leads to error during compile and run.
> 
> After delete all, install the latest version from Library Manager.

Feedback from test with auth: In the new version you cant use streamcallback for esp32 withouth auth or testmode. It works like two weeks ago but this one needs auth now.

mobizt, 2023-02-11:
You still can use test mode for RTDB but it's not recommended. There's nothing limit for auth and no-auth functional, but you need to check the library installation issue that I replied above.