mobizt/Firebase-ESP32 #267: I can't upload my text file to firebase storeage

dnshshkr, 2023-03-04:
I've followed the instructions and set all the parameter accordingly. The code seems fine when it comes to writing something into SPIFFS but when it comes to uploading it to the firbase storage it says "File not found". Please help, thanks.

mobizt, 2023-03-04:
This repository is for the library that works with RTDB database and legacy Firebase Cloud Messaging only.

mobizt, 2023-03-04:
You should read comments in the library examples carefully.

closed 2023-03-04 by mobizt

mobizt, 2023-03-04:
That is because file name you provide to the upload function does not exist in flash filesystem.

You can upload the sketch data using the upload tool plugin in Arduino IDE by creating the data folder with your files inside at the same path of your sketch file. 

The simplest way to test without upload sketch data is testing with download example first.
Then test upload example with the file that you download previously.

dnshshkr, 2023-03-04:
Hi, I’ve tried the DownloadFile and it worked fine. It can download and read the file in Firebase Storage. But when I swap to UploadFile and change mime type to “text/plain”, the same message appeared “File not found”.

mobizt, 2023-03-04:
Don't just say because it can't help anything.

You should post your code and all the debug message on serial to see what's wrong in your code.

mobizt, 2023-03-04:
Please use Github instead of using Email reply.

dnshshkr, 2023-03-04:
I’ve posted it on GitHub thanks.

mobizt, 2023-03-04:
As I said, you should post your code not snippets and all Serial debug.

Don't waste my time with just little information.

dnshshkr, 2023-03-04:
Sorry, I’ve posted the full one.

mobizt, 2023-03-04:
The issue will be fixed soon. I will inform you when it is available.