mobizt/Firebase-ESP32 #269: BUG

ecodetech, 2023-03-29:
**Device Name: ESP32 , ROM:4MB**
I am running web-server and Firebase tasks on core1 of ESP32. For some reason, after the Firebase.ready returns true,  and other RTDB and firestore operations are running or  not. The core gets so busy that I cannot pull web-server response from the device. Also I am running a mod-bus tasks on the same, it also gets stopped.
Previously I was using Firebase Lib version _4.2.7_. This issue did not occur in it and still doesn't if I compile with it. 
The issue occurred when I updated to _4.3.8_.
**No error is reported on UART.**
```
// some applciation
xTaskCreatePinnedToCore( runMainApp , "mainApp", 1024, NULL, 0, &runMainApp_handle, 1);
// WiFi connectivity management
  xTaskCreatePinnedToCore( appWiFiMgmtTask, "wifiTask", 2048, NULL, 0, &appWiFiMgmtTask_handle, 0);
// handling and serving Webserver API, html pages
  xTaskCreatePinnedToCore( start_webserver, "serverTask", 8000, NULL, 0, &start_webserver_handle, 1);
//  firebase task
  xTaskCreatePinnedToCore( iot_core_app , "iotTask", 20480, NULL, 0, &iot_core_app_handle, 1);
// mdobus task
  xTaskCreatePinnedToCore( customModbusApp, "modbus", 2048, NULL, 0, NULL, 1);
```
Almost all these tasks run fine with _v4.2.7_ for application. The only reason I am updating to _v4.3.8_ is that ```Firebase.ready()``` returns true even when it is not connected to Firebase.

Has anybody encountered similar issues?

I would be happy to use _v4.2.7_ only if I could fix ```Firebase.ready()``` to return false if not connected.

**Expected behavior**
No matter when I try to refresh the device webpage in browser, the web-server on device shall server me pages anyway. Also Firebase connectivity should be stable as it is.

**IDE and its version:**
 - Arduino,
 - Version [e.g. 1.8.18]

**ESP32 Arduino Core SDK version**
 - Version [e.g. 2.0.7]

mobizt, 2023-03-29:
Library is intensively test 24/7. I don't see library related issue from your code. The update is not the cause.

You should debug your code first. You should monitor the memory usage, memory leaks and free heap.

closed 2023-03-29 by mobizt

mobizt, 2023-03-29:
The function `Firebase.ready` will return true only when token is valid (not expired), legacy token was used, and network connection is ok otherwise return false.

You should not rely on ESP32 core v2.0.7 which contains bugs and should waits for newer version or stay at 2.0.6.

mobizt, 2023-03-29:
If you are using stream with callback in RTDB, please consider calling `Firebase.runStream()` in loop() function.

```
void loop()
{
    Firebase.runStream(); 

    // other codes here

}
```
The function Firebase.runStream will handle the stream under the main loop instead of internally create RTOS stream task.

The RTOS stream task will be deleted and free it reserved memory, and stream result will send to the callback as usual.

And if your problem is still persisted, you should create minimal, clean and clear code (no third-party libs included) that represent the issue and describe the issue then post it here.

mobizt, 2023-03-29:
You should set your `FreeRTOS` tasks priority more than 0 (`tskIDLE_PRIORITY`).

Some user reports that his Adafruit sensor library stops working when using stream with callback (the stream task created) which is very strange, and it may be false report.

The library's stream task has set to run on core 1 with priority of 3, 8192 bytes reserved stack and 10 ms delays for each iteration.

If this is the task priority issue, you can set your task priority to 3 or more.