mobizt/Firebase-ESP32 #281: ESP32 - Calling Firebase.setIdToken() cause crash

tjozsef, 2023-07-16:
**Describe the bug**
Calling     _Firebase.setIdToken(&config, "drhdtjhdfxtjdtj", 3600 /* expiry time */, "ydtjydtjydtjtydj" /* refresh token */);_ with non empty token, and refreshtoken arguments cause crash, if called before Firebase.begin().

**Steps to reproduce the behavior:**
1. Download the SignInWithIDToken example: https://github.com/mobizt/Firebase-ESP32/tree/master/examples/Authentications/SignInWithIDToken
2. Set your WiFi credentials in the sketch
3. Set random characters as token and refreshtoken calling Firebase.setIdToken() 
4. Build, and upload the sketch to the ESP32

**Expected behavior**
Calling Firebase.setIdToken() with non empty token and refreshtoken should not cause crash, even if called before Firebase.begin().

**Console log**
[  2742][D][WiFiGeneric.cpp:1098] _eventCallback(): STA IP: 192.168.1.9, MASK: 255.255.255.0, GW: 192.168.1.254
Connected with IP: 192.168.1.9
Firebase Client v4.3.16

Setting id token:
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400e3d68  PS      : 0x00060130  A0      : 0x800d7d2d  A1      : 0x3ffb2140  
A2      : 0x00000000  A3      : 0x0000000c  A4      : 0xffffffff  A5      : 0x3f401402
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800e3d68  A9      : 0x3ffb2110
A10     : 0x0000012f  A11     : 0x3f8008e0  A12     : 0x000000ff  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000000c  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x400858b5  LEND    : 0x400858bd  LCOUNT  : 0x00000027

Backtrace: 0x400e3d65:0x3ffb2140 0x400d7d2a:0x3ffb2160 0x400d32ac:0x3ffb21f0 0x400ea172:0x3ffb2290

**IDE and its version:**
 - PlatformIO Version v3.3.0
 
**ESP32 Arduino Core SDK version**
 - Version 2.0.0
 
**Additional context**
LilyGo-T-Call-SIM800 1.4v,
 Calling  Firebase.setCustomToken() with random characters as token does not cause crash, even if called before Firebase.begin().

mobizt, 2023-07-16:
Thanks for reporting this issue.

The [new version library](https://github.com/mobizt/Firebase-ESP32/releases/tag/v4.3.17) is available, please update.

closed 2023-07-16 by mobizt

tjozsef, 2023-07-17:
Thanks @mobizt for the fix. 
My goal was to use the deep sleep in ESP32, meanwhile I save the token which I aquired with email, and password signin, just to skip the token generation steps as much possible.