mobizt/Firebase-ESP32 #282: FCM OAUth2.0 with Ethernet not work

back920096, 2023-07-19:
hello author
I would like to ask how to use firebase FCM OAUth2.0 with Ethernet
I am using esp32s3
Irebase FCM OAUth2.0 can be used normally when I am on wifi
Ethernet doesn't work

mobizt, 2023-07-19:
I assumed that you mean HTTPv1 API cloud messaging.
This library does not support that, only legacy API supported. 

You should use [this library](https://github.com/mobizt/Firebase-ESP-Client) instead.

If you already used that library, you post in the wrong repository.

For ethernet with native support PHY (e.g., LAN8720) please follow [this example](https://github.com/mobizt/Firebase-ESP-Client/tree/main/examples/RTDB/BasicEthernet/ESP32).

For other SPI ethernet module (MAC), please use external client by follow [this example](https://github.com/mobizt/Firebase-ESP-Client/tree/main/examples/ExternalClient/RTDB/Ethernet/ESP32/Basic) and [this example](https://github.com/mobizt/Firebase-ESP-Client/tree/main/examples/ExternalClient/RTDB/Ethernet/Others/Basic).

When you use external ethernet client, it is not in this library scope, and you should debug your ethernet client yourself.

closed 2023-07-19 by mobizt

mobizt, 2023-07-19:
You should post your code or create the simple code to replicate the issue and describe more error details.

back920096, 2023-07-19:
But won't the old version of FCM stop serving next year?

Suwatchai K.

mobizt, 2023-07-19:
I have no idea. You should ask Google supports.

back920096, 2023-07-20:
Hello author, I would like to ask if there is any way to get the Bearer
token in the esp32+w5500 solution?
I plan to send FCM via http after getting the token
I refer to https://github.com/mobizt/ESP-Signer
I want to get token through this project
Create an account according to the process and implement it but it is not
successful
Here is my code and error message

Thank you very much for your help
[image: image.png]
/**
* This example show how to get access token using ESP32 and LAN8720
Ethernet module.
*
* Created by K. Suwatchai (Mobizt)
*
* Email: ***@***.***
*
* Github: https://github.com/mobizt/ESP-Mail-Client
*
* Copyright (c) 2023 mobizt
*
*/

/**
* There are many sources for LAN8720 and ESP32 interconnection on the
internet which may
* work for your LAN8720 board.
*
* Some methods worked unless no IP is available.
*
* This modification and interconnection provided in this example are mostly
worked as
* the 50 MHz clock was created internally in ESP32 which GPIO 17 is set to
be output of this clock
* and feeds to the LAN8720 chip XTAL input.
*
* The on-board LAN8720 50 MHz XTAL chip will be disabled by connect its
enable pin or pin 1 to GND.
*
* Please see the images in the folder "modified_LAN8720_board_images" for
how to modify the LAN8720 board.
*
* The LAN8720 Ethernet modified board and ESP32 board wiring connection.
*
* ESP32 LAN8720
*
* GPIO17 - EMAC_CLK_OUT_180 nINT/REFCLK - LAN8720 XTAL1/CLKIN 4k7 Pulldown
* GPIO22 - EMAC_TXD1 TX1
* GPIO19 - EMAC_TXD0 TX0
* GPIO21 - EMAC_TX_EN TX_EN
* GPIO26 - EMAC_RXD1 RX1
* GPIO25 - EMAC_RXD0 RX0
* GPIO27 - EMAC_RX_DV CRS
* GPIO23 - MDC MDC
* GPIO18 - MDIO MDIO
* GND GND
* 3V3 VCC
*
*/

//In case of Gmail, to send the Email via port 465 (SSL), less secure app
option should be enabled in the account settings.
https://myaccount.google.com/lesssecureapps?pli=1

#include <Arduino.h>
#include <Ethernet.h>
#include <ESP_Signer.h>

/** These credentials are taken from Service Account key file (JSON)
* https://cloud.google.com/iam/docs/service-accounts
*/
// See
https://github.com/mobizt/ESP-Signer#how-to-create-service-account-private-key

#define PROJECT_ID "The project ID" //Taken from "project_id" key in JSON
file.
#define CLIENT_EMAIL "Client Email" //Taken from "client_email" key in JSON
file.
const char PRIVATE_KEY[] PROGMEM = "-----BEGIN PRIVATE KEY-----\\n-----END
PRIVATE KEY-----\n";

#define WIZNET_RESET_PIN 8 // Connect W5500 Reset pin to GPIO 8 of ESP32
#define WIZNET_CS_PIN 10 // Connect W5500 CS pin to GPIO 10 of ESP32
#define WIZNET_MISO_PIN 13 // Connect W5500 MISO pin to GPIO 13 of ESP32
#define WIZNET_MOSI_PIN 11 // Connect W5500 MOSI pin to GPIO 11 of ESP32
#define WIZNET_SCLK_PIN 12 // Connect W5500 SCLK pin to GPIO 125 of ESP32
uint8_t Eth_MAC[] = {0x02, 0xF0, 0x0D, 0xBE, 0xEF, 0x01};
bool eth_connected = false;

void networkConnection()
{

Ethernet.init(WIZNET_CS_PIN);
ResetEthernet();
Serial.println("Starting Ethernet connection...");
while (Ethernet.linkStatus() != LinkON){
Serial.println("wait ethline");
Serial.println(Ethernet.localIP());
delay(500);
}
Serial.println("ethline finish");
IPAddress ip = Ethernet.localIP();
// Ethernet.begin(Eth_MAC);
while(!Ethernet.begin(Eth_MAC,1000,4000))
{
Serial.println(".");
}

while (ip == IPAddress(0, 0, 0, 0)) {
delay(500);
ip = Ethernet.localIP();
Serial.println("wait ip");
}
Serial.println(Ethernet.localIP());
Serial.println(ip);
while (ip == IPAddress(0, 0, 0, 0)) {
delay(500);
ip = Ethernet.localIP();
Serial.println("wait ip");
}
unsigned long to = millis();
while (Ethernet.linkStatus() == LinkOFF)
{
Serial.println("Wait Ethernet connection...");
delay(100);
}
Serial.print("Connected with IP ");
eth_connected = true;
Serial.println(Ethernet.localIP());

}

void ResetEthernet()
{
Serial.println("Resetting WIZnet W5500 Ethernet Board... ");
pinMode(WIZNET_RESET_PIN, OUTPUT);
digitalWrite(WIZNET_RESET_PIN, HIGH);
delay(200);
digitalWrite(WIZNET_RESET_PIN, LOW);
delay(50);
digitalWrite(WIZNET_RESET_PIN, HIGH);
delay(200);
}

SignerConfig config;

bool beginReady = false;

void tokenStatusCallback(TokenInfo info);

void begin()
{
beginReady = true;

/* Assign the sevice account credentials and private key (required) */
config.service_account.data.client_email = CLIENT_EMAIL;
config.service_account.data.project_id = PROJECT_ID;
config.service_account.data.private_key = PRIVATE_KEY;

/* Expired period in seconds (optional). Default is 3600 sec.*/
config.signer.expiredSeconds = 3600;

/* Seconds to refresh the token before expired (optional). Default is 60
sec.*/
config.signer.preRefreshSeconds = 60;

config.token_status_callback = tokenStatusCallback;

config.signer.tokens.scope = "
https://www.googleapis.com/auth/spreadsheets,https://www.googleapis.com/auth/drive,https://www.googleapis.com/auth/drive.file
";
Signer.begin(&config);
}

void setup()
{
Serial.begin(115200);
Serial.println();
networkConnection();

}

void loop()
{
if (eth_connected && !beginReady)
{
begin();
}

//Check to status and also refresh the access token
bool ready = Signer.tokenReady();
if (ready)
{
int t = Signer.getExpiredTimestamp() - config.signer.preRefreshSeconds -
time(nullptr);

Serial.print("Remaining seconds to refresh the token, ");
Serial.println(t);
delay(1000);
}
}

void tokenStatusCallback(TokenInfo info)
{
if (info.status == esp_signer_token_status_error)
{
Serial.printf("Token info: type = %s, status = %s\n", Signer.getTokenType(
info).c_str(), Signer.getTokenStatus(info).c_str());
Serial.printf("Token error: %s\n", Signer.getTokenError(info).c_str());
}
else
{
Serial.printf("Token info: type = %s, status = %s\n", Signer.getTokenType(
info).c_str(), Signer.getTokenStatus(info).c_str());
if (info.status == esp_signer_token_status_ready)
Serial.printf("Token: %s\n", Signer.accessToken().c_str());
}
}

Suwatchai K.

mobizt, 2023-07-20:
You don't need to use ESP-Signer library.

You can update the library to [latest version 4.3.18](https://github.com/mobizt/Firebase-ESP-Client/releases/tag/v4.3.18).

And please don't forget to define this macro in [FirebaseFS.h](https://github.com/mobizt/Firebase-ESP-Client/blob/main/src/FirebaseFS.h)
`#define FB_ENABLE_EXTERNAL_CLIENT`

Then try [this example](https://github.com/mobizt/Firebase-ESP-Client/blob/main/examples/ExternalClient/FCM/Ethernet/ESP32/HTTPv1/SendMessage/SendMessage.ino).