mobizt/Firebase-ESP32 #290: "Connection Refused" after 10 - 15 minutes of using Basic.ino code.

rinzlerz, 2023-09-18:
So I want to test out the Firebase RTDB using Basic.ino code given in repo. However, I monitored it for quite some time and the serial monitor shows "Connection refused". I do not modify anything to the code yet except things that are required such as WIFI SSID,WIFI PASSWORD and etc as I just want to see it works for at least 1 hour for my project from Examples. Also this is the latest library of FirebaseESP32.

```
#include <Arduino.h>
#include <WiFi.h>
#include <FirebaseESP32.h>

// Provide the token generation process info.
#include <addons/TokenHelper.h>

// Provide the RTDB payload printing info and other helper functions.
#include <addons/RTDBHelper.h>

/* 1. Define the WiFi credentials */
#define WIFI_SSID ""
#define WIFI_PASSWORD ""

// For the following credentials, see examples/Authentications/SignInAsUser/EmailPassword/EmailPassword.ino

/* 2. Define the API Key */
#define API_KEY ""

/* 3. Define the RTDB URL */
#define DATABASE_URL "" //<databaseName>.firebaseio.com or <databaseName>.<region>.firebasedatabase.app

/* 4. Define the user Email and password that already registered or added to your project */
#define USER_EMAIL ""
#define USER_PASSWORD ""

// Define Firebase Data object
FirebaseData fbdo;

FirebaseAuth auth;
FirebaseConfig config;

unsigned long sendDataPrevMillis = 0;

unsigned long count = 0;

void setup()
{

  Serial.begin(115200);

  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  Serial.printf("Firebase Client v%s\n\n", FIREBASE_CLIENT_VERSION);

  /* Assign the api key (required) */
  config.api_key = API_KEY;

  /* Assign the user sign in credentials */
  auth.user.email = USER_EMAIL;
  auth.user.password = USER_PASSWORD;

  /* Assign the RTDB URL (required) */
  config.database_url = DATABASE_URL;

  /* Assign the callback function for the long running token generation task */
  config.token_status_callback = tokenStatusCallback; // see addons/TokenHelper.h
[... 153 lines cut ...]
 * called in between.
 *
 * If you needed to call arbitrary "get" and "set" based on condition or event, use another FirebaseData object to avoid the session
 * closing and reopening.
 *
 * 3. Use of delay or hidden delay or blocking operation to wait for hardware ready in the third party sensor libraries, together with stream functions e.g. Firebase.RTDB.readStream and fbdo.streamAvailable in the loop.
 *
 * Please use non-blocking mode of sensor libraries (if available) or use millis instead of delay in your code.
 *
 * 4. Blocking the token generation process.
 *
 * Let the authentication token generation to run without blocking, the following code MUST BE AVOIDED.
 *
 * while (!Firebase.ready()) <---- Don't do this in while loop
 * {
 *     delay(1000);
 * }
 *
*

```
[image]

 - Arduino IDE
 - Version e.g. 1.8.15

mobizt, 2023-09-18:
The error showed internet connection problem.

Please check your internet connection (not WiFi).

closed 2023-09-18 by mobizt

rinzlerz, 2023-09-18:
The internet connection works fine on my side. I did some testing and at exactly average of 12 minutes it started to show "connection refused"

mobizt, 2023-09-18:
It's not always work fine then you will see "connection refused".

mobizt, 2023-09-18:
Do not argue on this topic because it is your device, and you should debug it yourself.

rinzlerz, 2023-09-19:
I'm not arguing, I'm just discussing on this issue as I debugged everything from the code and hardware itself. Or else, I won't open this issue. Maybe you have a reference that would help for the situation that i'm having and that would be enough.