mobizt/Firebase-ESP32 #302: Token error: code: -4, message: connection lost

Nikitanagar, 2023-10-25:
I am using the GSM with esp32 and this error shows:

Token info: type = id token (GITKit token), status = error
Token error: code: -4, message: connection lost
Token info: type = id token (GITKit token), status = on request
Token info: type = id token (GITKit token), status = error
Token error: code: -4, message: connection lost
Token info: type = id token (GITKit token), status = on request

but with wifi, it is working fine.

mobizt, 2023-10-25:
You should follow [these examples](https://github.com/mobizt/Firebase-ESP32/tree/master/examples/ExternalClient/GSM) and adapt it with your GSM module.

closed 2023-10-25 by mobizt

Nikitanagar, 2023-10-25:
Yeah I use this example

mobizt, 2023-10-25:
You should make your GSM module works with TinyGSMClient before using the library.

This is not a library issue.

mobizt, 2023-10-25:
Your GSM module is actually SIM7600 compatible AT commands.

You should read every single line of comment in example carefully to set it up.

mobizt, 2023-10-25:
These pins should be changed to match your module.
`BAT_EN`, `RESET` and `PWR_PIN`

Nikitanagar, 2023-10-25:
In [this](src/FirebaseFS.h) file where i can change.

Nikitanagar, 2023-10-25:
Should i have define #define TINY_GSM_MODEM_SIM7600 only , or any other line also

mobizt, 2023-10-25:
Read it again (skip the typo lines)
```
// To allow TinyGSM library integration, the following macro should be defined in src/ESP_Mail_FS.h or
// your custom config file src/Custom_ESP_Mail_FS.h.
// #define TINY_GSM_MODEM_SIM7600
```

mobizt, 2023-10-25:
> These two lines comments are typo.
> 
> ```
> // To allow TinyGSM library integration, the following macro should be defined in src/ESP_Mail_FS.h or
> // your custom config file src/Custom_ESP_Mail_FS.h.
> ```
> 
> You should define the SIM7600 macro in [this file](https://github.com/mobizt/Firebase-ESP32/blob/master/src/FirebaseFS.h) instead. `#define TINY_GSM_MODEM_SIM7600`

Nikitanagar, 2023-10-25:
#include "./core/Firebase_Client_Version.h"
#if !FIREBASE_CLIENT_VERSION_CHECK(40408)
#error "Mixed versions compilation."
#endif

#ifndef FirebaseFS_H
#define FirebaseFS_H
#define TINY_GSM_MODEM_SIM7600 // Defined here is it right

#include <Arduino.h>
#include "./mbfs/MB_MCU.h"

mobizt, 2023-10-25:
It's ok, actually you can define it any place in that file.

Nikitanagar, 2023-10-25:
Thanks a ton mobizt. It's working fine.

avinashboy, 2024-01-31:
Hi, i'm also getting the same error 

```
17:02:48.807 -> Token info: type = id token (GITKit token), status = on request
17:02:53.795 -> Token info: type = id token (GITKit token), status = error
17:02:53.795 -> Token error: code: -4, message: connection lost
```

here is my code:

```
#define TINY_GSM_MODEM_SIM7600
#define SerialAT Serial1
#include <TinyGsmClient.h>
#include <Arduino.h>
#include <Firebase_ESP_Client.h>

#include "addons/TokenHelper.h"
#include "addons/RTDBHelper.h"

// Insert Firebase project API Key
#define API_KEY ""

// Insert Authorized Username and Corresponding Password
#define USER_EMAIL ""
#define USER_PASSWORD ""

const int MODEM_RX = 27;
const int MODEM_TX = 26;
const int powerPin = 4;
const int LED_BUILTINS = 2;

// Configure TinyGSM
TinyGsm modem(SerialAT);
TinyGsmClient client(modem);

// Insert RTDB URLefine the RTDB URL
#define DATABASE_URL ""

// Define Firebase objects
FirebaseData stream;
FirebaseData fbdo;
FirebaseAuth auth;
FirebaseConfig config;

const char* myConstant = "";
const char apn[] = "";
const char gprsUser[] = "";
const char gprsPass[] = "";

#define GSM_PIN ""

// Variables to save database paths
String listenerPath = "type/" + String(myConstant) + "/method";

void streamCallback(FirebaseStream data) {
  Serial.printf("stream path, %s\nevent path, %s\ndata type, %s\nevent type, %s\n\n",
                data.streamPath().c_str(),
                data.dataPath().c_str(),
                data.dataType().c_str(),
                data.eventType().c_str());
  //printResult(data);  //see addons/RTDBHelper.h
  Serial.println();

  Serial.printf("Received stream payload size: %d (Max. %d)\n\n", data.payloadLength(), data.maxPayloadLength());
}

void streamTimeoutCallback(bool timeout) {
  if (timeout)
    Serial.println("stream timeout, resuming...\n");
  if (!stream.httpConnected())
    Serial.printf("error code: %d, reason: %s\n\n", stream.httpCode(), stream.errorReason().c_str());
}

void simConInit(){
  pinMode(powerPin, OUTPUT);
  pinMode(LED_BUILTINS, OUTPUT);
  digitalWrite(powerPin, LOW);

  SerialAT.begin(115200, SERIAL_8N1, MODEM_RX, MODEM_TX);
  
  // Restart takes quite some time
  // To skip it, call init() instead of restart()
  Serial.println("Initializing modem...");
  if (!modem.init()) {
    Serial.println("Failed to restart modem, delaying 10s and retrying");
    return;
  }

  // Restart takes quite some time
  // To skip it, call init() instead of restart()
  Serial.println("Initializing modem...");
  if (!modem.restart()) {
    Serial.println("Failed to restart modem, delaying 10s and retrying");
    return;
  }

  Serial.println("Waiting for network...");
  if (!modem.waitForNetwork()) {
    Serial.println(" fail");
    delay(10000);
    return;
  }
  Serial.println(" success");

  if (modem.isNetworkConnected()) {
    Serial.println("Network connected");
  }

  // GPRS connection parameters are usually set after network registration
  Serial.print(F("Connecting to "));
  Serial.print(apn);
  if (!modem.gprsConnect(apn,gprsUser,gprsPass)) {
    Serial.println(" fail");
    delay(10000);
    return;
  }
  Serial.println(" success");

  if (modem.isGprsConnected()) {
    Serial.println("LTE module connected");
  }

  modem.setNetworkMode(38);

  digitalWrite(LED_BUILTINS, HIGH);

  // Assign the api key (required)
  config.api_key = API_KEY;

  // Assign the user sign in credentials
  auth.user.email = USER_EMAIL;
  auth.user.password = USER_PASSWORD;

  // Assign the RTDB URL (required)
  config.database_url = DATABASE_URL;

  // Assign the callback function for the long running token generation task */
  config.token_status_callback = tokenStatusCallback;  //see addons/TokenHelper.h

  stream.setGSMClient(&client, &modem, GSM_PIN, apn, gprsUser, gprsPass);

  Firebase.reconnectWiFi(true);

  stream.setBSSLBufferSize(2048 /* Rx buffer size in bytes from 512 - 16384 */, 1024 /* Tx buffer size in bytes from 512 - 16384 */);

  // Assign the maximum retry of token generation
  config.max_token_generation_retry = 5;

  // Initialize the library with the Firebase authen and config
  Firebase.begin(&config, &auth);

  if (!Firebase.RTDB.beginStream(&stream, listenerPath.c_str()))
    Serial.printf("stream begin error, %s\n\n", stream.errorReason().c_str());

  // Assign a calback function to run when it detects changes on the database
  Firebase.RTDB.setStreamCallback(&stream, streamCallback, streamTimeoutCallback);

  delay(2000);
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  

  // SIMCON INIT
  simConInit();

}

void loop() {
  // put your main code here, to run repeatedly:
  if (Firebase.isTokenExpired()) {
    Firebase.refreshToken(&config);
    Serial.println("Refresh token");
  }

  if (!modem.isGprsConnected()) {
    Serial.println("GPRS connection lost. Reconnecting...");
    if (!modem.gprsConnect(apn)) {
      Serial.println("Failed to reconnect to cellular network.");
      // Consider delaying or restarting the ESP
      delay(10000);
      return;
    }
    Serial.println("Reconnected to cellular network.");
  }

}

```

I'm a newbie. Please, can anyone tell me what went wrong with this code?

mobizt, 2024-01-31:
@avinashboy 

https://github.com/mobizt/Firebase-ESP32/blob/9c63f30b1a72093f16395a6b83ad13ecde0b5ecb/examples/ExternalClient/GSM/Basic/Basic.ino#L15-L17

avinashboy, 2024-01-31:
I have created src/CustomFirebaseFS.h, but I don't understand its purpose. Can someone explain it to me?

mobizt, 2024-01-31:
@avinashboy 
https://github.com/mobizt/Firebase-ESP32?tab=readme-ov-file#predefined-options

avinashboy, 2024-01-31:
Yeah, I got it. But I'm sorry for asking such a basic question: where can I find the file src/FirebaseFS.h?

mobizt, 2024-01-31:
@avinashboy 

You are misunderstanding the basic library working concept.  

The file `FirebaseFS.h` is the libraries configuration file which comes with the library, you don't have to create it.
https://github.com/mobizt/Firebase-ESP32/blob/master/src/FirebaseFS.h

You should add this `#define TINY_GSM_MODEM_SIM7600` in that file.

avinashboy, 2024-01-31:
Yeah, I understood that. Where can I find the file on my computer? I'm not familiar with that part in the Arduino IDE.

mobizt, 2024-01-31:
@avinashboy 

Why don't you know where the library installed?

You should ask such basic question in community forum or google.