mobizt/Firebase-ESP32 #309: Tried to send value from accelerometer to my firebase RTDB but all x y z values are 0

slashplusdash, 2023-11-23:
If I try to test only the accelerometer, the x y z values are not 0

void loop() {
  // Read accelerometer values
  xaxis = analogRead(xPin);
  yaxis = analogRead(yPin);
  zaxis = analogRead(zPin);

  // Print accelerometer values for debugging
  Serial.println("X-Axis: " + String(xaxis));
  Serial.println("Y-Axis: " + String(yaxis));
  Serial.println("Z-Axis: " + String(zaxis));
  
  // Check Firebase readiness and send data
  if (Firebase.ready() && signupOK && (millis() - sendDataPrevMillis > 1000 || sendDataPrevMillis == 0)) {
    sendDataPrevMillis = millis();

    if (Firebase.RTDB.setInt(&fbdo, "test/x", xaxis)) {
      Serial.println("PASSED: x-axis sent to Firebase");
    } else {
      Serial.println("FAILED to send x-axis to Firebase");
      Serial.println("REASON: " + fbdo.errorReason());
    }

    if (Firebase.RTDB.setInt(&fbdo, "test/y", yaxis)) {
      Serial.println("PASSED: y-axis sent to Firebase");
    } else {
      Serial.println("FAILED to send y-axis to Firebase");
      Serial.println("REASON: " + fbdo.errorReason());
    }

    if (Firebase.RTDB.setInt(&fbdo, "test/z", zaxis)) {
      Serial.println("PASSED: z-axis sent to Firebase");
    } else {
      Serial.println("FAILED to send z-axis to Firebase");
      Serial.println("REASON: " + fbdo.errorReason());
    }
  }
  
  delay(1000); // Add a delay to avoid overwhelming Firebase with requests
}

mobizt, 2023-11-23:
Because all values are zero.

ADC2 pins cannot be used if WIFI was used.

https://randomnerdtutorials.com/esp32-adc-analog-read-arduino-ide/

closed 2023-11-23 by mobizt

slashplusdash, 2023-11-23:
Thanks and I read all the documentation, does it mean that all ADC2 GPIO pins are disabled and cannot be used?

mobizt, 2023-11-23:
You should read the blog that I share.

mobizt, 2023-11-23:
https://docs.espressif.com/projects/esp-idf/en/v4.4/esp32/api-reference/peripherals/adc.html#adc-limitations