mobizt/Firebase-ESP32 #316: ESP-IDF support

Murtaza7745, 2024-01-24:
Hey Mobizt,

I am using ESP-IDF v5.1.2 in VSCode as an extension for ESP32 firmware development.

I am able to successfully add arduino as a component and added repo in the library and updated the CMakeLists.txt to use [https://github.com/mobizt/Firebase-ESP32](url) in ESP-IDF.

The issue i am facing here is i am connecting to Wifi in sta mode using esp-idf's API and am able to connect to wifi, but when i am trying to connect to firebase in test mode its not connecting.

Looking for some help or guidance in the right direction for this to work.

mobizt, 2024-01-24:
You have to debug it.

The error reason is already available by calling fbdo.errorReason().

Library depends on Arduino WiFi library for internet connection.

You have to use it via Arduino core WiFi.

mobizt, 2024-01-24:
I focus on Arduino at this time because of its flexibility and platform supports.

It is not necessary for making WiFi feature library in esp-idf platform and you are able to use it as Arduino component.

closed 2024-01-24 by mobizt