mobizt/Firebase-ESP32 #318: Connection Refused

Agil1307, 2024-02-12:
#include "Base64.h"
#include <WiFi.h>
#include "soc/soc.h"
#include "soc/rtc_cntl_reg.h"
#include "esp_camera.h"
#include "FirebaseESP32.h"
#include <FirebaseJson.h>
#include <FirebaseESP32HTTPClient.h>

const char *ssid = "U";
const char *password = "12345678";

String FIREBASE_HOST = "kotakpaket-20c6a-default-rtdb.firebaseio.com";
String FIREBASE_AUTH = "9fV8VRgHkPGNROHCmIeUzK5doqcBRsboanoVUI2F";
FirebaseData firebaseData;
void setup()
{
  WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0);
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  Serial.println();
  Serial.println("SSID: " + (String)ssid);
  Serial.println("Password: " + (String)password);

  WiFi.begin(ssid, password);
  long int StartTime = millis();
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    if ((StartTime + 10000) < millis())
      break;
  }

  if (WiFi.status() == WL_CONNECTED)
  {
    Serial.println("");
    Serial.print("Camera Ready! Use 'http://");
    Serial.print(WiFi.localIP());
    Serial.println("' to connect");
  }
  else
  {
    Serial.println("Connection failed");
    return;
  }

  camera_config_t config;
  config.ledc_channel = LEDC_CHANNEL_0;
  config.ledc_timer = LEDC_TIMER_0;
  config.pin_d0 = 5; // Y2_GPIO_NUM
  config.pin_d1 = 18; // Y3_GPIO_NUM
  config.pin_d2 = 19; // Y4_GPIO_NUM
  config.pin_d3 = 21; // Y5_GPIO_NUM
  config.pin_d4 = 36; // Y6_GPIO_NUM
  config.pin_d5 = 39; // Y7_GPIO_NUM
  config.pin_d6 = 34; // Y8_GPIO_NUM
  config.pin_d7 = 35; // Y9_GPIO_NUM
  config.pin_xclk = 0; // XCLK_GPIO_NUM
  config.pin_pclk = 22; // PCLK_GPIO_NUM
  config.pin_vsync = 25; // VSYNC_GPIO_NUM
  config.pin_href = 23; // HREF_GPIO_NUM
  config.pin_sscb_sda = 26; // SIOD_GPIO_NUM
  config.pin_sscb_scl = 27; // SIOC_GPIO_NUM
  config.pin_pwdn = 32; // PWDN_GPIO_NUM
  config.pin_reset = -1; // RESET_GPIO_NUM
  config.xclk_freq_hz = 20000000;
  config.pixel_format = PIXFORMAT_JPEG;
  // init with high specs to pre-allocate larger buffers
  if (psramFound())
  {
    config.frame_size = FRAMESIZE_UXGA;
    config.jpeg_quality = 10; // 0-63 lower number means higher quality
    config.fb_count = 2;
  }
  else
  {
    config.frame_size = FRAMESIZE_SVGA;
    config.jpeg_quality = 12; // 0-63 lower number means higher quality
    config.fb_count = 1;
  }
  // camera init
  esp_err_t err = esp_camera_init(&config);
  if (err != ESP_OK)
  {
    Serial.printf("Camera init failed with error 0x%x", err);
    delay(1000);
    ESP.restart();
  }
  // drop down frame size for higher initial frame rate
  sensor_t *s = esp_camera_sensor_get();
  s->set_framesize(s, FRAMESIZE_QQVGA); // VGA|CIF|QVGA|HQVGA|QQVGA   ( UXGA? SXGA? XGA? SVGA? )
  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  Firebase.reconnectWiFi(true);
  Firebase.setMaxRetry(firebaseData, 3);
  Firebase.setMaxErrorQueue(firebaseData, 30);
  Firebase.enableClassicRequest(firebaseData, true);

  FirebaseJson json;

  //String jsonData = "{\"photo\":\"" + Photo2Base64() + "\"}";
  json.setJsonData("{\"photo\":\"" + Photo2Base64() + "\"}");
  String photoPath = "/esp32-cam";
  if (Firebase.pushJSON(firebaseData, photoPath, json))
  {
    Serial.println(firebaseData.dataPath());
    Serial.println(firebaseData.pushName());
    Serial.println(firebaseData.dataPath() + "/" + firebaseData.pushName());
  }
  else
  {
    Serial.println(firebaseData.errorReason());
  }
}

void loop()
{
  delay(10000);
}

String Photo2Base64()
{
  camera_fb_t *fb = NULL;
  fb = esp_camera_fb_get();
  if (!fb)
  {
    Serial.println("Gagal mengambil gambar dari kamera");
    return "";
  }

  String imageFile = "data:image/jpeg;base64,";
  String encoded = base64::encode(fb->buf, fb->len);
  imageFile += encoded;

  esp_camera_fb_return(fb);
  return imageFile;
}

String urlencode(String str)
{
  String encodedString = "";
  char c;
  char code0;
  char code1;
  char code2;
  for (int i = 0; i < str.length(); i++)
  {
    c = str.charAt(i);
    if (c == ' ')
    {
      encodedString += '+';
    }
    else if (isalnum(c))
    {
      encodedString += c;
    }
    else
    {
      code1 = (c & 0xf) + '0';
      if ((c & 0xf) > 9)
      {
        code1 = (c & 0xf) - 10 + 'A';
      }
      c = (c >> 4) & 0xf;
      code0 = c + '0';
      if (c > 9)
      {
        code0 = c - 10 + 'A';
      }
      code2 = '\0';
      encodedString += '%';
      encodedString += code0;
      encodedString += code1;
      encodedString += code2;
    }
    yield();
  }
  return encodedString;
}
why connection refused?

mobizt, 2024-02-12:
There is memory issue (out of memory)

Don't store large base64 string in memory, you should save it to file and send it with file function instead.

closed 2024-02-12 by mobizt

mobizt, 2024-02-13:
https://github.com/mobizt/Firebase-ESP32/tree/master/examples/File

Agil1307, 2024-02-13:
what should I do with it?

mobizt, 2024-02-13:
> Don't store large base64 string in memory, you should save it to file and send it with file function instead.

avinashboy, 2024-02-15:
Hi @mobizt , i have doubt

How to start the Firebase client ESP library in ESP32? I mean, there is no network for 2 hours, but it has the Wi-Fi signal. After some time, the library can't detect whether the network is available or not after losing the refreshed token.

mobizt, 2024-02-15:
@avinashboy

It is quite off topic.
You have to debug your code and WiFi.

There are the token status callback and the error reason from `FirebaseData` that provided information of error.

It is not possible to connect to server if your AP can't access the internet.