mobizt/Firebase-ESP32 #320: BUG: build warning in .map file

Murtaza7745, 2024-03-01:
I am using ESP-IDF v5.1.2 with Arduino as a component in VSCode.

**Describe the bug**
I have successfully added the Firebase-ESP32 library to the arduino component and I am able to use the Firebase-ESP32 function API to Signup, connect and stream data successfully.

The problem here is when i comment out the firebase part from the code, meaning commenting the firebase header files as well
I am getting no warning in the map file.

But when uncomment them, i get this warning as below.

```
WARNING: Given section not found in any memory region.
Check whether the LD file is compatible with the definitions in get_mem_regions in esp_idf_size
WARNING: Given section not found in any memory region.
Check whether the LD file is compatible with the definitions in get_mem_regions in esp_idf_size
WARNING: Given section not found in any memory region.
Check whether the LD file is compatible with the definitions in get_mem_regions in esp_idf_size
WARNING: Given section not found in any memory region.
Check whether the LD file is compatible with the definitions in get_mem_regions in esp_idf_size
Total sizes:
Used static DRAM:   71596 bytes (  52984 remain, 57.5% used)
      .data size:   25356 bytes
      .bss  size:   46240 bytes
Used static IRAM:  130378 bytes (  41654 remain, 75.8% used)
      .text size:  129351 bytes
   .vectors size:    1027 bytes
Used Flash size : 1735255 bytes
           .text: 1356175 bytes
         .rodata:  378824 bytes
Total image size: 1890989 bytes (.bin may be padded larger)
```

**To Reproduce**
Steps to reproduce the behavior:
1. Add Arduino as a component in the project using ESP-IDF v5.1.2.
2. Add the Firebase-ESP32 library using the ./add_lib.sh from the path "components/arduino/tools"
3. Build the project with adding #include <FirebaseESP32.h> in main.c
4. See warning four times

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]

mobizt, 2024-03-01:
This is Arduino library, port or use on other platform is your own risk and not support.

closed 2024-03-01 by mobizt

Murtaza7745, 2024-03-01:
This is used in arduino environment only

mobizt, 2024-03-01:
It's Arduino library and I don't have to make it support for specific platform.

If I want to make such Firebase library for esp-idf I have to create all from scratch and use its SSL engine and http client instead of Arduino component.

mobizt, 2024-03-01:
The compiler warning is not related to library as Arduino component.
Library is just code and does nothing with memory mapping.

Murtaza7745, 2024-03-01:
Okay, thanks @mobizt  for clearing that up.