mobizt/Firebase-ESP32 #325: Does this arduino library block some GPIO ports when reading the RTDB?

CerMax, 2024-03-20:
I was trying to run a stepper motor through an ESP32 where the speed is determined by the value given in the firebase but every time the ESP tries to read the firebase data it stops the stepper motor. Might be a GPIO issue or the Stepper motor library I am using but knowing if the firebase library does any blocking action would be beneficial. i can't really post my code due to NDA restrictions

mobizt, 2024-03-20:
It is not async library, mcu may take time for waiting during sending and reading data.

You can run your code on another RTOS task that can run with higher priority.

mobizt, 2024-03-20:
You can try this [new async Firebase client](https://github.com/mobizt/FirebaseClient) library.

closed 2024-03-20 by mobizt

CerMax, 2024-03-20:
Thank you, i appreciate you for taking the time out of your day for solving my query