mobizt/Firebase-ESP32 #328: FCM API V1 always fail

armanArk, 2024-06-22:
**Just ask something**
Hello 
[image]
Does this Lib support Firebase cloud messaging V1 ?
i test to send but always getting fail

**IDE and its version:**
 - Arduino, PlatformIO
 - Version [e.g. 1.8.9]

**ESP32 Arduino Core SDK version**
 - Version [e.g. 2.0.2]

**Additional context**
Add any other context about the problem here.

mobizt, 2024-06-22:
Read the documentation.

closed 2024-06-22 by mobizt

armanArk, 2024-06-22:
[image]
i have running with FirebaseEsp client https://github.com/mobizt/Firebase-ESP-Client.git still get -108 error code, 
did you have sugestion?

armanArk, 2024-06-22:
[image]
i have running with your FirebaseEspClient https://github.com/mobizt/Firebase-ESP-Client.git but still get -108 error code, 
did you have any suggestion?