mobizt/Firebase-ESP32 #329: ESP32 :REALTIME DATABASE :FAILED to read from Firebase: connection refused

Drinso, 2024-06-28:
HI, I'm doing our school project and I encounter an error message it is FAILED to read from Firebase: connection refused its working if push the reset button but a few seconds later its now working and in arduino serial monitor says FAILED to read from Firebase: connection refused if i didn't push the rst button it will be not working .
[image]
here my code:

#include <WiFi.h>
#include <Firebase_ESP_Client.h>
#include "addons/TokenHelper.h"
#include "addons/RTDBHelper.h"

#define WIFI_SSID "B310_D71AA"
#define WIFI_PASSWORD "Mariya<3"
#define API_KEY "   "  // i remove my api key
#define DATABASE_URL "   " // i remove my database_url 

#define LIGHT_PIN 12
#define LIGHT2_PIN 13
#define LIGHT3_PIN 14
#define LIGHT4_PIN 27

#define SWITCH_PIN 19
#define SWITCH2_PIN 4
#define SWITCH3_PIN 5
#define SWITCH4_PIN 18
#define MAINSWITCH_PIN 33

FirebaseData fbdo;
FirebaseAuth auth;
FirebaseConfig config;

unsigned long sendDataPrevMillis = 0;
unsigned long wifiReconnectPrevMillis = 0;
unsigned long wifiConnectTimeout = 10000;  // 10 seconds WiFi connection timeout
unsigned long wifiConnectStartTime = 0;

unsigned long switchDebouncePrevMillis = 0;
unsigned long switch2DebouncePrevMillis = 0;
unsigned long switch3DebouncePrevMillis = 0;
unsigned long switch4DebouncePrevMillis = 0;

unsigned long mainswitchDebounceMillis = 0;

const unsigned long debounceDelay = 50;  // debounce delay in milliseconds

bool signupOK = false;

bool lightStatus = false;
bool light2Status = false;
bool light3Status = false;
bool light4Status = false;

bool lastSwitchStatus = HIGH;
bool lastSwitch2Status = HIGH;
bool lastSwitch3Status = HIGH;
bool lastSwitch4Status = HIGH;

bool lastMainswitchStatus = HIGH;
bool wifiConnected = false;
bool mainswitchStatus = false;

void connectToWiFi() {
  WiFi.disconnect(true);
  delay(1000);
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to WiFi...");
  wifiConnectStartTime = millis();
}

void setup() {
  pinMode(LIGHT_PIN, OUTPUT);
  pinMode(LIGHT2_PIN, OUTPUT);
[... 225 lines cut ...]
    checkFirebase();
  } else {
    if (WiFi.status() == WL_CONNECTED) {
      wifiConnected = true;
      Serial.println("Connected to WiFi");
      config.api_key = API_KEY;
      config.database_url = DATABASE_URL;
      if (Firebase.signUp(&config, &auth, "", "")) {
        Serial.println("signUp OK");
        signupOK = true;
      } else {
        Serial.printf("signUp failed: %s\n", config.signer.signupError.message.c_str());
      }
      config.token_status_callback = tokenStatusCallback;
      Firebase.begin(&config, &auth);
      Firebase.reconnectWiFi(true);
    } else if (millis() - wifiConnectStartTime >= wifiConnectTimeout) {
      Serial.println("Failed to connect to WiFi within timeout. Proceeding with local control.");
      wifiConnectStartTime = millis() + wifiReconnectPrevMillis;
    } else if (millis() - wifiReconnectPrevMillis >= 10000) {
      wifiReconnectPrevMillis = millis();
      connectToWiFi();
    }
  }
}

mobizt, 2024-06-28:
It is network or internet issue, and you should solve on your side.

Anyway, this library was deprecated and not active for development.
https://github.com/mobizt/Firebase-ESP32?tab=readme-ov-file#we-have-moved-to-the-new-library

closed 2024-06-28 by mobizt

mobizt, 2024-06-28:
If you re-connect the WiFi with your own code, you should remove `Firebase.reconnectWiFi` or use `Firebase.reconnectWiFi(false)`.

This is the user misuse. 

The function `Firebase.reconnectWiFi` is for allowing library to reconnect the WiFi instead of user code or third-party library e.g. WiFi Manager, otherwise, WiFi reconnection interferences  conflicts can be occurred.

Drinso, 2024-06-28:
what is the best or suitable code for my project ?
can you help me ? i'm a beginner in this project

Drinso, 2024-06-28:
i remove this Firebase.reconnectWiFi but it same  FAILED to read from Firebase: connection refused

mobizt, 2024-06-28:
This is for library issue tracking not a community forum.

The cause of the issue I am already told, and you have to solve because it depends on many factors.

You should check internet access because your device cannot connect to internet.

You have to check free Heap of your device because to connect to the server you should have enough Heap memory at least 100k.

The malfunction router and unstable WiFi hotspot are other causes which you have to check.